pymodbus-dev/pymodbus #694: Modbus Device Identification details overwrite

GayanJayakody, 2021-12-18:
### Versions

* Python: 3.7.0
* OS: Windows 10
* Pymodbus: 2.5.2

### Pymodbus Specific
* Server: tcp, sync
* Client: tcp sync

### Description

I am running two Pymodbus servers on the same machine in two threads. I have given two sets of identification information for each server. But Device information gets overwritten. (i.e. I always get the device information of the last created server even I request information from the first server)

### Server Code 

from pymodbus.server.sync import StartTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.version import version
from threading import Thread 

class ModbusServer:

    def __init__(self,_ip,_port,_vendorName,_ProductCode,_VendorUrl,_ProductName,_ModelName):
        self._ip = _ip
        self._port = _port
        store = ModbusSlaveContext(
            di = ModbusSequentialDataBlock(0, [0]*100),
            co = ModbusSequentialDataBlock(0, [0]*100),
            hr = ModbusSequentialDataBlock(0, [0]*100),
            ir = ModbusSequentialDataBlock(0, [0]*100))
        self.context = ModbusServerContext(slaves=store, single=True)

        # initialize the server information
        self.identity = ModbusDeviceIdentification(self)
        self.identity.VendorName = _vendorName
        self.identity.ProductCode = _ProductCode
        self.identity.VendorUrl = _VendorUrl
        self.identity.ProductName = _ProductName
        self.identity.ModelName = _ModelName
        self.identity.MajorMinorRevision = version.short()

    def _run(self):
        StartTcpServer(self.context, identity=self.identity, address=(self._ip, int(self._port)))
        
    def run(self):
        Thread(target=self._run).start()

Device1=ModbusServer("127.0.0.1",1500,"sample1","sample1","sample1","sample1","sample1")
Device1.run()
Device2=ModbusServer("127.0.0.1",1501,"sample2","sample2","sample2","sample2","sample2")
Device2.run()

# client code

from pymodbus.client.sync import ModbusTcpClient
from pymodbus.mei_message import ReadDeviceInformationRequest
from pymodbus.device import *

client = ModbusTcpClient('127.0.0.1','1500')
rq = ReadDeviceInformationRequest(unit=1,read_code=3)
rr = client.execute(rq)
print(rr.information)
client.close()

# result
{0: b'sample2', 1: b'sample2', 2: b'2.5.2', 3: b'sample2', 4: b'sample2', 5: b'sample2'}

But this is the result I wanted  {0: b'sample1', 1: b'sample1', 2: b'2.5.2', 3: b'sample1', 4: b'sample1', 5: b'sample1'}

zmatlik117, 2022-02-24:
I think it's caused by __data dictionary here:
https://github.com/riptideio/pymodbus/blob/master/pymodbus/device.py#L191
being defined as a class attribute instead of class INSTANCE attribute - and unfortunately is also mutable.
I had to drink a hell of a lot of whiskey before i figured that out.

But i don't think i know enough about pymodbus architecture to attempt to fix it right now. I mean - i certainly can - but i'm 90% sure that i'm missing something important, because this looks like it was intentional.

janiversen, 2022-05-31:
You thought correct, that should not be a class variable.

janiversen, 2023-02-27:
Latest dev branch, should no longer have this restriction.

closed 2023-02-27 by janiversen