pymodbus-dev/pymodbus #696: Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received))

tiagomiguelrs, 2021-12-23:
### Versions

* Python: 3.8.8
* OS: Windows 10
* Pymodbus: 2.5.3
* Modbus Hardware (if used): Omron CP1E-N30

### Pymodbus Specific
* Client: tcp - sync/async

### Description

Hi! I have searched for possible answers online regarding the  pymodbus package but I didn't find anything that would solve my problem.
So, I am trying to communicate with a PLC through direct ethernet connection from my windows 10 computer. 
It seems the connection is established, but there is no response from the PLC side because I get:

`2021-12-23 16:30:08,733 MainThread      DEBUG    transaction    :303      Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received))`

I am just a rookie when it comes to modbus and electrotecnic stuff and I have only recently started dabbling with PLCs and computer communication.
Even so, I think all settings are ok in the PLC and it may be an issue from some coding setting?

### Code and Logs

# Code

```
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient('192.168.250.1', port=9600, timeout=10)
client.connect()

rq = client.write_coil(1, True)
rr = client.read_coils(1,1)

client.close()
```

# Logs

```
2021-12-23 16:47:52,833 MainThread      DEBUG    sync           :215      Connection to Modbus server established. Socket ('192.168.250.100', 53503)
2021-12-23 16:47:52,834 MainThread      DEBUG    transaction    :139      Current transaction state - IDLE
2021-12-23 16:47:52,834 MainThread      DEBUG    transaction    :144      Running transaction 1
2021-12-23 16:47:52,835 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x5 0x0 0x1 0xff 0x0
2021-12-23 16:47:52,835 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2021-12-23 16:47:52,835 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-12-23 16:48:02,837 MainThread      DEBUG    transaction    :303      Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)) 
2021-12-23 16:48:02,838 MainThread      DEBUG    socket_framer  :147      Processing: 
2021-12-23 16:48:02,840 MainThread      DEBUG    transaction    :465      Getting transaction 1
2021-12-23 16:48:02,841 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2021-12-23 16:48:02,856 MainThread      DEBUG    sync           :215      Connection to Modbus server established. Socket ('192.168.250.100', 53505)
2021-12-23 16:48:02,856 MainThread      DEBUG    transaction    :139      Current transaction state - TRANSACTION_COMPLETE
2021-12-23 16:48:02,857 MainThread      DEBUG    transaction    :144      Running transaction 2
2021-12-23 16:48:02,857 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x1 0x0 0x1 0x0 0x1
2021-12-23 16:48:02,857 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2021-12-23 16:48:02,858 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-12-23 16:48:08,854 MainThread      DEBUG    transaction    :303      Transaction failed. ([WinError 10054] An existing connection was forcibly closed by the remote host) 
2021-12-23 16:48:08,855 MainThread      DEBUG    socket_framer  :147      Processing: 
2021-12-23 16:48:08,857 MainThread      DEBUG    transaction    :465      Getting transaction 2
2021-12-23 16:48:08,858 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

Thanks

dhoomakethu, 2021-12-27:
There is no response coming in back from the slave,  Have you tried using a unit-id with the requests ?
```
rq = client.write_coil(1, True, unit=<unit-id>)
rr = client.read_coils(1,1, unit=<unit-id>)
```

dhoomakethu, 2021-12-27:
Also, port 9600 is not a traditional modbus TCP port, Are you sure this device supports Modbus TCP ?

tiagomiguelrs, 2021-12-28:
> Also, port 9600 is not a traditional modbus TCP port, Are you sure this device supports Modbus TCP ?

So, I have been looking into the matter and I have to appologise, my rookie self thought that it only needed a TCP port to support Modbus TCP. I called the Omron support and they indeed confirmed me that it doesn't support TCP. 

I am sorry for wasting your time, hopefully it wasn't much and I know better now.

closed 2021-12-28 by tiagomiguelrs