pymodbus-dev/pymodbus #698: Doesn't construct payload if unit is greater than 0x9, unit=0x9

JimitD, 2022-01-06:
### Versions

* Python: 2.7
* OS: Raspbian
* Pymodbus: 2.5.3
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async: RTU Over TCP - sync
* Client: tcp/rtu/ascii - sync/async: RTU -sync

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

Trying to read Modbus registers of a device who has slave id of 9. 
In that case if I keep unit while reading the holding register to 9, it doesn't construct payload correctly. It's work until unit id is 8

### Code and Logs

```python
 # code and logs here.
all_registers = client.read_holding_registers(
                200-1, 52, unit=0x9)
```

janiversen, 2022-05-31:
This works in my test setup, can you please provide debug logs, so we can see why it is not working for you (maybe you do slave unit id = 0x09 is not online).

janiversen, 2022-09-24:
Closing as no longer actual.

closed 2022-09-24 by janiversen