pymodbus-dev/pymodbus #701: Modbus Error: [Connection] Failed to connect

EduardoRonchi, 2022-01-13:
### Versions

* Python: 3.7.3
* OS: Raspian (Buster)
* Pymodbus: 2.5.3
* Modbus Hardware (if used): Raspberry Pi 4B

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

What were you trying, what has happened, what went wrong, and what did you expect?
I am trying to communitate from my raspberry pi (client) to my computer windows 10 (master). In windows 10 (ip:192.168.0.6, port: 5020) I am running the example code from here [(https://pymodbus.readthedocs.io/en/latest/source/example/synchronous_server.html)]. 
In raspberry pi I am running a simple example called client.py (see below). But I receive an error:

### Code and Logs

```python
 # code and logs here.
 # client.py

from pymodbus.client.sync import ModbusTcpClient

client = ModbusTcpClient('192.168.0.6', port=5020)
client.write_coil(1, True)
client.close()

```
Log and error
```
Traceback (most recent call last):
  File "client.py", line 8, in <module>
    client.write_coil(1, True)
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/common.py", line 70, in write_coil
    return self.execute(request)
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.6:5020)]
```

When I try the master and client in same computer it works fine! I have created a rule in windows firewall to accept conections at port 5020. Any ideas to solve this?

dhoomakethu, 2022-01-14:
This looks like an issue with the network and not with the pymodbus library.

EduardoRonchi, 2022-01-14:
I guess it is a network issue. Anyone faced similar issues with windows 10?

closed 2022-02-19 by github-actions[bot]