pymodbus-dev/pymodbus #703: Fragmented TCP messages are not handled correctly

andersjohnsen, 2022-01-18:
As seen here

https://github.com/riptideio/pymodbus/blob/c5772b35ae3f29d1947f3ab453d8d00df846459f/pymodbus/framer/socket_framer.py#L160

if a modbus TCP packet is split between two or more fragments, the frame will be reset in between and will thus fail parsing. As TCP is a fully streaming protocol that supports splitting and merging data at will, this is leading to unreliable behavior. Simply removing the `resetFrame` call (and perhaps the log?) appears to fix the issue.

Jon-Salmon, 2022-03-15:
I'm not sure why this issue has been closed prematurely, this is a substantial bug in the Modbus TCP implementation.

swamper123, 2022-03-15:
This is caused by a bot, which is closing issues, if no activity happens on this issue.
Will reopen it.

closed 2022-09-05 by janiversen