pymodbus-dev/pymodbus #704: Slow read operation when sync serial client and servers are running at the same time

sjpark608, 2022-01-21:
### Versions

* Python:python3.7
* OS: Linux raspberrypi 5.10.63-v7l+
* Pymodbus: 2.5.3
* Modbus Hardware (if used):  none

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async: rtu-sync-serial
* Client: tcp/rtu/ascii - sync/async: rtu-sync-serial

### Description

I have two serial ports each configured as server and client. I am doing a speed test of reading a single and bulk reading of the holding register. Each server/client is configured with 300ms timeout and 9600 bps. The bulk read performs a single read command with reading 120 registers. The client measures the duration of the read operation using perf_counter() to measure the elapsed time of the read operation.  100ms for reading a single register and 300ms for reading 120 registers seems too slow.

When the client performs the same task with an external device, the performance becomes much better. Reading a single register takes about 20ms. This gives me an idea of where the issue is coming from the sync-server code. 

### Code and Logs
reading operation 
```
        read_start = perf_counter()
        data = modbus.read_regs(slave, address, num_of_regs)
        read_end = perf_counter()
```

Modbus context definition
```
    store = ModbusSlaveContext(
        di=ModbusSparseDataBlock({0x32:dummy_data}),
        co=ModbusSparseDataBlock({0x32:dummy_data}),
        hr=ModbusSparseDataBlock({0x32:dummy_data}),
        ir=ModbusSparseDataBlock({0x32:dummy_data})
    )
```

```python
 # code and logs here.

```

bsense-rius, 2022-01-26:
Same issue here.  In fact my guess is that the issue is in sync version of the server (not a hardware / rasp)

If i run the twisted version of the server, both sync and async client examples run ok, wihtout no issues passing all the asster tests.

BUT when i run the server in sync mode, results are really unereliable. Some times timesout, others frames are not complete etc.

I guess that deserves some investigation. 

@sjpark608 I suggest you try the asynchronous versions of the master (uses twisted package). I am not a fan of twisted, because for whatever reason when i install it, seems to change python filterwarnings(), some deprecationwarnings pop up my code. So, I test it and then uninstall, because i do not really need a serial slave simulator.

sjpark608, 2022-01-26:
@bsense-rius Thanks for sharing your experience and your suggestions. I will try the async version and let you know the progress.

dhoomakethu, 2022-01-27:
Do you have a pic of your setup ? Are the serial ports real (USB ports etc) or virtual serial ports ?

sjpark608, 2022-01-27:
> 

Hi, the schematic is deemed confidential, but both serial ports are two separate pairs of physical serial lines (RX1/TX1 and RX2/TX2) connected to a raspberry pi. I configured each line as a server and a client.

I also tried connecting one serial port on the raspberry pi to a laptop through a USB to serial cable using FTDI chip. And configured the laptop as server/client and the raspberry pi as client/server.  All synchronous server and client configs are synchronous. This also showed the same delay issue.

I have tried asynchronous server and synchronous client. The issue persists.

dhoomakethu, 2022-01-28:
Can you enable debug logs on both server and client and paste the output here?

sjpark608, 2022-01-29:
> Can you enable debug logs on both server and client and paste the output here?

Here is debug output of the server and client.

Server config port: ttyAMA1, baudrate 9600, timeout: 300ms
Client config port: ttyAMA0, baudrate 9600, timeout:300ms

Client: 10 single-register read operations
```
single regs test
Enter number of reads: 10
INFO:__main__:Reading Slave 247, 1 register at address 50
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0xf7 0x3 0x0 0x32 0x0 0x1 0x31 0x53
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0xf7 0x3 0x2 0x0 0x1 0xb1 0x91
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x2 0x0 0x1
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 247
DEBUG:pymodbus.transaction:Getting transaction 247
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
INFO:modbus_ftx:read regs: [1]
INFO:__main__:Read duration: 100ms
INFO:__main__:Reading Slave 247, 1 register at address 50
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:SEND: 0xf7 0x3 0x0 0x32 0x0 0x1 0x31 0x53
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1643419963.086268, Current Time stamp - 1643419963.091312
DEBUG:pymodbus.framer.rtu_framer:Waiting for 3.5 char before next send - 4.01 ms
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0xf7 0x3 0x2 0x0 0x1 0xb1 0x91
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x2 0x0 0x1
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 247
DEBUG:pymodbus.transaction:Getting transaction 247
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
INFO:modbus_ftx:read regs: [1]
INFO:__main__:Read duration: 303ms
INFO:__main__:Reading Slave 247, 1 register at address 50
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 3
DEBUG:pymodbus.transaction:SEND: 0xf7 0x3 0x0 0x32 0x0 0x1 0x31 0x53
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1643419963.389234, Current Time stamp - 1643419963.395681
DEBUG:pymodbus.framer.rtu_framer:Waiting for 3.5 char before next send - 4.01 ms
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0xf7 0x3 0x2 0x0 0x1 0xb1 0x91
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x2 0x0 0x1
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 247
DEBUG:pymodbus.transaction:Getting transaction 247
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
INFO:modbus_ftx:read regs: [1]
INFO:__main__:Read duration: 301ms
INFO:__main__:Reading Slave 247, 1 register at address 50
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 4
DEBUG:pymodbus.transaction:SEND: 0xf7 0x3 0x0 0x32 0x0 0x1 0x31 0x53
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1643419963.692148, Current Time stamp - 1643419963.697948
DEBUG:pymodbus.framer.rtu_framer:Waiting for 3.5 char before next send - 4.01 ms
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
[... 162 lines cut ...]
2022-01-28 17:32:44,892 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x0 0x32 0x0 0x1
2022-01-28 17:32:44,892 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-01-28 17:32:44,892 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-01-28 17:32:44,893 MainThread      DEBUG    context        :64       validate: fc-[3] address-51: count-1
2022-01-28 17:32:44,893 MainThread      DEBUG    context        :78       getValues fc-[3] address-51: count-1
2022-01-28 17:32:44,893 MainThread      DEBUG    sync           :144      send: [ReadHoldingRegistersResponse (1)]- b'f703020001b191'
2022-01-28 17:32:45,194 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x0 0x32 0x0 0x1
2022-01-28 17:32:45,195 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-01-28 17:32:45,195 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
[... cut ...]
2022-01-28 17:32:45,802 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-01-28 17:32:45,802 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-01-28 17:32:45,803 MainThread      DEBUG    context        :64       validate: fc-[3] address-51: count-1
2022-01-28 17:32:45,803 MainThread      DEBUG    context        :78       getValues fc-[3] address-51: count-1
2022-01-28 17:32:45,803 MainThread      DEBUG    sync           :144      send: [ReadHoldingRegistersResponse (1)]- b'f703020001b191'
```

dhoomakethu, 2022-01-30:
Thanks for the logs, I see you are logging the total consumed for each transaction. However, I want to know the time spent in each stage. Could you please update your logging to include the timestamp ?
```
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)
```
Apologies for not being explicit. Thanks in advance.

sjpark608, 2022-01-31:
> Thanks for the logs, I see you are logging the total consumed for each transaction. However, I want to know the time spent in each stage. Could you please update your logging to include the [...]

I just realized the client logging configuration was different from the server. The server and client configurations are the same as the previous logging.

<details><summary>Client: 10 single-register read operations</summary>

```

2022-01-31 09:58:37,142 MainThread      INFO     test_modbus    :23       Reading Slave 247, 1 register at address 50
2022-01-31 09:58:37,143 MainThread      DEBUG    transaction    :140      Current transaction state - IDLE
2022-01-31 09:58:37,143 MainThread      DEBUG    transaction    :145      Running transaction 1
2022-01-31 09:58:37,144 MainThread      DEBUG    transaction    :273      SEND: 0xf7 0x3 0x0 0x32 0x0 0x1 0x31 0x53
2022-01-31 09:58:37,144 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-01-31 09:58:37,145 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-01-31 09:58:37,216 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-01-31 09:58:37,216 MainThread      DEBUG    transaction    :297      RECV: 0xf7 0x3 0x2 0x0 0x1 0xb1 0x91
2022-01-31 09:58:37,217 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x2 0x0 0x1
2022-01-31 09:58:37,217 MainThread      DEBUG    factory        :266      Factory Response[ReadHoldingRegistersResponse: 3]
2022-01-31 09:58:37,217 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-01-31 09:58:37,218 MainThread      DEBUG    transaction    :454      Adding transaction 247
2022-01-31 09:58:37,218 MainThread      DEBUG    transaction    :465      Getting transaction 247
2022-01-31 09:58:37,218 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-01-31 09:58:37,219 MainThread      INFO     modbus_ftx     :148      read regs: [1]
2022-01-31 09:58:37,219 MainThread      INFO     test_modbus    :28       Read duration: 76ms
2022-01-31 09:58:37,219 MainThread      INFO     test_modbus    :23       Reading Slave 247, 1 register at address 50
2022-01-31 09:58:37,220 MainThread      DEBUG    transaction    :140      Current transaction state - TRANSACTION_COMPLETE
2022-01-31 09:58:37,220 MainThread      DEBUG    transaction    :145      Running transaction 2
2022-01-31 09:58:37,220 MainThread      DEBUG    transaction    :273      SEND: 0xf7 0x3 0x0 0x32 0x0 0x1 0x31 0x53
2022-01-31 09:58:37,221 MainThread      DEBUG    rtu_framer     :269      Changing state to IDLE - Last Frame End - 1643651917.215939, Current Time stamp - 1643651917.221246
2022-01-31 09:58:37,221 MainThread      DEBUG    rtu_framer     :277      Waiting for 3.5 char before next send - 4.01 ms
2022-01-31 09:58:37,226 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-01-31 09:58:37,226 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-01-31 09:58:37,520 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-01-31 09:58:37,520 MainThread      DEBUG    transaction    :297      RECV: 0xf7 0x3 0x2 0x0 0x1 0xb1 0x91
2022-01-31 09:58:37,521 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x2 0x0 0x1
2022-01-31 09:58:37,521 MainThread      DEBUG    factory        :266      Factory Response[ReadHoldingRegistersResponse: 3]
2022-01-31 09:58:37,522 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-01-31 09:58:37,522 MainThread      DEBUG    transaction    :454      Adding transaction 247
2022-01-31 09:58:37,522 MainThread      DEBUG    transaction    :465      Getting transaction 247
2022-01-31 09:58:37,523 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-01-31 09:58:37,523 MainThread      INFO     modbus_ftx     :148      read regs: [1]
2022-01-31 09:58:37,523 MainThread      INFO     test_modbus    :28       Read duration: 303ms
2022-01-31 09:58:37,523 MainThread      INFO     test_modbus    :23       Reading Slave 247, 1 register at address 50
2022-01-31 09:58:37,524 MainThread      DEBUG    transaction    :140      Current transaction state - TRANSACTION_COMPLETE
2022-01-31 09:58:37,524 MainThread      DEBUG    transaction    :145      Running transaction 3
2022-01-31 09:58:37,525 MainThread      DEBUG    transaction    :273      SEND: 0xf7 0x3 0x0 0x32 0x0 0x1 0x31 0x53
2022-01-31 09:58:37,525 MainThread      DEBUG    rtu_framer     :269      Changing state to IDLE - Last Frame End - 1643651917.520397, Current Time stamp - 1643651917.52533
2022-01-31 09:58:37,525 MainThread      DEBUG    rtu_framer     :277      Waiting for 3.5 char before next send - 4.01 ms
2022-01-31 09:58:37,530 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-01-31 09:58:37,530 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
[... cut ...]
2022-01-31 09:58:39,936 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-01-31 09:58:39,937 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-01-31 09:58:39,937 MainThread      DEBUG    context        :64       validate: fc-[3] address-51: count-1
2022-01-31 09:58:39,938 MainThread      DEBUG    context        :78       getValues fc-[3] address-51: count-1
2022-01-31 09:58:39,938 MainThread      DEBUG    sync           :144      send: [ReadHoldingRegistersResponse (1)]- b'f703020001b191'
```

</details>

dhoomakethu, 2022-02-01:
@sjpark608 thanks for the logs, I will take a look and try to reproduce the issue.

sjpark608, 2022-04-25:
@dhoomakethu is there any update regarding this issue?

aasick, 2022-04-28:
@dhoomakethu eagerly waiting for your response.

I am also trying same, like two serial cable connected with rtu configuration. One COM port act as server and another COM port act as server. but setting up address on server and reading exact value on slave I have facing issue.

I have created issue request on following link https://github.com/riptideio/pymodbus/issues/860

dhoomakethu, 2022-04-28:
@sjpark608 apologies, this got derailed. I will spend some time to see if this could be reproduced.

janiversen, 2022-08-18:
We now have a test that among others addresses this issue test/test_examples.py, furthermore I did some testing:
examples/server_sync + examples/client_sync using 2 USB rs-485 adapters, no delay, limiting factor were the baud rate.
examples/server_sync + examples/client_async using 2 USB rs-485 adapters, no delays same speed as above.
examples/server_async + examples/client_async using 2 USB rs-485 adapters, a bit faster but the limiting factor is still baud rate.

In general using a sync server is really asking for something slow.

janiversen, 2022-10-14:
3.0.0 solves this issue.

closed 2022-10-14 by janiversen