pymodbus-dev/pymodbus #706: creating an AsyncModbusSerialClient from within an existing loop hangs

bsense-rius, 2022-01-26:
### Versions

* Python: 3.8
* OS: Ubuntu 20.04
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: None
* Client: serial rtu - async

### Description

When calling AsynModbusSerialClient inside a running asyncio task (from a loop):

async def myCoro():
    ...
    loop, client = AsyncModbusSerialClient(schedulers.ASYNC_IO, loop=asyncio.get_running_loop(), ...)

Gets hang in async_io_serial() at future.result()

loop, client = AsyncModbusSerialClient(schedulers.ASYNC_IO, port=connPort, baudrate=connSpeed, method=connMethod, timeout=commTimeout, loop=runningLoop)

File "/home/smartvending/smartvending/venv/lib/python3.8/site-packages/pymodbus/client/asynchronous/serial.py", line 75, in __new__
    yieldable = factory_class(framer=framer, port=port, **kwargs)

File "/home/smartvending/smartvending/venv/lib/python3.8/site-packages/pymodbus/client/asynchronous/factory/serial.py", line 108, in async_io_factory
    future.result()

File "/usr/lib/python3.8/concurrent/futures/_base.py", line 434, in result
    self._condition.wait(timeout)

File "/usr/lib/python3.8/threading.py", line 302, in wait
    waiter.acquire()

I know rhis is something that has been somehow reported as an issue before but current implementation does not seem to integreate as expected.

I have worked it around by skipping the auto connection feature in async_io_serial()
    #coro = client.connect()
    #if loop.is_running():
    #    future = asyncio.run_coroutine_threadsafe(coro, loop=loop)
    #    future.result()
    #else:
    #    loop.run_until_complete(coro)
    #return loop, client

And simply calling await client.connect() at my code. Then works like a charm

PS: I did not actually patched my local lib, I just passed a fake loop object that has a loop.is_running() function that always returns False. It also has a loop.run_until_complete() function that does nothing inside. This makes the async_io_serial() to skip the autoconnection. In my code I then fix loop reference in client by the real running loop, before calling await client connect().
I am ashamed to share this workaround, but works without patching your nice code!

class fake_async_loop:
    def is_running(self):
        return False
    def run_until_complete(self, obj):
        return

async def myTask():
             fakeLoop = fake_async_loop()
             loop, client =  AsyncModbusSerialClient(schedulers.ASYNC_IO, port=connPort, baudrate=connSpeed, method=connMetho>
             # Fix loop. AsyncModbusSerialClient
             client.loop = asyncio.get_running_loop()
             await client.connect()

### Code and Logs

```python
 # code and logs here.

```

dhoomakethu, 2022-01-26:
what version of pymodbus is this?

bsense-rius, 2022-01-26:
Sorry! 
I missed the topic. It is pymodbus 2.5.3

dhoomakethu, 2022-01-27:
@bsense-rius can you share your script with out the work around? Thanks

bsense-rius, 2022-01-28:
Sure, here it goes: 

```
#
# TESTS MODBUS TEMPERATURE SENSOR WITH pymodbus LIB
# IN SYNC / ASYNC MODES
#

# MAIN CONFIGURATION VARIABLES (RUNMODE AND SENSOR TYPE)
runMode      = "ASYNC"        # "ASYNC" for async reading, "SYNC" for sync reading
clientCreate = "inside_loop"  # "inside_loop" / "outside_loop"
sensor       = "BOX"          # "TUBE" for tube shaped sensor, "BOX" for box shaped sensor
connPort     = '/dev/ttyS2'   # '/dev/ttyS3', '/tmp/ptyp0' => for simulated

# --------------------------------------------------------------------------- #
# slave parameters
# --------------------------------------------------------------------------- #
connMethod  = 'rtu'          # 'binary' 'ascii'

if sensor == "BOX":
    connSpeed   = 9600       # BOX SHAPED
else:
    connSpeed   = 4800       # TUBE SHAPED

commTimeout = 5              # Affects to all operations not just connection init!
UNIT        = 0x1            # Slave Id aka Slave Address ala Unit Id in modbus world

# --------------------------------------------------------------------------- #
# import the various client implementations
# --------------------------------------------------------------------------- #
import asyncio
from pymodbus.client.asynchronous import schedulers
from pymodbus.client.asynchronous.serial import AsyncModbusSerialClient
from pymodbus.client.sync import ModbusSerialClient as SyncModbusSerialClient

# --------------------------------------------------------------------------- #
# configure the client logging
# --------------------------------------------------------------------------- #
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s ' '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

import time

# --------------------------------------------------------------------------- #
# sync function
# --------------------------------------------------------------------------- #
def run_sync_client():
    log.debug(f'MODBUS SYNC CLIENT: Trying to open {connPort} @ {connSpeed} in {connMethod} mode for {sensor} sensor')
    client = SyncModbusSerialClient(method=connMethod, port=connPort, timeout=commTimeout, baudrate=connSpeed)
    client.connect()
    log.debug(f'CONNECTION SUCCESSFUL!')

    try:
        readsLeft = 10

        while readsLeft > 0:
            if sensor == "BOX":
                resp = client.read_input_registers(1, 2, unit=UNIT)
                assert(not resp.isError())          # test that we are not an error
                log.debug(f"Temperature {resp.registers[0]/10}ºC - Humidity {resp.registers[1]/10}%")

            else: # TUBE SHAPED
                resp = client.read_holding_registers(0, 2, unit=UNIT)
                assert(not resp.isError())          # test that we are not an error
                log.debug(f"Temperature {resp.registers[1]/10}ºC - Humidity {resp.registers[0]/10}%")

            time.sleep(1)
[... 74 lines cut ...]
# --------------------------------------------------------------------------- #
# main function
# --------------------------------------------------------------------------- #
if __name__ == "__main__":

    if runMode == "SYNC":
        run_sync_client()

    else:
        if clientCreate == "outside_loop":
        # - CREATE A CONNECTION (client) OUTSIDE ASYNCIO LOOP. It Works, we can the pass the client an asyncio loop with tasks
        # - BUT in many cases we need to create the client inside a running loop
        #
            log.debug(f'ASYNC MODBUS CLIENT OUTSIDE ASYNC TASK LOOP: Connect to {connPort} @ {connSpeed} in {connMethod} mode for {sensor} sensor')
            loop, client = AsyncModbusSerialClient(schedulers.ASYNC_IO, port=connPort, baudrate=connSpeed, method=connMethod, timeout=commTimeout)
            log.debug(f'CONNECTION SUCCESSFUL!')

            loop.run_until_complete(task_launcher(client))
            loop.close()

        else:   # clientCreate == "inside_loop"
            asyncio.run(task_launcher(None))
```

This has been tested with Modbus devices (different types of temperature sensors) and with your slave simulator.

tomas-bozek, 2022-04-12:
Hello,

i had a simmilar problem, 
this resolves problems for me

changed in `AsyncModbusSerialClient`

from
```py3
from self._connected_event = asyncio.Event() 
```

to 
```py3
self._connected_event = asyncio.Event(loop=loop)
```

janiversen, 2022-04-12:
This looks like a correct solution, but I suspect it affects the other client types as well.

janiversen, 2022-06-04:
Solved on dev (3.0.0prerelease).

closed 2022-06-04 by janiversen

wjbridge, 2022-06-15:
I am having this hang problem on `2.5.3`. When I try  `3.0.0dev4` and the latest commit as of today. I am getting a type error that `coro` is not a coroutine in file `serial.py`. I am trying to create  `AsyncModbusSerialClient(schedulers.ASYNC_IO, port=connPort, baudrate=connSpeed, method=connMethod, timeout=commTimeout)` in an existing running loop.

I don't think the coroutine was created correctly. What do you think?

Old:
`future = asyncio.run_coroutine_threadsafe(coro, loop=loop)` 

Proposed:
` future = asyncio.run_coroutine_threadsafe(coro(), loop=loop)`

``` py
def async_io_factory(port=None, framer=None, **kwargs):
    """Create asyncio based asynchronous serial clients.
    :param port:  Serial port
    :param framer: Modbus Framer
    :param kwargs: Serial port options
    :return: asyncio event loop and serial client
    """
    try:
        loop = kwargs.pop("loop", None) or asyncio.get_running_loop()
    except RuntimeError:
        loop = asyncio.new_event_loop()

    proto_cls = kwargs.get("proto_cls") or ModbusClientProtocol

    client = AsyncioModbusSerialClient(port, proto_cls, framer, loop, **kwargs)
    coro = client.connect
    if not loop.is_running():
        loop.run_until_complete(coro())
    else:  # loop is not asyncio.get_event_loop():
        future = asyncio.run_coroutine_threadsafe(coro, loop=loop)
        future.result()

    return loop, client
```