pymodbus-dev/pymodbus #713: updating_server.py error

Franck-PowerUp, 2022-02-09:
### Versions

* Python: 3.7
* OS: Ubuntu 20
* Pymodbus: 2.5.0
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: rtu - sync
* Client: rtu - sync

### Description

Error when i try the "updating_server.py" example, Exception occured line 55.

### Correction

the error is due to the data access, the example goes deeper than needed.

i propose:

```python
def updating_writer(a):
    """ A worker process that runs every so often and
    updates live values of the context. It should be noted
    that there is a race condition for the update.
    :param arguments: The input arguments to the call
    """
    log.debug("updating the context")
    register = 4  
    slave_id = 0x00
    slave = a[slave_id]
    address = 0x10
    values = slave.getValues(register, address, count=5)
    values = [v + 1 for v in values]
    log.debug("new values: " + str(values))
    slave.setValues(register, address, values)

```

dhoomakethu, 2022-02-12:
@Franck-PowerUp can you paste the traceback please?

Franck-PowerUp, 2022-02-14:
```bash
...
  File "/mnt/c/work/Temp/updating_server.py", line 55, in updating_writer
    values = context[slave_id].getValues(register, address, count=5)
TypeError: 'ModbusSlaveContext' object is not subscriptable
```

Franck-PowerUp, 2022-02-14:
Not an error, I've used threading.Timer instead of initial twisted.

closed 2022-02-14 by Franck-PowerUp