pymodbus-dev/pymodbus #717: windows10, asynchronous_server.py got an unexpected keyword argument 'timeout'

makaay2007, 2022-02-27:
### Versions

* Python:3.8
* OS:windows 10
* Pymodbus:2.5.3
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: rtu - async

### Description

I use the asynchronous_server.py as an example:
but I got an error, 
  File "C:\Users\22804\Documents\myPycharmEnv\pymodbusEnv\lib\site-packages\pymodbus\server\asynchronous.py", line 338, in StartSerialServer
    SerialPort(protocol, port, reactor, baudrate=baudrate, parity=parity,
TypeError: __init__() got an unexpected keyword argument 'timeout'

### Code and Logs

    StartSerialServer(context, identity=identity,
                      port='COM2', framer=ModbusRtuFramer)

```

twriventa, 2022-03-09:
I too am encountering the same issue with an asynchronous serial rtu server.

Python 3.8
Windows 10
pymodbus 2.5.3

File "C:\Riventa\Source\HMI\modbus_server\venv\lib\site-packages\pymodbus\server\asynchronous.py", line 338, in StartSerialServer
    SerialPort(protocol, port, reactor, baudrate=baudrate, parity=parity,
TypeError: __init__() got an unexpected keyword argument 'timeout'

`StartSerialServer(context, framer=ModbusRtuFramer, identity=identity,
                      port='COM4', baudrate=9600)`

janiversen, 2022-04-04:
Solved on dev (pre 3.0.0) but continue to exist in 2.5.3

closed 2022-06-14 by dhoomakethu