pymodbus-dev/pymodbus #719: REPL history file path hardcoded to ../.pymodhis

ehn, 2022-03-10:
### Versions

* Python: 3.8.3
* OS: macOS 12.2.1
* Pymodbus: 2.5.3
* Modbus Hardware (if used): N/A

### Description

It [looks like](https://github.com/riptideio/pymodbus/blob/c5772b35ae3f29d1947f3ab453d8d00df846459f/pymodbus/repl/client/main.py#L172) the history file of the REPL is hardcoded to `../.pymodhis`. If I run `pymodbus.console` in my home directory, I get this error:

```
Unhandled exception in event loop:
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/input/vt100.py", line 168, in callback_wrapper
    callback()
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/application/application.py", line 691, in read_from_input
    self.key_processor.process_keys()
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/key_binding/key_processor.py", line 274, in process_keys
    self._process_coroutine.send(key_press)
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/key_binding/key_processor.py", line 186, in _process
    self._call_handler(matches[-1], key_sequence=buffer[:])
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/key_binding/key_processor.py", line 329, in _call_handler
    handler.call(event)
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/key_binding/key_bindings.py", line 102, in call
    result = self.handler(event)
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/shortcuts/prompt.py", line 798, in _accept_input
    self.default_buffer.validate_and_handle()
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/buffer.py", line 1877, in validate_and_handle
    self.append_to_history()
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/buffer.py", line 1385, in append_to_history
    self.history.append_string(self.text)
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/history.py", line 73, in append_string
    self.store_string(string)
  File "/Users/ehn/.pyenv/versions/3.8.3/lib/python3.8/site-packages/prompt_toolkit/history.py", line 294, in store_string
    with open(self.filename, "ab") as f:

Exception [Errno 13] Permission denied: '../.pymodhis'
```

If I run it somewhere where I have write access to `..`, I don't get the error message. Can we replace `../.pymodhis` with something like `~/.pymodhis`?

PaulKGrimes, 2022-04-21:
This is a not an enhancement request.  It prevents this software project being used.

janiversen, 2022-04-21:
well this is also something you can easily patch, so even though it is a bug it does not prevent you from using pymodbus.

janiversen, 2022-04-21:
not to mention, that you could make a sub directory in a place where you have access and start your program there. So lot of solutions.

PaulKGrimes, 2022-04-21:
Yes, there are workarounds.  But it's still a bug, not an enhancement.

PaulKGrimes, 2022-04-21:
> That depends on the view, if a program does not work as documented it is a
> bug, if others want it to work differently it is an enhancement.

It does not work as documented, depending on the permissions of the parent
directory to your current working directory.  The program crashes
unexpectedly, when following the example given in the readme, so there is a
bug.

janiversen, 2022-04-21:
if you say do….but I do not think it is documented that it should be able to run in a place where you do not have permissions.

But lets cut it short, this is a open source projects, so feel free to submit a patch.

PaulKGrimes, 2022-04-21:
The example does not run in a working directory where you do have permissions, because it requires permissions in the parent of the working directory.  This is very unusual behavior for a project.

janiversen, 2022-04-21:
as I wrote earlier, patches are welcome.

ehn, 2022-04-22:
Separately from not working unless you have write permissions to the parent directory of where you run the software, it will also spread history files throughout your directory structure if invoked in various places, and the history won't work as expected, since it will depend on where in the directory structure the software is invoked. This can hardly be the intended behavior.

dhoomakethu, 2022-04-22:
I agree it could be considered as bug depending on the perspective, in either way its just a one line fix in my opinion. It will be part of the new `3.0.0.rc4` or will land in `dev` branch soon. Feel free to raise a PR if you think its something that has a major side effect on the overall codebase.

dhoomakethu, 2022-04-22:
@PaulKGrimes Thanks for the PR

janiversen, 2022-04-22:
@dhoomakethu I added a suggestion along your idea, because I need that in order to be able to use it as an app on my Mac.

The big pylint round is complete, so the time is right for a 3.0.0.rc4 in my opinion.

janiversen, 2022-05-23:
on dev.

closed 2022-05-23 by janiversen