pymodbus-dev/pymodbus #723: Modbus Tcp Client Connection closing unexpectedly

sarathtv, 2022-03-24:
### Versions

* Python: 3.9.6
* OS: Windows 10
* Pymodbus: 2. 3.5
* Modbus Hardware (if used): USR-TCP232-304 

### Pymodbus Specific
* Client: tcp - sync

### Description

I am trying to communicate to an Smart Meter from Schneider Conzerv EM6436 . Initially i used the RS485 to USB converter to read data from the smart meter and i was able to do this without any problem. 
Then i wanted to try the Ethernet to RS485 convertor (USR-TCP232-304). i tried a basic code for connecting with smart meter and reading a holding register data . The client.connect() returned true but when i tried reading the register i got an exception

#### ConnectionException: Modbus Error: [Connection] ModbusTcpClient(172.17.148.200:8000): Connection unexpectedly closed 0.000000 seconds into read of 8 bytes without response from unit before it closed connection

### Code and Logs

```python
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

UNIT = 0x1

def run_sync_client():  
    client = ModbusClient('172.17.148.200', port=8000)
    client.connect()

    log.debug("Write to a holding register and read back")
    rr = client.read_holding_registers(3926, 1, unit=UNIT)
    assert(not rr.isError())    
    client.close()

if __name__ == "__main__":
    run_sync_client()
```

```python
## Log 
2022-03-24 15:51:06,473 MainThread      DEBUG    sync           :215      Connection to Modbus server established. Socket ('172.17.148.95', 4485)
2022-03-24 15:51:06,474 MainThread      DEBUG    2889022486     :15       Write to a holding register and read back
2022-03-24 15:51:06,475 MainThread      DEBUG    transaction    :139      Current transaction state - IDLE
2022-03-24 15:51:06,475 MainThread      DEBUG    transaction    :144      Running transaction 1
2022-03-24 15:51:06,476 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0xf 0x56 0x0 0x1
2022-03-24 15:51:06,476 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-03-24 15:51:06,476 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'

ConnectionException: Modbus Error: [Connection] ModbusTcpClient(172.17.148.200:8000): Connection unexpectedly closed 0.000000 seconds into read of 8 bytes without response from unit before it closed connection
```
I had checked the previous issues on the github page but i couldnt get any proper solution for this. 

Any help is appreciated. 
Thanks in advance

dhoomakethu, 2022-03-25:
Looks like remote host is closing the connection. Does this work well with other modbus clients ?

sarathtv, 2022-03-25:
Hai, @dhoomakethu 
I had used a raspberry pi3 to fetch data from the smart meter using the USB to RS485 converter . I had no problems in that set up.

dhoomakethu, 2022-03-25:
No that is Modbus RTU , what I meant was have you ever tried with any Modbus TCP client for this unit ?

dhoomakethu, 2022-03-27:
You can try the same code from a raspberry pi and see if you get the same behaviour, there are multiple stand-alone modbus clients for windows which you can try to see if the behaviour is consistent or the issue is only with pymodbus

sarathtv, 2022-03-28:
ok. thank you

janiversen, 2022-04-27:
Did you find a solution?

janiversen, 2022-05-29:
You are using the wrong framer. You are addressing a usb-rs485 adapter via TCP, but that does normally means you have to use the rtuovertcp framer.

closed 2022-05-29 by janiversen

basking-in-the-sun2000, 2023-04-17:
I'm also getting a similar error, for also a solar inverter application. The code had been running fine for almost 3 years, and for about a week I been having issues. Think I tried to upgrade the apt components a little earlier. 

- rpi 4 8 GB
- Linux raspberrypi 6.1.11-v8+ #1630 SMP PREEMPT Fri Feb 10 12:11:31 GMT 2023 aarch64 GNU/Linux
- Python 3.9.2
- pymodbus            3.2.2 (was running 3.1.1? until today)

At first I suspected the inverter, but has been flawless for the past 3 years, and hasn't received an update since.

Did thought of the framer, since it was a new addition(started with pymodbus 2), but didn't make a difference. However only tried ascii and binary framers. 

The code runs fine for a while and then starts having issues. These can last a few minutes, to most of the day. Thought it might not be clearing the connection, so I added a 5m delay between closing the modbus connection and trying again. 

Would consider a network issue, but having the connection closed in only 50µs doesn't sound as if this was enough time. Unless it got closed and pymodbus recognized the disconnect before receiving another request

```
DEBUG:pymodbus.logging:Current transaction state - IDLE
2023-04-16 17:58:02 DEBUG    Current transaction state - IDLE
DEBUG:pymodbus.logging:Running transaction 1
2023-04-16 17:58:02 DEBUG    Running transaction 1
DEBUG:pymodbus.logging:SEND: 0x3a 0x30 0x30 0x30 0x33 0x37 0x35 0x33 0x30 0x30 0x30 0x30 0x46 0x34 0x39 0xd 0xa
2023-04-16 17:58:02 DEBUG    SEND: 0x3a 0x30 0x30 0x30 0x33 0x37 0x35 0x33 0x30 0x30 0x30 0x30 0x46 0x34 0x39 0xd 0xa
DEBUG:pymodbus.logging:New Transaction state "SENDING"
2023-04-16 17:58:02 DEBUG    New Transaction state "SENDING"
DEBUG:pymodbus.logging:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-04-16 17:58:02 DEBUG    Changing transaction state from "SENDING" to "WAITING FOR REPLY"
modbus read_registers error: Modbus Error: [Connection] ModbusTcpClient(192.168.88.15:502): Connection unexpectedly closed 6.0558319091796875e-05 seconds into read of 5 bytes without response from slave before it closed connection

2023-04-16 18:13:59,826 MainThread      DEBUG    logging        :102      Current transaction state - IDLE
2023-04-16 18:13:59,826 MainThread      DEBUG    logging        :102      Running transaction 1
2023-04-16 18:13:59,827 MainThread      DEBUG    logging        :102      SEND: 0x7b 0x0 0x3 0x75 0x30 0x0 0xf 0x1e 0x1c 0x7d
2023-04-16 18:13:59,828 MainThread      DEBUG    logging        :102      New Transaction state "SENDING"
2023-04-16 18:13:59,828 MainThread      DEBUG    logging        :102      Changing transaction state from "SENDING" to "WAITING FOR REPLY"
modbus read_registers error: Modbus Error: [Connection] ModbusTcpClient(192.168.88.15:502): Connection unexpectedly closed 1.6991147994995117 seconds into read of 3 bytes without response from slave before it closed connection	
```
﻿
```
if config.inverter_type == "TCP":
    from pymodbus.client import ModbusTcpClient as ModbusClient
elif config.iinverter_type == "RTU":
    from pymodbus.client import ModbusSerialClient as ModbusClient
from pymodbus.framer.binary_framer import ModbusBinaryFramer as myFramer

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def connect_bus(ip=config.inverter_ip, PortN = config.inverter_port, timeout = 5):
    if config.inverter_type == "TCP":
        client = ModbusClient(host=ip, port=PortN, timeout=timeout, RetryOnEmpty = False, retries = 5, framer = myFramer,  )
    elif config.inverter_type == "RTU":
        client = ModbusClient(method = "rtu", port=config.connection_port, timeout=timeout, stopbits = config.connection_stopbits, 
                 bytesize = config.connection_bytesize, parity = config.connection_parity, baudrate=config.connection_baudrate)        
    else:
        raise Exception("Unknown modbus type")
    
    print(client)
    
    time.sleep(1)
    client.connect()
    time.sleep(1)
    return client

def close_bus(client):
    client.close()
    time.sleep(5)
    del client

def read_registers(client, UnitID, data):
    global ok
    try:
        nb = int(data['registers'])
        if data['method'] == "hold":
            result = client.read_holding_registers(address = int(data['addr']), count=nb, slave=UnitID)
        elif data['method'] == "input":
            result = client.read_input_registers(address =  int(data['addr']), count=nb, slave=UnitID)
 
        print(result)
 
        ns = time.time()
        while (time.time() - ns < 1.5):
            try:
                if not result.isError():
                    j = result.registers[nb - 1]
                    break
            except Exception as e:
                print("modbus read_registers error 1: %s" % str(e))
                time.sleep(0.2)
                if 'exception_code' in vars(result):
                    break
                continue
        time.sleep(0.025)
        return result
    except Exception as e:
        print("modbus read_registers error: %s" % str(e))
        ok = False
        return -1
```