pymodbus-dev/pymodbus #747: Cleanup branches to avoid confusion

janiversen, 2022-04-03:
I suggest to clean up the branches in the repo, primarily to avoid confusion but also to get an overview of which branches are still in development.

A) The following branches have been merged, and therefore can be deleted without problems:
- **dev-rebase**
- **Fix-#657**
- **home-assit-fix-#629**
- **test-fixes**
- **2.5.0**, identical to the v2.5.0 release tag,
- **revert-536-use_hexlify_packets**
- **#515-Fix-Repl-Broadcast**
- **asyncio-bug-fixes**
- **fix-docs**
- **pymodbus-2.2.0**
- **gh-pages-update**
- **dev3.7**
- **#353-Error-Reading-Registers**
- **#139-SqlDb-Context**
- **pymodbus-2.1.0**
- **sneaky-twisted**
- **pymodbus-repl**
- **#326-Use-Existing-Loop**
- **#254-Backward-Comaptibility**
- **downloas-badge**

B) The following branches have not had commits in years so they seem abandoned and thus can be deleted:
- **revert-520-dev**
- **travis-osx**
- **#357-Custom-Function**
- **gh-pages**
- **codecov-experiment**
- **#333-Infinite-Loop**
- **repl-improvements**
- **patch-307**
- **dev-1.5.0**
- **async-updates**

C) These branches are being changed as part of the new setup, and should be deleted:
- **3.0.0**, the content is in **dev**, and once released it should be a tag
- **dev3.x", the content is in **dev**, and it is confusing as to where development takes place

D) These branches seems to have commits not yet in dev, and could be active ?
- **dev2x**, no commits last 8 month

E) These branches are active development/release branches and should not be touched:
- **dev**, development branch (new default)
- **3.0.0-black** used to find black problems
- **master**, release branch (push should be prohibited, unless when making a release)
- **2.5.3**, identical to the v2.5.3 release tag, good to have if we one day need to make 2.5.4

When we have an agreement I volunteer to do the cleanup (delete A,B and C)

I will update the list according to the comments received.

janiversen, 2022-04-03:
@dhoomakethu this is one where I really need your input and opinion.

dhoomakethu, 2022-04-05:
@janiversen 
* `A`  can be deleted straight forward
*  From `B` `gh-pages` was intendend at creating github page @ pymodbus.github.io , I need to check if its really required.
*  Also from `B` please hold on with deleting `repl-improvements` , I will double confirm if there are any pending changes in there which needs to be in `dev`
* `3.0.0` branch needs to be available for some more time as we have 3 releases made out of it.
*  `dev3.x` cane be deleted as I agree it causes confusion and so is `dev2x`

Thanks for taking this up. Appreciate.

janiversen, 2022-04-05:
Done now we are down to:
```
  remotes/upstream/2.5.3
  remotes/upstream/3.0.0 (needs to go before releasing 3.0.0)
  remotes/upstream/3.0.0-black
  remotes/upstream/dev
  remotes/upstream/gh-pages (awaiting judgement)
  remotes/upstream/master
  remotes/upstream/repl-improvements (awaiting judgement)
```

closed 2022-04-13 by janiversen