pymodbus-dev/pymodbus #776: TLS Modbus Server crash when unit id is different than 0

tomas-bozek, 2022-04-13:
Python: 3.8
OS: Windows
Pymodbus: 2.5.3

Example:

```py3
loop = asyncio.get_event_loop()
from pymodbus.datastore import ModbusServerContext, ModbusSlaveContext, ModbusSequentialDataBlock
from pymodbus.server.async_io import ModbusConnectedRequestHandler, ModbusTlsServer
from pymodbus.transaction import ModbusTlsFramer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.client.asynchronous.async_io import ReconnectingAsyncioModbusTlsClient, ModbusClientProtocol
from pymodbus.factory import ClientDecoder
from ssl import create_default_context, Options
# ---
slave_context = ModbusSlaveContext(hr=ModbusSequentialDataBlock(address=30001,
                                                                            values=[152, 276]),
                                               zero_mode=True)
context = ModbusServerContext(slaves={2: slave_context}, single=False)
identity = ModbusDeviceIdentification()
identity.VendorName = 'Example'
identity.VendorUrl = 'http://www.example.com'
identity.ProductCode = 'MdbTcpSrv'
identity.ProductName = 'Modbus TCP Server'
identity.ModelName = 'Modbus TCP Server'
identity.MajorMinorRevision = '1.0.0'
identity.UserApplicationName = 'Exm'
# ---
host = "0.0.0.0"
port = 802
cert_file = "path to cert file"
key_file = "path to key file"
key_file_password = None
# ssl context
ssl_context = create_default_context()
if cert_file or key_file:
    ssl_context.load_cert_chain(certfile=cert_file, keyfile=key_file, password=key_file_password)
# According to MODBUS/TCP Security Protocol Specification, it is TLSv2 at least
ssl_context.options |= Options.OP_NO_TLSv1_1
ssl_context.options |= Options.OP_NO_TLSv1
ssl_context.options |= Options.OP_NO_SSLv3
ssl_context.options |= Options.OP_NO_SSLv2
# ---
modbus_server = ModbusTlsServer(context=context,
                                framer=ModbusTlsFramer,
                                identity=identity,
                                address=(host, port),
                                sslctx=ssl_context,
                                handler=ModbusConnectedRequestHandler,
                                cert_file=cert_file,
                                key_file=key_file,
                                defer_start=False,
                                loop=loop,
                                ignore_missing_slaves=False,
                                broadcast_enable=False,
                                allow_reuse_address=False,
                                allow_reuse_port=False)
modbus_server.server = await modbus_server.server_factory

# --- CLIENT
host = "127.0.0.1"
modbus_client = ReconnectingAsyncioModbusTlsClient(protocol_class=ModbusClientProtocol,
                                                   loop=loop,
                                                   framer=ModbusTlsFramer(ClientDecoder()))

# ssl context
ssl_context = create_default_context(cafile=cert_file)
# According to MODBUS/TCP Security Protocol Specification, it is TLSv2 at least
ssl_context.options |= Options.OP_NO_TLSv1_1
ssl_context.options |= Options.OP_NO_TLSv1
ssl_context.options |= Options.OP_NO_SSLv3
ssl_context.options |= Options.OP_NO_SSLv2
# ---
await modbus_client.start(host=host,
                          port=port,
                          sslctx=ssl_context)
# ---
result = await modbus_client.protocol.read_holding_registers(address=30001, count=1, unit=2)
# ---
modbus_client.stop()
modbus_server.server_close()
```

When unit == 0 then everything is OK, when is something else (for example 2) then is problem

An Error is in:
- ModbusTlsFramer does not know anything about `unit`, class does not encode/decode nothing only function_code 
- ModbusFramer in `_validate_unit_id` checks `self._header` for `uid`, but ModbusTlsFramer  header is empty dictionary, therefore crash and  raised `pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request`

janiversen, 2022-04-13:
please attach the backtrace so we can see where it breaks.

tomas-bozek, 2022-04-13:
exception

```text
2022-04-13 12:29:41,381 ERROR    pymodbus.server.async_io Unknown exception ''uid'' on stream ('127.0.0.1', 50850) forcing disconnect
2022-04-13 12:29:41,381 INFO     pymodbus.client.asynchronous.async_io Protocol lost connection.
2022-04-13 12:29:41,381 ERROR    Test            Modbus Error: [Connection] Connection lost during request
Traceback (most recent call last):
  File "C:\Bender\runtime\tests\tests_30_pymodbus\runners\test_runner.py", line 103, in runner
    result = await modbus_client.protocol.read_holding_registers(address=30001, count=1, unit=2)
  File "C:\_VENV\lib\site-packages\pymodbus\client\asynchronous\async_io\__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "C:\Program Files\Python38\lib\asyncio\tasks.py", line 494, in wait_for
    return fut.result()
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
```

janiversen, 2022-04-13:
This is a timeout, are you sure your devices responds to the request?

tomas-bozek, 2022-04-13:
problem is in `ERROR    pymodbus.server.async_io Unknown exception ''uid'' on stream ('127.0.0.1', 50850) forcing disconnect`

like i wrote before 
ModbusFramer in _validate_unit_id checks self._header for uid, but ModbusTlsFramer header is empty dictionary, it crashs and connection is lost.

When i change unit on server side and client side to value == 0, everything is ok, because `_validate_unit_id` return always True when unit is 0.

```py3
    def _validate_unit_id(self, units, single):
        """
        Validates if the received data is valid for the client
        :param units: list of unit id for which the transaction is valid
        :param single: Set to true to treat this as a single context
        :return:         """

        if single:
            return True
        else:
            if 0 in units or 0xFF in units:
                # Handle Modbus TCP unit identifier (0x00 0r 0xFF)
                # in asynchronous requests
                return True
            return self._header['uid'] in units
```

tomas-bozek, 2022-04-13:
I modified `pymodbus.server.async_io` row 184

from 

```py3
                    _logger.error("Unknown exception '{}' on stream {} "
                                  "forcing disconnect".format(e, client_addr))
```

to

```py3
                    _logger.error("Unknown exception '{}' on stream {} "
                                  "forcing disconnect".format(e, client_addr), exc_info=e)
```

and tracebacks

```text
2022-04-13 12:54:18,442 ERROR    pymodbus.server.async_io Unknown exception ''uid'' on stream ('127.0.0.1', 50966) forcing disconnect
Traceback (most recent call last):
  File "C:\_VENV\lib\site-packages\pymodbus\server\async_io.py", line 171, in handle
    self.framer.processIncomingPacket(
  File "C:\_VENV\lib\site-packages\pymodbus\framer\tls_framer.py", line 133, in processIncomingPacket
    if self._validate_unit_id(unit, single):
  File "C:\_VENV\lib\site-packages\pymodbus\framer\__init__.py", line 33, in _validate_unit_id
    return self._header['uid'] in units
KeyError: 'uid'
2022-04-13 12:54:18,459 INFO     pymodbus.client.asynchronous.async_io Protocol lost connection.
2022-04-13 12:54:18,459 ERROR    Test            Modbus Error: [Connection] Connection lost during request
Traceback (most recent call last):
  File "C:\Bender\runtime\tests\tests_30_pymodbus\runners\test_runner.py", line 103, in runner
    result = await modbus_client.protocol.read_holding_registers(address=30001, count=1, unit=2)
  File "C:\_VENV\lib\site-packages\pymodbus\client\asynchronous\async_io\__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "C:\Program Files\Python38\lib\asyncio\tasks.py", line 494, in wait_for
    return fut.result()
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
```

closed 2022-05-20 by github-actions[bot]

ZHI-ANG, 2022-08-05:
I also got stuck on this problem and managed to troubleshoot. The problem comes from the default framer used by ModbusTlsServer and ModbusTlsClient, ModbusTlsFramer, which doesn't contain the uid of a slave:
`
    """ Modbus TLS Frame controller

    No prefix MBAP header before decrypted PDU is used as a message frame for
    Modbus Security Application Protocol.  It allows us to easily separate
    decrypted messages which is PDU as follows:

        [ Function Code] [ Data ]
          1b               Nb
    """
`
So instead of using the default framer, we should assign a ModbusSocketFramer in both client and server, like this:
`
# in client.py
client = ModbusTlsClient('127.0.0.1', port=8020, sslctx=sslctx,framer=ModbusSocketFramer)
# in server.py
    StartTlsServer(context, identity=identity, certfile="server-cert.pem",
                    keyfile="server-key.pem",framer=ModbusSocketFramer, address=("0.0.0.0", 8020))
`
Hope it is helpful.

janiversen, 2022-08-05:
If you use ModbusSocketFramer the connection runs as if it is a normal tcp connection, which means that there is no encryption which is the whole purpose of tls.