pymodbus-dev/pymodbus #778: AttributeError when using uvloop

tomas-bozek, 2022-04-13:
Linux: Ubuntu 20.04
Python: 3.8
Pymodbus: 2.5.3

Exception when using `uvloop`

```
2022-04-13 13:16:13,035 ERROR    pymodbus.server.async_io Datastore unable to fulfill request: 'uvloop.loop.UDPTransport' object has no attribute '_sock'; Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/pymodbus/server/async_io.py", line 109, in connection_lost
    self._log_exception()
  File "/usr/local/lib/python3.8/dist-packages/pymodbus/server/async_io.py", line 61, in _log_exception
    sock_name = self.protocol._sock.getsockname()
AttributeError: 'uvloop.loop.UDPTransport' object has no attribute '_sock'
```

dhoomakethu, 2022-04-13:
`uvloop` is not supported , use `asyncio` loop instead.

tracernz, 2022-04-14:
Can <BaseTransport>.get_extra_info('socket') not be used instead of accessing the private _sock member? It appears to only be used to get the socket name.

https://docs.python.org/3/library/asyncio-protocol.html#asyncio.BaseTransport.get_extra_info

janiversen, 2022-04-14:
try it, patches are welcome.

janiversen, 2022-04-21:
Closing as non-supported.

closed 2022-04-21 by janiversen