pymodbus-dev/pymodbus #860: PyModbus server client payload

aasick, 2022-04-28:
Hello Team,

I am trying to implement server and client modbus simulator with pymodbus using virtual serial connector (COM1 <-> COM2).
I have two python codes one is for modbus server and another one for client. I have attached both codes below 
-----------------------------------server.py-------------------------------------------------------------------------------------
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
import logging
# --------------------------------------------------------------------------- #
# import the payload builder
# --------------------------------------------------------------------------- #

from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadBuilder
from pymodbus.transaction import ModbusRtuFramer
from pymodbus.server.sync import StartSerialServer

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():
    builder = BinaryPayloadBuilder(byteorder=Endian.Little,
                                   wordorder=Endian.Little)

    builder.add_8bit_int(-0x12)
    builder.add_8bit_uint(0x12)

    block = ModbusSequentialDataBlock(1, builder.to_registers())
    store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
    context = ModbusServerContext(slaves=store, single=True)

    StartSerialServer(context, framer=ModbusRtuFramer, port='COM2', stopbits=1, bytesize=8, parity='E', baudrate=9600, timeout=1)

if __name__ == "__main__":
    run_server()
-----------------------------------client.py-------------------------------------------------------------------------------------
from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.client.sync import ModbusSerialClient
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_client():
    client = ModbusSerialClient("rtu", port='COM1', stopbits=1, bytesize=8, parity='E', baudrate=9600, timeout=1)
    client.connect()

    data = client.read_holding_registers(address=0x01, count=10, unit=1)

    print(data)

if __name__ == "__main__":
    run_client()
--------------------------------------------------------------------------------------------------------------------------------------------

**when I am running server code I am getting below error on server side**
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x0 0x1 0x0 0xa
DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.datastore.context:validate: fc-[3] address-2: count-10
ERROR:pymodbus.pdu:Exception Response(131, 3, IllegalAddress)
DEBUG:pymodbus.server.sync:send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'

**when I am running client ode I am getting below error on clientside**
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0x1 0x0 0xa 0x94 0xd
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x83 0x2 0xc0 0xf1
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x83 0x2
DEBUG:pymodbus.factory:Factory Response[131]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Exception Response(131, 3, IllegalAddress)

aasick, 2022-04-28:
@[dhoomakethu](https://github.com/dhoomakethu) can you take a look

dhoomakethu, 2022-04-28:
what version of pymodbus is this ?

aasick, 2022-04-28:
following version available in my system 
\lib\site-packages (from pymodbus) (1.16.0)
\lib\site-packages (from pymodbus) (3.5)

swamper123, 2022-04-28:
1.16.0 would be a quite old version of Modbus (2.5.3 is the latest and 3.0 is incoming).
3.5 hasn't been released ... do you got the right numbers?

swamper123, 2022-04-28:
Just took a deeper look in the release history...there is neither a 1.16.0

aasick, 2022-04-28:
I have 2.5.3 on my system

aasick, 2022-04-28:
is my code looking good? server and client?

swamper123, 2022-04-28:
Haven't worked with the rtu and server part so far.
My guttfeeling tells me, that `Exception Response(131, 3, IllegalAddress)`  says you tried to read sth. on an address that is not correct.

janiversen, 2022-04-28:
Your feeling is correct, the server tells that the client requested a non existing address.

aasick, 2022-04-28:
yeah, address is main issue.
my server I started with StartSerialServer() method call. following is server code

builder = BinaryPayloadBuilder(byteorder=Endian.Little,
                                   wordorder=Endian.Little)

    builder.add_8bit_int(-0x12)
    builder.add_8bit_uint(0x12)

    block = ModbusSequentialDataBlock(1, builder.to_registers())
    store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
    context = ModbusServerContext(slaves=store, single=True)

    StartSerialServer(context, framer=ModbusRtuFramer, port='COM2', stopbits=1, bytesize=8, parity='E', baudrate=9600, timeout=1)

swamper123, 2022-04-28:
So what are your registers on the Server?
And what are you trying to read in your command?
Explain it preciscely (and maybe I got an idea more ^^).

aasick, 2022-04-28:
can you help me, how to mention register before starting server with StartSerialServer() method

aasick, 2022-04-28:
I haven't find right example for starting rtu based server with StartSerialServer()

dhoomakethu, 2022-04-28:
Try this example

```
#!/usr/bin/env python3
""" Pymodbus Server Payload Example
--------------------------------------------------------------------------

If you want to initialize a server context with a complicated memory
layout, you can actually use the payload builder.
"""
import logging

# --------------------------------------------------------------------------- #
# import the various server implementations
# --------------------------------------------------------------------------- #
from pymodbus.version import version
from pymodbus.server.sync import StartTcpServer, StartSerialServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadBuilder
from pymodbus.framer.rtu_framer import ModbusRtuFramer

# --------------------------------------------------------------------------- #
# configure the service logging
# --------------------------------------------------------------------------- #
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT) #NOSONAR
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_payload_server():
    """ Run payload server. """
    # ----------------------------------------------------------------------- #
    # build your payload
    # ----------------------------------------------------------------------- #
    builder = BinaryPayloadBuilder(byteorder=Endian.Little,
                                   wordorder=Endian.Little)
    builder.add_string('abcdefgh')
    builder.add_bits([0, 1, 0, 1, 1, 0, 1, 0])
    builder.add_8bit_int(-0x12)
    builder.add_8bit_uint(0x12)
    builder.add_16bit_int(-0x5678)
    builder.add_16bit_uint(0x1234)
    builder.add_32bit_int(-0x1234)
    builder.add_32bit_uint(0x12345678)
    builder.add_16bit_float(12.34)
    builder.add_16bit_float(-12.34)
    builder.add_32bit_float(22.34)
    builder.add_32bit_float(-22.34)
    builder.add_64bit_int(-0xDEADBEEF)
    builder.add_64bit_uint(0x12345678DEADBEEF)
    builder.add_64bit_uint(0xDEADBEEFDEADBEED)
    builder.add_64bit_float(123.45)
    builder.add_64bit_float(-123.45)

    # ----------------------------------------------------------------------- #
    # use that payload in the data store
    # ----------------------------------------------------------------------- #
    # Here we use the same reference block for each underlying store.
    # ----------------------------------------------------------------------- #

    block = ModbusSequentialDataBlock(1, builder.to_registers())
    store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
    context = ModbusServerContext(slaves=store, single=True)

    # ----------------------------------------------------------------------- #
    # initialize the server information
    # ----------------------------------------------------------------------- #
    # If you don't set this or any fields, they are defaulted to empty strings.
    # ----------------------------------------------------------------------- #
    identity = ModbusDeviceIdentification(info_name= {
        'VendorName': 'Pymodbus',
        'ProductCode': 'PM',
        'VendorUrl': 'http://github.com/riptideio/pymodbus/', #NOSONAR
        'ProductName': 'Pymodbus Server',
        'ModelName': 'Pymodbus Server',
        'MajorMinorRevision': version.short(),
    })
    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #
    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity,
                      port='/tmp/ttyp0', timeout=.005, baudrate=9600)

if __name__ == "__main__":
    run_payload_server()
```

dhoomakethu, 2022-04-28:
change `port` and other `serial` properties appropriately

dhoomakethu, 2022-04-28:
Logs
```
2022-04-28 12:52:43,444 MainThread      DEBUG    payload        :121      [24930, 25444, 25958, 26472, 23278, 4744, 43316, 4812, 60927, 65400, 22068, 4652, 18988, 51794, 47282, 16722, 47282, 49425, 16722, 8703, 65535, 65519, 48813, 56952, 22068, 4845, 48813, 57071, 48813, 57037, 52428, 52428, 56414, 16589, 52428, 52428, 56414, 49152]
2022-04-28 12:52:43,450 MainThread      DEBUG    sync           :45       Client Connected [('/tmp/ttyp0', '/tmp/ttyp0')]
2022-04-28 12:52:43,450 MainThread      DEBUG    sync           :580      Started thread to serve client
2022-04-28 12:52:48,086 MainThread      DEBUG    rtu_framer     :181      Getting Frame - 0x3 0x0 0x0 0x0 0x1
2022-04-28 12:52:48,086 MainThread      DEBUG    factory        :215      Factory Request[ReadHoldingRegistersRequest: 3]
2022-04-28 12:52:48,086 MainThread      DEBUG    rtu_framer     :105      Frame advanced, resetting header!!
2022-04-28 12:52:48,086 MainThread      DEBUG    context        :64       validate: fc-[3] address-1: count-1
2022-04-28 12:52:48,086 MainThread      DEBUG    context        :78       getValues: fc-[3] address-1: count-1
2022-04-28 12:52:48,086 MainThread      DEBUG    sync           :143      send: [ReadHoldingRegistersResponse (1)]- b'0103026162103d'
2022-04-28 12:52:49,241 MainThread      DEBUG    rtu_framer     :181      Getting Frame - 0x3 0x0 0x0 0x0 0x1
2022-04-28 12:52:49,242 MainThread      DEBUG    factory        :215      Factory Request[ReadHoldingRegistersRequest: 3]
2022-04-28 12:52:49,242 MainThread      DEBUG    rtu_framer     :105      Frame advanced, resetting header!!
2022-04-28 12:52:49,242 MainThread      DEBUG    context        :64       validate: fc-[3] address-1: count-1
2022-04-28 12:52:49,242 MainThread      DEBUG    context        :78       getValues: fc-[3] address-1: count-1
2022-04-28 12:52:49,242 MainThread      DEBUG    sync           :143      send: [ReadHoldingRegistersResponse (1)]- b'0103026162103d'
2022-04-28 12:52:50,235 MainThread      DEBUG    rtu_framer     :181      Getting Frame - 0x3 0x0 0x0 0x0 0x1
2022-04-28 12:52:50,235 MainThread      DEBUG    factory        :215      Factory Request[ReadHoldingRegistersRequest: 3]
2022-04-28 12:52:50,235 MainThread      DEBUG    rtu_framer     :105      Frame advanced, resetting header!!
2022-04-28 12:52:50,235 MainThread      DEBUG    context        :64       validate: fc-[3] address-1: count-1
2022-04-28 12:52:50,235 MainThread      DEBUG    context        :78       getValues: fc-[3] address-1: count-1
2022-04-28 12:52:50,235 MainThread      DEBUG    sync           :143      send: [ReadHoldingRegistersResponse (1)]- b'0103026162103d'
```

aasick, 2022-04-28:
yeah getting same output logs

2022-04-28 13:20:19,057 MainThread      DEBUG    payload        :125      [24930, 25444, 25958, 26472, 23278, 4744, 43316, 4812, 60927, 65400, 22068, 4652, 18988, 51794, 47282, 16722, 47282, 49425, 16722, 8703, 65535, 65519, 48813, 56952
, 22068, 4845, 48813, 57071, 48813, 57037, 52428, 52428, 56414, 16589, 52428, 52428, 56414, 49152]
2022-04-28 13:20:19,057 MainThread      DEBUG    sync           :46       Client Connected [COM2:COM2]
2022-04-28 13:20:19,057 MainThread      DEBUG    sync           :581      Started thread to serve client
2022-04-28 13:20:27,136 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x0 0x1 0x0 0xa
2022-04-28 13:20:27,136 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-04-28 13:20:27,136 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-04-28 13:20:27,136 MainThread      DEBUG    context        :63       validate: fc-[3] address-2: count-10
2022-04-28 13:20:27,136 MainThread      DEBUG    context        :77       getValues fc-[3] address-2: count-10
2022-04-28 13:20:27,136 MainThread      DEBUG    sync           :144      send: [ReadHoldingRegistersResponse (10)]- b'0103146364656667685aee1288a93412ccedffff785634fe15'
2022-04-28 13:21:53,752 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x0 0x1 0x0 0xa
2022-04-28 13:21:53,753 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-04-28 13:21:53,753 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-04-28 13:21:53,753 MainThread      DEBUG    context        :63       validate: fc-[3] address-2: count-10
2022-04-28 13:21:53,753 MainThread      DEBUG    context        :77       getValues fc-[3] address-2: count-10
2022-04-28 13:21:53,753 MainThread      DEBUG    sync           :144      send: [ReadHoldingRegistersResponse (10)]- b'0103146364656667685aee1288a93412ccedffff785634fe15'

aasick, 2022-04-28:
In this case how our client should be....? is below code is right ? because from our server side we not mention any address right?

`    client = ModbusSerialClient("rtu", port='COM1', stopbits=1, bytesize=8, parity='E', baudrate=9600, timeout=1)
    client.connect()

    data = client.read_holding_registers(address=0x01, count=10, unit=1)

    print(data)`

aasick, 2022-04-28:
I need to send following hex data as payload, what was the right method to use -> 01 2B 0E 01 01 00 00 03 00 0A 53 69 65 6D 65 6E 73 20 41 47 01 07 50 41 43 32 32 30 30

dhoomakethu, 2022-05-31:
Slave will have an address defined, please refer to the manuals if you are using a real modbus slave , in case using modbus server, the option can be passed as part of server initialization.