pymodbus-dev/pymodbus #873: ValueError: filedescriptor out of range in select()

gieljnssns, 2022-05-02:
### Versions

* Python: 3.9
* OS: HAOS
* Pymodbus: 2.5.3 and 3.0.0dev3
* Modbus Hardware (if used):  SolarEdge SE3000

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

I'm using a custom component for Home Assistant and I'm alway having this error after ±25min

Original issue https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/24
Similar issue https://github.com/riptideio/pymodbus/issues/721

### Code and Logs

```python
Logger: homeassistant
Source: custom_components/solaredge_modbus/__init__.py:187
Integration: SolarEdge Modbus Multi-Inverter ([documentation](https://github.com/WillCodeForCats/solaredge-modbus-multi), [issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues))
First occurred: 12:51:25 (188 occurrences)
Last logged: 15:58:26

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 156, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 197, in read_modbus_data
    self.read_modbus_data_inverters()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 207, in read_modbus_data_inverters
    inverter_data = self.read_holding_registers(
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 109, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 174, in execute
    response, last_exception = self._transact(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 294, in _transact
    result = self._recv(response_length, full)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 326, in _recv
    read_min = self.client.framer.recvPacket(min_size)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/framer/__init__.py", line 49, in recvPacket
    return self.client.recv(size)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 89, in recv
    def recv(self, size):
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 293, in _recv
    while recv_size > 0:
ValueError: filedescriptor out of range in select()

```

janiversen, 2022-05-02:
That is solved in #722 and merged into dev, however it have not been released yet.

If you want a fast solution simply copy/paste the change.

gieljnssns, 2022-05-02:
I use 
```json
{
  "domain": "solaredge_modbus",
  "name": "SolarEdge Modbus Multi-Inverter",
  "documentation": "https://github.com/WillCodeForCats/solaredge-modbus-multi",
  "issue_tracker": "https://github.com/WillCodeForCats/solaredge-modbus-multi/issues",
  "requirements": ["pymodbus==3.0.0.dev3"],
  "dependencies": [],
  "codeowners": ["@WillCodeForCats"],
  "config_flow": true,
  "iot_class": "local_polling",
  "version": "1.1.2"
}
```
as a manifest file

janiversen, 2022-05-02:
ok, but it is not in dev3, it will be in dev4.

Simply control if sync.py contained the patch I pointed you to.

janiversen, 2022-05-02:
Seems so, please try to change sync.py line 295 from:
```
except ValueError:
```
to
```
except:
```
Seems you are getting another exception that expected.

gieljnssns, 2022-05-02:
I was able to change the file with vi, I will report back...

gieljnssns, 2022-05-02:
```
Deze fout is ontstaan door een aangepaste integratie.

Logger: homeassistant
Source: custom_components/solaredge_modbus/__init__.py:187
Integration: SolarEdge Modbus Multi-Inverter (documentation, issues)
First occurred: 17:35:58 (2 occurrences)
Last logged: 17:36:58

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 295, in _recv
    ready = select.select([self.socket], [], [], end - time_)
ValueError: filedescriptor out of range in select()

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 156, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 197, in read_modbus_data
    self.read_modbus_data_inverters()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 207, in read_modbus_data_inverters
    inverter_data = self.read_holding_registers(
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 110, in execute
    return self.transaction.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 174, in execute
    response, last_exception = self._transact(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 294, in _transact
    result = self._recv(response_length, full)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 326, in _recv
    read_min = self.client.framer.recvPacket(min_size)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/framer/__init__.py", line 49, in recvPacket
    return self.client.recv(size)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 90, in recv
    return self._recv(size)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 297, in _recv
    return self._handle_abrupt_socket_close(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 347, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000024 seconds into read of 8 bytes without response from unit before it closed connection

```

janiversen, 2022-05-02:
This is a wanted exception, because it informs the application that a connection was broken.

gieljnssns, 2022-05-05:
[image]
[image]
[image]

gieljnssns, 2022-05-05:
pymodbus debug logs
```
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.79.54', 50909)
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Running transaction 23
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x0 0x17 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x6c
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] RECV: 0x0 0x17 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x46 0xcd 0x46 0xcd 0xff 0xff 0xff 0xff 0xff 0xfc 0x9 0x1a 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x10 0x54 0xff 0xff 0xc3 0x6a 0xff 0xfd 0x10 0x76 0xff 0xff 0x15 0x18 0xff 0xfe 0xd9 0x44 0xff 0xfe 0x1 0x22 0x74 0x72 0x0 0x0 0x2f 0x27 0xff 0xfc 0xd 0xbb 0xff 0xff 0x10 0x93 0xff 0xff 0x80 0x0 0x7 0x13 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x17 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x46 0xcd 0x46 0xcd 0xff 0xff 0xff 0xff 0xff 0xfc 0x9 0x1a 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x10 0x54 0xff 0xff 0xc3 0x6a 0xff 0xfd 0x10 0x76 0xff 0xff 0x15 0x18 0xff 0xfe 0xd9 0x44 0xff 0xfe 0x1 0x22 0x74 0x72 0x0 0x0 0x2f 0x27 0xff 0xfc 0xd 0xbb 0xff 0xff 0x10 0x93 0xff 0xff 0x80 0x0 0x7 0x13 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 23
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 23
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.payload] [21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 13104, 12336, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12339, 11826, 13619, 14080, 0, 0, 0, 14150, 12597, 12341, 13622, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 101, 50, 18125, 18125, 65535, 65535, 65532, 2330, 65535, 65535, 65535, 65535, 65535, 65535, 4180, 65535, 50026, 65533, 4214, 65535, 5400, 65534, 55620, 65534, 290, 29810, 0, 12071, 65532, 3515, 65535, 4243, 65535, 32768, 1811, 32768, 32768, 65534, 4, 0, 65535, 65535, 65535]
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.payload] [b'\x01"', b'tr']
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x69 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 105
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x69 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x1 0xf2 0xf 0xae 0xf 0xb1 0xf 0xb8 0x9 0x13 0x9 0xa 0x9 0x13 0x0 0x0 0x2 0xd0 0x0 0x0 0x2 0xc9 0x0 0x0 0x2 0xe3 0x0 0x0 0x1 0xe1 0x0 0x0 0x1 0xe0 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x8b 0x25 0xa2 0x1 0x7c 0xb 0xb8 0x2 0x0 0x0 0x0 0x62 0x67 0x98 0x48 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0xe9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0x52 0x68 0x0 0x9 0xdc 0x42 0x62 0x67 0xb6 0x35 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x25
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x69 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x1 0xf2 0xf 0xae 0xf 0xb1 0xf 0xb8 0x9 0x13 0x9 0xa 0x9 0x13 0x0 0x0 0x2 0xd0 0x0 0x0 0x2 0xc9 0x0 0x0 0x2 0xe3 0x0 0x0 0x1 0xe1 0x0 0x0 0x1 0xe0 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x8b 0x25 0xa2 0x1 0x7c 0xb 0xb8 0x2 0x0 0x0 0x0 0x62 0x67 0x98 0x48 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0xe9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0x52 0x68 0x0 0x9 0xdc 0x42 0x62 0x67 0xb6 0x35 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x25
[... cut ...]
2022-04-26 09:08:53 DEBUG (MainThread) [pymodbus.transaction] Running transaction 25
2022-04-26 09:08:53 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x0 0x19 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x6c
2022-04-26 09:08:53 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-04-26 09:08:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-04-26 09:08:53 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading inverter at id 1: filedescriptor out of range in select()
```

janiversen, 2022-05-05:
HA uses pymodbus 2.5.3, so I suppose you miss the patch.

gieljnssns, 2022-05-08:
I use the dev version
[image]

I have some debug logs
```
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.79.54', 55027)
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - IDLE
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Running transaction 1
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x6c
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] RECV: 0x0 0x1 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x27 0xf2 0x27 0xf2 0xff 0xff 0xff 0xff 0xff 0xfd 0x9 0x71 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x60 0x40 0xff 0xff 0xc3 0x51 0xff 0xfd 0x60 0x6d 0xff 0xff 0x3a 0x34 0xff 0xfe 0xd9 0x3 0xff 0xfe 0x1 0x25 0x63 0xfc 0x0 0x0 0x1a 0xd6 0xff 0xfd 0xe 0x39 0xff 0xff 0x61 0xb7 0xff 0xff 0x80 0x0 0x9 0x62 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x27 0xf2 0x27 0xf2 0xff 0xff 0xff 0xff 0xff 0xfd 0x9 0x71 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x60 0x40 0xff 0xff 0xc3 0x51 0xff 0xfd 0x60 0x6d 0xff 0xff 0x3a 0x34 0xff 0xfe 0xd9 0x3 0xff 0xfe 0x1 0x25 0x63 0xfc 0x0 0x0 0x1a 0xd6 0xff 0xfd 0xe 0x39 0xff 0xff 0x61 0xb7 0xff 0xff 0x80 0x0 0x9 0x62 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.payload] [21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 13104, 12336, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12339, 11826, 13619, 14080, 0, 0, 0, 14150, 12597, 12341, 13622, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 101, 50, 10226, 10226, 65535, 65535, 65533, 2417, 65535, 65535, 65535, 65535, 65535, 65535, 24640, 65535, 50001, 65533, 24685, 65535, 14900, 65534, 55555, 65534, 293, 25596, 0, 6870, 65533, 3641, 65535, 25015, 65535, 32768, 2402, 32768, 32768, 65534, 4, 0, 65535, 65535, 65535]
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.payload] [b'\x01%', b'c\xfc']
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.79.54', 52049)
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Running transaction 2
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x6c
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] RECV: 0x0 0x2 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0 [... cut ...]
2022-05-08 10:47:49 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-05-08 10:47:49 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-05-08 10:47:49 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000026 seconds into read of 8 bytes without response from unit before it closed connection
```

dhoomakethu, 2022-09-08:
@AnonHax0r can you give us an approximate count of clients you are trying to handle ? `fd`'s are OS specific and if it exhausts there is nothing a library can do but throw errors. But if you think pymodbus is not releasing the handles and draining the fd's thats something we will have to debug.

AnonHax0r, 2022-09-08:
@dhoomakethu I am running around 600 devices or more, and growing. can only run so many with a select call, upgrading the client to use a poll call prevents this error and maintains much better performance. I can share the code snipped I used to make this possible.

AnonHax0r, 2022-09-08:
`from asyncio import events
import socket
import selectors
import serial
import time
import ssl
import sys
from functools import partial
from pymodbus.constants import Defaults
from pymodbus.utilities import hexlify_packets, ModbusTransactionState
from pymodbus.factory import ClientDecoder
from pymodbus.exceptions import NotImplementedException, ParameterException
from pymodbus.exceptions import ConnectionException
from pymodbus.transaction import FifoTransactionManager
from pymodbus.transaction import DictTransactionManager
from pymodbus.transaction import ModbusSocketFramer, ModbusBinaryFramer
from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
from pymodbus.transaction import ModbusTlsFramer
from pymodbus.client.common import ModbusClientMixin

# --------------------------------------------------------------------------- #
# Logging
# --------------------------------------------------------------------------- #
import logging
_logger = logging.getLogger(__name__)

# --------------------------------------------------------------------------- #
# The Synchronous Clients
# --------------------------------------------------------------------------- #
EVENT_READ = (1 << 0)
EVENT_WRITE = (1 << 1)

class BaseModbusClient(ModbusClientMixin):
    """
    Inteface for a modbus synchronous client. Defined here are all the
    methods for performing the related request methods.  Derived classes
    simply need to implement the transport methods and set the correct
    framer.
    """

    def __init__(self, framer, **kwargs):
        """ Initialize a client instance

        :param framer: The modbus framer implementation to use
        """
        self.framer = framer
        self.transaction = DictTransactionManager(self, **kwargs)
        self._debug = False
        self._debugfd = None
        self.selector = selectors.PollSelector()
        self.broadcast_enable = kwargs.get(
            'broadcast_enable', Defaults.broadcast_enable)

    # ----------------------------------------------------------------------- #
    # Client interface
    # ----------------------------------------------------------------------- #
    def connect(self):
        """ Connect to the modbus remote host

        :returns: True if connection succeeded, False otherwise
        """
        raise NotImplementedException(
            "Method not implemented by derived class")

    def close(self):
        """ Closes the underlying socket connection
        """
        pass

    def is_socket_open(self):
[... 700 lines cut ...]
            else:
                return self.socket.isOpen()
        return False

    def __str__(self):
        """ Builds a string representation of the connection

        :returns: The string representation
        """
        return "ModbusSerialClient(%s baud[%s])" % (self.method, self.baudrate)

    def __repr__(self):
        return (
            "<{} at {} socket={self.socket}, method={self.method}, "
            "timeout={self.timeout}>"
        ).format(self.__class__.__name__, hex(id(self)), self=self)

# --------------------------------------------------------------------------- #
# Exported symbols
# --------------------------------------------------------------------------- #

__all__ = [
    "ModbusTcpClient", "ModbusTlsClient", "ModbusUdpClient", "ModbusSerialClient"
]
`

AnonHax0r, 2022-09-08:
`class BaseModbusClient(ModbusClientMixin):
    """
    Inteface for a modbus synchronous client. Defined here are all the
    methods for performing the related request methods.  Derived classes
    simply need to implement the transport methods and set the correct
    framer.
    """

    def __init__(self, framer, **kwargs):
        """ Initialize a client instance

        :param framer: The modbus framer implementation to use
        """
        self.framer = framer
        self.transaction = DictTransactionManager(self, **kwargs)
        self._debug = False
        self._debugfd = None
        self.selector = selectors.PollSelector()
        self.broadcast_enable = kwargs.get(
            'broadcast_enable', Defaults.broadcast_enable)

    # ----------------------------------------------------------------------- #
    # Client interface
    # ----------------------------------------------------------------------- #
    def connect(self):
        """ Connect to the modbus remote host

        :returns: True if connection succeeded, False otherwise
        """
        raise NotImplementedException(
            "Method not implemented by derived class")

    def close(self):
        """ Closes the underlying socket connection
        """
        pass

    def is_socket_open(self):
        """
        Check whether the underlying socket/serial is open or not.

        :returns: True if socket/serial is open, False otherwise
        """
        raise NotImplementedException(
            "is_socket_open() not implemented by {}".format(self.__str__())
        )

    def send(self, request):
        if self.state != ModbusTransactionState.RETRYING:
            _logger.debug("New Transaction state 'SENDING'")
            self.state = ModbusTransactionState.SENDING
        return self._send(request)

    def _send(self, request):
        """ Sends data on the underlying socket

        :param request: The encoded request to send
        :return: The number of bytes written
        """
        raise NotImplementedException(
            "Method not implemented by derived class")

    def recv(self, size):
        return self._recv(size)

    def _recv(self, size):
        """ Reads data from the underlying descriptor

        :param size: The number of bytes to read
        :return: The bytes read
        """
        raise NotImplementedException(
            "Method not implemented by derived class")

    # ----------------------------------------------------------------------- #
    # Modbus client methods
    # ----------------------------------------------------------------------- #
    def execute(self, request=None):
        """
        :param request: The request to process
        :returns: The result of the request execution
        """
        if not self.connect():
            raise ConnectionException(
                "Failed to connect[%s]" % (self.__str__()))
        return self.transaction.execute(request)

    # ----------------------------------------------------------------------- #
    # The magic methods
    # ----------------------------------------------------------------------- #
    def __enter__(self):
        """ Implement the client with enter block

        :returns: The current instance of the client
        """
        if not self.connect():
            raise ConnectionException(
                "Failed to connect[%s]" % (self.__str__()))
        return self

    def __exit__(self, klass, value, traceback):
        """ Implement the client with exit block """
        self.close()

    def idle_time(self):
        """
        Bus Idle Time to initiate next transaction
        :return: time stamp
        """
        if self.last_frame_end is None or self.silent_interval is None:
            return 0
        return self.last_frame_end + self.silent_interval

    def debug_enabled(self):
        """
        Returns a boolean indicating if debug is enabled.
        """
        return self._debug

    def set_debug(self, debug):
        """
        Sets the current debug flag.
        """
        self._debug = debug

    def trace(self, writeable):
        if writeable:
            self.set_debug(True)
        self._debugfd = writeable

    def _dump(self, data, direction):
        fd = self._debugfd if self._debugfd else sys.stdout
        try:
            fd.write(hexlify_packets(data))
        except Exception as e:
            _logger.debug(hexlify_packets(data))
            _logger.exception(e)

    def register(self, function):
        """
        Registers a function and sub function class with the decoder
        :param function: Custom function class to register
        :return:
        """
        self.framer.decoder.register(function)

    def __str__(self):
        """ Builds a string representation of the connection

        :returns: The string representation
        """
        return "Null Transport"`

dhoomakethu, 2022-09-08:
@AnonHax0r This sounds interesting, can you raise a PR so that we can review and see if it helps other folks as well ? Thanks.

AnonHax0r, 2022-09-08:
@dhoomakethu  of course, give me a little time to clean it up and get it working in the actual repo and I will do that. I have only tested this for ModbusTCP clients. I tried to use Asyncio but performance just wasn't there and faced similar issued with the fd count. I then went to try something synchronous combined with asyncio and came up with what I believe is a pretty good solution. This approach forced me to examine the pymodbus library and I found the select call was the issue. I will gladly share and update the small changes which I think will take the library to the next level for more than just a handful of local clients and allow users to create applications for scada dashboards that can scale a thousand or more devices in real time for large spans of registers.  I will try to get the PR pushed this weekend for you to review, but since it uses the selector package you could even pass in a param to the client to decide which selector class the base client uses incase there is a strict requirement for select over poll, but the API exposes the same calls for each class you chose. It appears to be a nice abstraction.

janiversen, 2022-09-08:
Be aware, that the sync server implementation no longer exist, we only have async implementation and a sync interface (to allow sync applications).

In a very short timeframe the same will happen to the sync clients.

janiversen, 2022-09-08:
We do not want to add extra complexity by giving the users possibility of selecting selector package, that is something internal.

AnonHax0r, 2022-09-08:
@janiversen well that is good news! And that is also understandable, it was unfortunately a requirement for me to modify for handling large quantities of connections. I will avoid submitting a PR then, hopefully the future release is able to handle the large sum via async, which I have no doubts it will knowing it will be python 3.7 or newer only.  Async is preferred by myself and I have both synchronous and asynchronous modes for my application so switching back will be a breeze. The issue async seemed to be causing me in 2.5.3 and older is that the requests would come back incomplete if I tried to handle multiple requests at once to complete a single 'payload'. There seemed to be a large latency issue and the amount of fd's i could have open at the time along with the packets seeming to struggle to complete. I will be glad to help testing 3.0 with a large scale application once it is ready. Until then I will be patient and leave these comments incase they help someone else trying to maintain thousands of connections.

janiversen, 2022-09-08:
You misunderstood me, please submit a PR and we will review it, I am not fan of the selector methods (I too do SCADA development), so I would very much like to see changes. The async version could have the same problem, apart from the fact that we do not handle the actual socket.

I just wanted to give you a heads up because you use the sync version, so you could focus better. Our current client async implementation is already a bit faster. Async in 2.5.3 is somewhat "rocky"; which is why the modbus implementation in Home Assistant still uses sync.

Multiple request in a client is something I want to do long term, but suggestions and enhancements (based on dev) are more than welcome.

Dev have come a long way, but if people like you do not test, we will not get the bugs solved. Minimum version is python 3.8

AnonHax0r, 2022-09-08:
I definitely did misunderstand, i havent collabed much via github so didnt want to step on toes, I appreciate the reply and all the work you guys have done. I will go ahead and submit that PR, and maybe it ends up helping with development in the future! I currently use python3.8 for my project, would love to jump to 3.10 but havent had time to test how it works with the libraries. I will see what other suggestions I can find on the async side as well as I would love to see this project continue to thrive, its been very useful for my development!

janiversen, 2022-09-08:
Thanks for the kind words.

Do not be afraid to step on toes, most maintainers have very thick skin 🥲 . The difference is that people who only complain are a real PITA, while those who help out are highly valued, please do not forget this is open source so all helping hands are welcome.

I have invested quite a lot of time the last year so for sure I want this project to grow, but it is hard to do alone, another reason to help out.

WillCodeForCats, 2022-09-30:
I spent a bunch of time with another user of my Home Assistant integration trying to figure out a similar issue, and the only thing we could come up with was that it was wifi related with their inverter, and that after the problem popped up restarting Home Assistant would fix the issue. My inverter is hardwired so I could never duplicate it. I'm also still too new to Python to effectively debug anything like the fd handle exhaustion idea.

Maybe worth a read in case anyone else has an idea:
[ModbusTcpClient(redacted:1502): Connection unexpectedly closed](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/80)

janiversen, 2022-09-30:
Be aware that it uses version 2.5.3 just like home assistant and we have since made a number of changes in pymodbus which should solve the issue.

Home assistant will be updated when we release 3.0.0

WillCodeForCats, 2022-11-04:
> Be aware that it uses version 2.5.3 just like home assistant and we have since made a number of changes in pymodbus which should solve the issue.
> 
> Home assistant will be updated when we release 3.0.0

Are these the only changes needed in HA to work with 3.0.0 ?

https://github.com/home-assistant/core/compare/dev...WillCodeForCats:home-assistant-core:pymodbus-3-0-0

janiversen, 2022-11-04:
No there are quite a number of other changes needed. As the current maintainer of the modbus integration in HA I am planning to bump the library version when 3.0.1 is available,

However do not expect big differences, I work on making the modbus integration use pymodbus async, which will bring a much better performance and solve a number of issues.

Using 3.0.0 in HA will lead to problems, just see what is fixed  on our current dev.

WillCodeForCats, 2022-11-04:
I already pinned my custom integration to 3.0.0 becasue it fixed the communication issues that were reported, so while it does fix those issues it will not coexist if someone is also loading the HA core modbus integration.

I'll tell my users that for now they have to wait for HA to catch up.

roykrikke, 2022-11-05:
I ran into the problem as described in: https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/162

After some reading and searching for a solution I came across a new release: [pymodbus 3.0.2](https://pymodbus.readthedocs.io/en/latest/changelog.html#version-3-0-2)

I hope the issue is resolved soon. In the meantime, we wait patiently. Feel free to ping me if a tester is needed.

janiversen, 2022-11-05:
??? Do you have the problem in the new release 3.0.2 ? I am confused as to what your problem really is.

The issue was reported with pretty old versions of pymodbus, and quite a lot have changed since then especially in this area. I for one can no longer provoke this problem.

janiversen, 2022-11-05:
Be aware the link you added are about HomeAssistant which still uses pymodbus 2.5.3 (I should know as I am the active maintainer of that integration). HomeAssistant are due to be upgraded very soon.

roykrikke, 2022-11-06:
> ??? Do you have the problem in the new release 3.0.2 ? I am confused as to what your problem really is.
> 
> The issue was reported with pretty old versions of pymodbus, and quite a lot have changed since then especially in this area. I for one can no longer provoke this problem.

Sorry, I expressed myself a bit clumsily. I have no issues other then shared in https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/162 Just wanted to report that pymodbus  3.0.2 was released yesterday. In the posts above versions 3.0.0 and 3.0.1 of pymodbus were mentioned. I should have been clear about that! Anyway, I understand your feedback. Thanks for pointing it out to me, hope this post provides some clarity.

I understood that upgrading the HA modbus integration is a bit more complicated than expected. Good luck with the integration and thank you for your commitment to the HA modbus integration

janiversen, 2022-12-07:
Should be solved on dev.

closed 2022-12-07 by janiversen