pymodbus-dev/pymodbus #879: Creating context and decoding on client

aasick, 2022-05-05:
I try to create server context and tried to decode and print on client side. But always it's throwing error.
PyModbus example code works fine. when I try to modify always registers error occurred.

**Server code:**
    builder.add_8bit_int(-0x12)
    builder.add_8bit_uint(0x12)
    builder.add_16bit_int(-0x5678)
    builder.add_16bit_uint(0x1234)
    builder.add_32bit_int(-0x1234)
    builder.add_32bit_uint(0x12345678)
    builder.add_16bit_float(12.34)
    builder.add_16bit_float(-12.34)
    builder.add_32bit_float(22.34)
    builder.add_32bit_float(-22.34)
    builder.add_64bit_int(-0xDEADBEEF)
    builder.add_64bit_uint(0x12345678DEADBEEF)
    builder.add_64bit_uint(0xDEADBEEFDEADBEED)
    builder.add_64bit_float(123.45)
    builder.add_64bit_float(-123.45)

    # ----------------------------------------------------------------------- #
    # use that payload in the data store
    # ----------------------------------------------------------------------- #
    # Here we use the same reference block for each underlying store.
    # ----------------------------------------------------------------------- #
    print(builder.to_registers())
    block = ModbusSequentialDataBlock(1, builder.to_registers())
    store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
    context = ModbusServerContext(slaves=store, single=True)

**Client code:**
    result = client.read_holding_registers(address=0x00, count=17, unit=1)
    print("-result", result)
    print("Registers")
    print("-" * 60)
    print(result.registers)
    print("\n")
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.Little, wordorder=Endian.Little)
    decoded = OrderedDict([
        # ('string', decoder.decode_8bit_int()),
        # ('bits', decoder.decode_8bit_int()),
        ('8int', decoder.decode_8bit_int()),
        ('8int', decoder.decode_8bit_int()),
        ('8int', decoder.decode_8bit_int()),
        ('8int', decoder.decode_8bit_int()),
        ('8int', decoder.decode_8bit_int()),

    ])

**Server error:**
2022-05-05 10:47:05,013 MainThread      DEBUG    payload        :125      [61166, 61166, 60928]
[61166, 61166, 60928]
2022-05-05 10:47:05,013 MainThread      DEBUG    payload        :125      [61166, 61166, 60928]
2022-05-05 10:47:05,015 MainThread      DEBUG    sync           :46       Client Connected [COM2:COM2]
2022-05-05 10:47:05,016 MainThread      DEBUG    sync           :581      Started thread to serve client
2022-05-05 10:47:09,052 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x0 0x0 0x0 0x11
2022-05-05 10:47:09,053 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-05-05 10:47:09,055 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-05-05 10:47:09,057 MainThread      DEBUG    context        :63       validate: fc-[3] address-1: count-17
2022-05-05 10:47:09,058 MainThread      ERROR    pdu            :107      Exception Response(131, 3, IllegalAddress)
2022-05-05 10:47:09,059 MainThread      DEBUG    sync           :144      send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'

**Client error:**
2022-05-05 10:47:08,627 MainThread      DEBUG    transaction    :139      Current transaction state - IDLE
2022-05-05 10:47:08,627 MainThread      DEBUG    transaction    :144      Running transaction 1
2022-05-05 10:47:08,628 MainThread      DEBUG    transaction    :273      SEND: 0x1 0x3 0x0 0x0 0x0 0x11 0x85 0xc6
2022-05-05 10:47:08,628 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-05-05 10:47:08,628 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-05-05 10:47:09,060 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-05-05 10:47:09,061 MainThread      DEBUG    transaction    :297      RECV: 0x1 0x83 0x2 0xc0 0xf1
2022-05-05 10:47:09,062 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x83 0x2
2022-05-05 10:47:09,063 MainThread      DEBUG    factory        :266      Factory Response[131]
2022-05-05 10:47:09,063 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-05-05 10:47:09,064 MainThread      DEBUG    transaction    :454      Adding transaction 1
2022-05-05 10:47:09,064 MainThread      DEBUG    transaction    :465      Getting transaction 1
2022-05-05 10:47:09,065 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
-result Exception Response(131, 3, IllegalAddress)
Registers
------------------------------------------------------------
Traceback (most recent call last):
  File "sender.py", line 67, in <module>
    run_server()
  File "sender.py", line 30, in run_server
    print(result.registers)
AttributeError: 'ExceptionResponse' object has no attribute 'registers'

C:\github\tool-protocol-simulator\Modbus_simulator>

aasick, 2022-05-05:
https://github.com/dhoomakethu can you take a look, I need to modify payload for power meter readings

dhoomakethu, 2022-05-05:
Have you looked in to this example ? https://github.com/riptideio/pymodbus/blob/dev/examples/common/modbus_payload_server.py

aasick, 2022-05-09:
@dhoomakethu we have any example, based on address request from slave need to send payload from server