pymodbus-dev/pymodbus #900: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

axelfatherofpeace, 2022-05-19:
### Versions

* Python: 3.6
* OS: Win 10
* Pymodbus: 
* Modbus Hardware (if used):  DELTA PLC- DVP-10SX

### Pymodbus Specific
* Server: ascii - sync
* Client: ascii - sync

### Description

Want to read the output from the plc to pc using python . 

### Code and Logs
from pymodbus.client.sync import ModbusSerialClient

from pymodbus.client.sync import ModbusTcpClient

import serial

client= ModbusSerialClient(method = "ASCII", port="COM3",stopbits = 1, bytesize = 8, parity = 'N', baudrate= 9600)

client.connect()

result = client.read_coils(2048,10, unit = 1)
print(result)

Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
```python

def onRedLight():
    client.write_coil(2048,1)

def offRedLight():
    client.write_coil(2048,0)

def onGreenLight():
    client.write_coil(2049,1)

def offGreenLight():
    client.write_coil(2049,0)

client.write_coil(2049,1)

onGreenLight()

onRedLight()

offRedLight()

offGreenLight()

client.close()

 # code and logs here.

```

dhoomakethu, 2022-05-21:
@axelfatherofpeace based on the gitter chat, is this issue still open?

axelfatherofpeace, 2022-05-25:
..

closed 2022-05-25 by axelfatherofpeace