pymodbus-dev/pymodbus #938: multiple slave unable to run on thread

aasick, 2022-06-13:
I am getting following error

pymodbus.exceptions.NoSuchSlaveException: Modbus Error: [No Such Slave] slave - 0x01 does not exist, or is out of range

On my code I try to run updating server with two slaves with thread. but for single slave it works fine and updating value as expected. But for two slaves I unable to update values. at the time of running getting above error

def updating_writer(a):
    log.debug("updating the context")
    context  = a[0]
    register = 3
    power_data = {'0x01': {0x01: 8, 65: 8, 961: 8}, '0x02': {0x01: 8, 19: 8, 21: 8, 23: 8, 6687: 8}}
    for slave_id in power_data:
        for address, write_count in power_data[slave_id].items():
            values = context[slave_id].getValues(register, address, count=write_count)
            txt1 = f"old values: {str(values)}"
            log.debug(txt1)
            values = [v + 1 for v in values]
            txt = f"new values: {str(values)}"
            log.debug(txt)
            context[slave_id].setValues(register, address, values)

janiversen, 2022-06-13:
The line “ context = a[0]” looks wrong, context should have the same number of entries as the highest sleve_id, because later you do “ context[slave_id]”.

aasick, 2022-06-13:
@janiversen for now I have two slaves, so can I make it as "context = a[1]" this will help me?

janiversen, 2022-06-13:
a[0] and a[1] are both just a single element in a, you need to ensure that a contains both [0] and [1] and then assign a to context.

aasick, 2022-06-13:
below is my exact code for updating server with two slaves and need to add another two also in future
@janiversen @dhoomakethu you have idea on this

block = ModbusSparseDataBlock({64002: device_info_model, 65: [0]*12, 961: [0]*12,
                                   0: [0]*12}, mutable=False)
si_datastore = ModbusSlaveContext(hr=block)

block_slave2 = ModbusSparseDataBlock({64000: device_info_model, 19: active_power, 21: active_power2, 23: active_power3,
                                          6687: active_energy, 0: [0]*12}, mutable=False)
sie_datastore = ModbusSlaveContext(hr=block_slave2)

slaves = {0x01: si_datastore, 0x02: sie_datastore}
context = ModbusServerContext(slaves=slaves, single=False)

server = ModbusSerialServer(context, framer=ModbusRtuFramer, port='COM9', stopbits=1, bytesize=8,
                                parity='E', baudrate=9600, timeout=1)

t = threading.Thread(target=server.serve_forever, daemon=True)
t.start()
log.debug(context)
loop = LoopingCall(f=updating_writer, a=(context,))
loop.start(update_time, now=True)

reactor.run()

  def updating_writer(a):
      log.debug("updating the context")
      context  = a[0]
      register = 3
      power_data = {'0x01': {0x01: 8, 65: 8, 961: 8}, '0x02': {0x01: 8, 19: 8, 21: 8, 23: 8, 6687: 8}}
      for slave_id in power_data:
          for address, write_count in power_data[slave_id].items():
              values = context[slave_id].getValues(register, address, count=write_count)
              txt1 = f"old values: {str(values)}"
              log.debug(txt1)
              values = [v + 1 for v in values]
              txt = f"new values: {str(values)}"
              log.debug(txt)
              context[slave_id].setValues(register, address, values)

dhoomakethu, 2022-06-14:
`power_data = {'0x01': {0x01: 8, 65: 8, 961: 8}, '0x02': {0x01: 8, 19: 8, 21: 8, 23: 8, 6687: 8}}`

Your `slave_id` is string instead of an integer. Please update it accordingly and it should work.  Also you don't have to pass `context` as part of tuple in the looping call. This should avoid getting the actual context by index.

```
...
...
loop = LoopingCall(f=updating_writer, context=context)

def updating_writer(context: ModbusServerContext):
      log.debug("updating the context")
      register = 3
     ....

```

closed 2022-06-14 by dhoomakethu