pymodbus-dev/pymodbus #939: Function code 43 fails because of incomplete receival of bytes

balaARR, 2022-06-13:
### Versions

* Python: 3.6.5
* OS: Windows 11
* Pymodbus: 2.5.3
* Hardware ():  
         - ic7 Danfoss drive with RS485 support
                 - Drive has been directly connected to the Test PC through DTech( USB to RS485) cable and tested through a small python code.
      

### Pymodbus Specific
* Server(master): rtu- async
* Client(slave): rtu - async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

We were trying to execute function code 43/14 through the pymodbus package in the above-mentioned drive specification.
For any baud rate less than **115200**, the execution of function code 43/14 fails as pymodbus package fails to read all the response bytes. Possible root cause/issue found in _wait_for_data() function**(available in sync.py)**. This function sleeps for 10 ms after each iteration of reading from serial port, if no new data/bytes are received in the next iteration, it breaks the loop and returns the incomplete response. So in some cases like ours (with FC43/14) and at lower baud rates this delay is not sufficient.

### Code and Logs

**The below-mentioned sample python code has been used to execute the function code 43/14**

from pymodbus.client.sync import ModbusTcpClient as ModbusTCPClient
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.payload import BinaryPayloadBuilder
from pymodbus.mei_message import *
from pymodbus.diag_message import *
from pymodbus.constants import *
from pymodbus.pdu import ExceptionResponse
from pymodbus.framer.socket_framer import *
from pymodbus.framer.rtu_framer import *
from pymodbus.factory import ClientDecoder
from pymodbus.transaction import ModbusTransactionManager
from pymodbus.payload import BinaryPayloadBuilder
from pymodbus.pdu import ModbusExceptions
from pymodbus.exceptions import ModbusException
#from utility.logger import Logger
import re
import struct
from pymodbus.bit_read_message import *
from pymodbus.bit_write_message import *
from pymodbus.register_read_message import *
from pymodbus.register_write_message import *
from pymodbus.utilities import computeCRC
from pymodbus.transaction import ModbusTransactionManager
from pymodbus.payload import BinaryPayloadBuilder
from pymodbus.mei_message import ReadDeviceInformationRequest
from pymodbus.pdu import ExceptionResponse
from pymodbus.pdu import ModbusExceptions
from pymodbus.exceptions import ModbusException
import time

class ModbusRTU():
  def init(self):
    self.connection = None

  def connect(self, method, port):
    self.connection = ModbusClient(method=method, port=port, baudrate=19200, parity='E', strict=False)
    if self.connection.connect():
      print("RTU Connected")
  
  def read_device_information(self):
    request = ReadDeviceInformationRequest(1, 1, unit=1)
    result = self.connection.execute(request)
    print(result)

modbusrtu = ModbusRTU()
modbusrtu.connect('rtu', 'COM4')
modbusrtu.read_device_information()

The following image is the output of the above-mentioned snippet:
[image]

Here in our scenario, we are expecting a response of byte size 82, but actually, we are not receiving the complete bytes. Hence that fails our test scenario.

Could you please consider this issue and provide us with some solution for that?

dhoomakethu, 2022-06-14:
Will increasing read timeout helps ?

janiversen, 2022-06-14:
very likely.

balaARR, 2022-06-16:
**Will increasing read timeout helps?**

@dhoomakethu 
If you are pointing out the timeout which is getting specified as part of "connect" method, then increasing that timeout is not helping us. 

Else, if you are mentioning time.sleep in sync.py file, then yes it works for function 43/14.

dhoomakethu, 2022-06-17:
@balaARR I would suggest to use a patched version of pymodbus if you think that helps. I couldn't find datasheet or modbus spec for the device (`ic7 Danfoss drive with RS485 support` ) and with out logs or anything else to take a more closer look there is nothing much we can do here.

balaARR, 2022-06-22:
@dhoomakethu 

Could you please let me know what kind of logs we need to provide for this issue? So, we will try to provide those logs accordingly.

dhoomakethu, 2022-07-13:
@balaARR just enable debug logs

janiversen, 2022-08-13:
Still on the list to be corrected.

janiversen, 2022-11-07:
@balaARR please check with version 3.0.2, it should be solved.

balaARR, 2022-11-07:
@janiversen  Yeah sure will try. Thank you for sharing this information.

janiversen, 2022-11-11:
I just added a commit on dev, that might help a bit more.

janiversen, 2022-12-12:
still open, getting close to the top of my list

closed 2023-02-20 by github-actions[bot]