pymodbus-dev/pymodbus #956: pymodbus.exceptions.ModbusIOException('No Response received from the remote unit/Unable to decode response',1)

tiagomiguelrs, 2022-07-10:
### Versions

* Python: 3.8.8
* OS: Win 10
* Pymodbus: 2.5.3
* Modbus Hardware: Omron CP1L-EL (ethernet built-in port)

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

Hi, I am trying to connect to an Omron CP1L-EL PLC because I need to create an app for system monitoring.
What happens is described bellow in the debug log. I command the PLC to read the first coil. However. the PLC sends a message that can't be decoded by pymodbus.
I have found that there is connection between the PLC and the PC because the `client.open()` command returns `True`. But everytime I ask it to do something, it doesn't respond as expected.
I am not an expert, but I have tried changing the unit ID both in the PLC and the code, IP numbers, ports, everything that is most common.
Also, I feel like the sending message from pymodbus seems to have too many bytes? just speculating. And I can't make sense of the recieving message as well...
Also, to operate with this PLC in Modbus I had to download a program developped by Omron that I send to the PLC's memory, but within it, the "connected" contact never turns on when I open connection with pymodbus...
I'll leave the pdf file here that contains the Omron program:

[MTCP_CP1L_Client_E.pdf](https://github.com/riptideio/pymodbus/files/9079096/MTCP_CP1L_Client_E.pdf)

### Code and Logs

```python
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.constants import Defaults

#---------------------------------------------------------------------------# 
# configure the client logging
#---------------------------------------------------------------------------# 
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

#---------------------------------------------------------------------------# 
# choose the client you want
#---------------------------------------------------------------------------# 
# make sure to start an implementation to hit against. For this
# you can use an existing device, the reference implementation in the tools
# directory, or start a pymodbus server.
#---------------------------------------------------------------------------# 

client = ModbusClient('192.168.250.100', port=9600, timeout=10)
client.connect()

#---------------------------------------------------------------------------# 
# example requests
#---------------------------------------------------------------------------# 
# simply call the methods that you would like to use. An example session
# is displayed below along with some assert checks.
#---------------------------------------------------------------------------# 
rr = client.read_coils(1,1, unit=1)
assert(rr.bits[0] == True)          # test the expected value

client.close()

```

```Debug log

2022-07-10 13:40:42,230 MainThread      DEBUG    sync           :215      Connection to Modbus server established. Socket ('192.168.250.200', 50284)
2022-07-10 13:40:42,231 MainThread      DEBUG    transaction    :139      Current transaction state - IDLE
2022-07-10 13:40:42,231 MainThread      DEBUG    transaction    :144      Running transaction 1
2022-07-10 13:40:42,231 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x1 0x0 0x1
2022-07-10 13:40:42,231 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-07-10 13:40:42,232 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-07-10 13:40:42,235 MainThread      DEBUG    transaction    :367      Incomplete message received, Expected 6 bytes Recieved 8 bytes !!!!
2022-07-10 13:40:42,235 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-07-10 13:40:42,236 MainThread      DEBUG    transaction    :297      RECV: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
2022-07-10 13:40:42,236 MainThread      DEBUG    socket_framer  :147      Processing: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
2022-07-10 13:40:42,236 MainThread      DEBUG    socket_framer  :159      Frame check failed, ignoring!!
2022-07-10 13:40:42,236 MainThread      DEBUG    transaction    :465      Getting transaction 1
2022-07-10 13:40:42,237 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

janiversen, 2022-07-10:
try with read_coil instead of read_coils, some devices have problems responding to a multiple coil request.

The send packet looks OK, but it look like the device responds with an extra coil.

tiagomiguelrs, 2022-07-10:
@janiversen I don't think there is a `read_coil` function in pymodbus in the docs. I tried anyways, but it returns an error saying """'ModbusTcpClient' object has no attribute 'read_coil'"""

janiversen, 2022-07-10:
Sorry it is in my development version.

tiagomiguelrs, 2022-07-10:
I used it because it is similar to what is in the documentation example. Removing doesn't solve it. The code returns the same as in the debug log above except the number of the unit is 0 instead of 1.

tiagomiguelrs, 2022-07-13:
I tried this:

```
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.constants import Defaults

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient('192.168.250.1', port=9600, timeout=10)
client.connect()

rq = client.write_coil(1, True, unit=1)
rr = client.read_coils(2, 1, unit=0x1)

rq = client.write_coils(1, [True]*8, unit=1)
rr = client.read_discrete_inputs(1, 8, unit=1)

rq = client.write_register(1, 10, unit=1)
rr = client.read_holding_registers(1, 1, unit=1)

rq = client.write_registers(1, [10]*8, unit=1)
rr = client.read_input_registers(1, 8, unit=1)

client.close()
```

And got the following debug log:

```
2022-07-13 12:53:10,597 MainThread      DEBUG    sync           :215      Connection to Modbus server established. Socket ('192.168.250.200', 61358)
2022-07-13 12:53:10,598 MainThread      DEBUG    transaction    :139      Current transaction state - IDLE
2022-07-13 12:53:10,599 MainThread      DEBUG    transaction    :144      Running transaction 1
2022-07-13 12:53:10,599 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x5 0x0 0x1 0xff 0x0
2022-07-13 12:53:10,600 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-07-13 12:53:10,600 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-07-13 12:53:10,603 MainThread      DEBUG    transaction    :367      Incomplete message received, Expected 6 bytes Recieved 8 bytes !!!!
2022-07-13 12:53:10,604 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-07-13 12:53:10,605 MainThread      DEBUG    transaction    :297      RECV: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
2022-07-13 12:53:10,605 MainThread      DEBUG    socket_framer  :147      Processing: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
2022-07-13 12:53:10,606 MainThread      DEBUG    socket_framer  :159      Frame check failed, ignoring!!
2022-07-13 12:53:10,606 MainThread      DEBUG    transaction    :465      Getting transaction 1
2022-07-13 12:53:10,607 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-07-13 12:53:10,610 MainThread      DEBUG    sync           :215      Connection to Modbus server established. Socket ('192.168.250.200', 61359)
2022-07-13 12:53:10,612 MainThread      DEBUG    transaction    :139      Current transaction state - TRANSACTION_COMPLETE
2022-07-13 12:53:10,612 MainThread      DEBUG    transaction    :144      Running transaction 2
2022-07-13 12:53:10,612 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x2 0x0 0x1
2022-07-13 12:53:10,612 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-07-13 12:53:10,614 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-07-13 12:53:10,618 MainThread      DEBUG    transaction    :367      Incomplete message received, Expected 6 bytes Recieved 8 bytes !!!!
2022-07-13 12:53:10,619 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-07-13 12:53:10,619 MainThread      DEBUG    transaction    :297      RECV: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
2022-07-13 12:53:10,620 MainThread      DEBUG    socket_framer  :147      Processing: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
2022-07-13 12:53:10,620 MainThread      DEBUG    socket_framer  :159      Frame check failed, ignoring!!
2022-07-13 12:53:10,621 MainThread      DEBUG    transaction    :465      Getting transaction 2
2022-07-13 12:53:10,621 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-07-13 12:53:10,629 MainThread      DEBUG    sync           :215      Connection to Modbus server established. Socket ('192.168.250.200', 61360)
2022-07-13 12:53:10,629 MainThread      DEBUG    transaction    :139      Current transaction state - TRANSACTION_COMPLETE
2022-07-13 12:53:10,630 MainThread      DEBUG    transaction    :144      Running transaction 3
2022-07-13 12:53:10,631 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x3 0x0 0x0 0x0 0x8 0x1 0xf 0x0 0x1 0x0 0x8 0x1 0xff
2022-07-13 12:53:10,631 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-07-13 12:53:10,632 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-07-13 12:53:10,636 MainThread      DEBUG    transaction    :367      Incomplete message received, Expected 6 bytes Recieved 8 bytes !!!!
2022-07-13 12:53:10,636 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-07-13 12:53:10,637 MainThread      DEBUG    transaction    :297      RECV: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
2022-07-13 12:53:10,637 MainThread      DEBUG    socket_framer  :147      Processing: 0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10
[... 46 lines cut ...]
2022-07-13 12:53:10,709 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
[... cut ...]
2022-07-13 12:53:10,733 MainThread      DEBUG    socket_framer  :159      Frame check failed, ignoring!!
2022-07-13 12:53:10,733 MainThread      DEBUG    transaction    :465      Getting transaction 8
2022-07-13 12:53:10,734 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

So, for all function codes, my PLC always returns the same message: `0x46 0x49 0x4e 0x53 0x0 0x0 0x0 0x10`. This means that I need to find a way to reconfgure the modbus connection within the PLC right?

janiversen, 2022-08-13:
Yes the PLC is telling you there are something it does not like.

closed 2022-08-13 by janiversen