pymodbus-dev/pymodbus #962: UDP-Client: Unexpected log, `Expected 1024 bytes`

Eideen, 2022-07-17:
### Versions

* Python: `3.10.4`
* OS: `Ubuntu 22.04 LTS`
* Pymodbus: `2.5.3`

### Pymodbus Specific
* Server: UDP - sync
* Client: UDP - sync

### Description

Trying to test/learn Modbus TCP/UDP.

Change `common` example to UDP.

Got a unexpected log errors:
````python
2022-07-17 17:04:45,010 MainThread      DEBUG    transaction    :367      Incomplete message received, Expected 1024 bytes Recieved 10 bytes !!!!
````

### Code and Logs

#### Log output

````python
2022-07-17 17:04:40,004 MainThread      DEBUG    synchronous_client:87       Reading Coils
2022-07-17 17:04:45,009 MainThread      DEBUG    transaction    :139      Current transaction state - IDLE
2022-07-17 17:04:45,009 MainThread      DEBUG    transaction    :144      Running transaction 1
2022-07-17 17:04:45,009 MainThread      DEBUG    transaction    :273      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x1 0x0 0x2
2022-07-17 17:04:45,009 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2022-07-17 17:04:45,009 MainThread      DEBUG    transaction    :287      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-07-17 17:04:45,010 MainThread      DEBUG    transaction    :367      Incomplete message received, Expected 1024 bytes Recieved 10 bytes !!!!
2022-07-17 17:04:45,010 MainThread      DEBUG    transaction    :375      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-07-17 17:04:45,010 MainThread      DEBUG    transaction    :297      RECV: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x3
2022-07-17 17:04:45,010 MainThread      DEBUG    socket_framer  :147      Processing: 0x0 0x1 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x3
2022-07-17 17:04:45,010 MainThread      DEBUG    factory        :266      Factory Response[ReadCoilsResponse: 1]
2022-07-17 17:04:45,010 MainThread      DEBUG    transaction    :454      Adding transaction 1
2022-07-17 17:04:45,010 MainThread      DEBUG    transaction    :465      Getting transaction 1
2022-07-17 17:04:45,010 MainThread      DEBUG    transaction    :224      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-07-17 17:04:50,013 MainThread      DEBUG    synchronous_client:91       ReadCoilsResponse(8)

````

#### File changes

note: @@ -110,22 +110,15 @@ def run_server(): changes is different bug.

````diff
--- a/examples/common/synchronous_client.py
+++ b/examples/common/synchronous_client.py
@@ -13,7 +13,7 @@ import logging
 # --------------------------------------------------------------------------- #
 # import the various client implementations
 # --------------------------------------------------------------------------- #
-from pymodbus.client.sync import ModbusTcpClient as ModbusClient
+from pymodbus.client.sync import ModbusUdpClient as ModbusClient
 
 # from pymodbus.client.sync import ModbusUdpClient as ModbusClient
 # from pymodbus.client.sync import ModbusSerialClient as ModbusClient

--- a/examples/common/synchronous_server.py
+++ b/examples/common/synchronous_server.py
@@ -19,7 +19,7 @@ from pymodbus.datastore import (
 # from pymodbus.server.sync import StartUdpServer
 # from pymodbus.server.sync import StartSerialServer
 from pymodbus.device import ModbusDeviceIdentification
-from pymodbus.server.sync import StartTcpServer
+from pymodbus.server.sync import StartUdpServer
 
 # --------------------------------------------------------------------------- #
 # import the various server implementations
@@ -110,22 +110,15 @@ def run_server():
     # ----------------------------------------------------------------------- #
     # If you don"t set this or any fields, they are defaulted to empty strings.
     # ----------------------------------------------------------------------- #
-    identity = ModbusDeviceIdentification(
-        info_name={
-            "VendorName": "Pymodbus",
-            "ProductCode": "PM",
-            "VendorUrl": "http://github.com/riptideio/pymodbus/",
-            "ProductName": "Pymodbus Server",
-            "ModelName": "Pymodbus Server",
-            "MajorMinorRevision": version.short(),
-        }
-    )
+    identity = ModbusDeviceIdentification(   )
+
+
 
     # ----------------------------------------------------------------------- #
     # run the server you want
     # ----------------------------------------------------------------------- #
     # Tcp:
-    StartTcpServer(context, identity=identity, address=("", 5020))
+    StartUdpServer(context, identity=identity, address=("", 5020))
     #
     # TCP with different framer
     # StartTcpServer(context, identity=identity,

````

janiversen, 2022-07-17:
Thanks for reporting the issue, it looks like a bug. I will try to simulate it.

janiversen, 2022-07-24:
First test, running sync client/server example without your modifications, SUCCESS.

Next up, test your modifications.

PR #978 (2022-07-25): Update synchronous client/server examples

Simplified/cleaned sync client to prepare for updated examples.

The new client allow for easy switching between comm methods and frames.

this is done to test/verify/solve
fixes #962
closed 2022-07-25 by janiversen