pymodbus-dev/pymodbus #963: Server: Unable to run with `info_name`

Eideen, 2022-07-17:
### Versions

* Python: `3.10.4`
* OS: `Ubuntu 22.04 LTS`
* Pymodbus: `2.5.3`

### Pymodbus Specific
* Server: UDP - sync
* Client: UDP - sync

### Description

Trying to test/learn Modbus TCP/UDP.

Unable to run without removing `info_name`.
Work if ` info_name` was removed

### Code and Logs

#### Log output

````python
17:39 $ ./synchronous_server.py 
Traceback (most recent call last):
  File "/home/torstein/Dokumenter/git-workfolder/modbus_test/pymodbus/examples/common/./synchronous_server.py", line 166, in <module>
    run_server()
  File "/home/torstein/Dokumenter/git-workfolder/modbus_test/pymodbus/examples/common/./synchronous_server.py", line 113, in run_server
    identity = ModbusDeviceIdentification(
TypeError: ModbusDeviceIdentification.__init__() got an unexpected keyword argument 'info_name'

````

#### File changes

````diff

--- a/examples/common/synchronous_server.py
+++ b/examples/common/synchronous_server.py
@@ -19,7 +19,7 @@ from pymodbus.datastore import (
 # from pymodbus.server.sync import StartUdpServer
 # from pymodbus.server.sync import StartSerialServer
 from pymodbus.device import ModbusDeviceIdentification
-from pymodbus.server.sync import StartTcpServer
+from pymodbus.server.sync import StartUdpServer
 
 # --------------------------------------------------------------------------- #
 # import the various server implementations
@@ -110,22 +110,15 @@ def run_server():
     # ----------------------------------------------------------------------- #
     # If you don"t set this or any fields, they are defaulted to empty strings.
     # ----------------------------------------------------------------------- #
-    identity = ModbusDeviceIdentification(
-        info_name={
-            "VendorName": "Pymodbus",
-            "ProductCode": "PM",
-            "VendorUrl": "http://github.com/riptideio/pymodbus/",
-            "ProductName": "Pymodbus Server",
-            "ModelName": "Pymodbus Server",
-            "MajorMinorRevision": version.short(),
-        }
-    )
+    identity = ModbusDeviceIdentification(   )

````

janiversen, 2022-07-17:
Seems like this example was never updated.

Eideen, 2022-07-20:
@dhoomakethu I have test with the `3.0.0.dev4` tag, and no errors!

I probably use the `2.5.3`  server together with the  `3.0.0.dev4` example files.

closed 2022-07-20 by Eideen