pymodbus-dev/pymodbus #968: UDP - server: Cosmetic  Opening/Closing connections on calls.

Eideen, 2022-07-23:
### Versions

* Python: `3.10.4`
* OS: `Ubuntu 22.04 LTS`
* Pymodbus: `dev`

### Pymodbus Specific
* Server: UDP - sync
* Client: UDP - sync

### Description

Trying to test/learn Modbus TCP/UDP.

It looks like the server is unable to "reuse" the connection, when it is running is UDP mode. For firewalls it is normal to have 60 sec delay before the connection is closed.
It may be missing just a timeout, or a array that keeps track of it.

Change `common` example to UDP.

Got a unexpected log:
````python
2022-07-23 11:14:31,487 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)
2022-07-23 11:14:31,488 Thread-32 (process_request_thread) DEBUG    sync           :44       Client Connected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,488 Thread-32 (process_request_thread) DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 53689)]
````

### Code and Logs

#### Log output

````python
2022-07-23 11:14:31,487 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)
2022-07-23 11:14:31,488 Thread-32 (process_request_thread) DEBUG    sync           :44       Client Connected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,488 Thread-32 (process_request_thread) DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,489 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)
2022-07-23 11:14:31,489 Thread-33 (process_request_thread) DEBUG    sync           :44       Client Connected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,490 Thread-33 (process_request_thread) DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,490 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)
2022-07-23 11:14:31,490 Thread-34 (process_request_thread) DEBUG    sync           :44       Client Connected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,491 Thread-34 (process_request_thread) DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,492 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)
2022-07-23 11:14:31,492 Thread-35 (process_request_thread) DEBUG    sync           :44       Client Connected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,492 Thread-35 (process_request_thread) DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,492 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)
2022-07-23 11:14:31,493 Thread-36 (process_request_thread) DEBUG    sync           :44       Client Connected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,493 Thread-36 (process_request_thread) DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,493 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)
2022-07-23 11:14:31,493 Thread-37 (process_request_thread) DEBUG    sync           :44       Client Connected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,494 Thread-37 (process_request_thread) DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 53689)]
2022-07-23 11:14:31,494 MainThread      DEBUG    sync           :516      Started thread to serve client at ('127.0.0.1', 53689)

````

#### File changes

note: @@ -110,22 +110,15 @@ def run_server(): changes is different bug.

````diff
--- a/examples/common/synchronous_client.py
+++ b/examples/common/synchronous_client.py
@@ -13,7 +13,7 @@ import logging
 # --------------------------------------------------------------------------- #
 # import the various client implementations
 # --------------------------------------------------------------------------- #
-from pymodbus.client.sync import ModbusTcpClient as ModbusClient
+from pymodbus.client.sync import ModbusUdpClient as ModbusClient
 
 # from pymodbus.client.sync import ModbusUdpClient as ModbusClient
 # from pymodbus.client.sync import ModbusSerialClient as ModbusClient

--- a/examples/common/synchronous_server.py
+++ b/examples/common/synchronous_server.py
@@ -19,7 +19,7 @@ from pymodbus.datastore import (
 # from pymodbus.server.sync import StartUdpServer
 # from pymodbus.server.sync import StartSerialServer
 from pymodbus.device import ModbusDeviceIdentification
-from pymodbus.server.sync import StartTcpServer
+from pymodbus.server.sync import StartUdpServer

 
     # ----------------------------------------------------------------------- #
     # run the server you want
     # ----------------------------------------------------------------------- #
     # Tcp:
-    StartTcpServer(context, identity=identity, address=("", 5020))
+    StartUdpServer(context, identity=identity, address=("", 5020))
     #
     # TCP with different framer
     # StartTcpServer(context, identity=identity,

````

janiversen, 2022-07-23:
Yes it is not part of the sync ModbusUDPServer.

janiversen, 2022-07-23:
UDP is connectionless, to you could more correctly claim the debug output is false. ModbusUDP uses sendTo() without a previous connection.

UDP do have have timeout, because there are no acknowledgment of the message being received, so nothing to wait for.

Closing the issue as the current handling is correct.

closed 2022-07-23 by janiversen