pymodbus-dev/pymodbus #969: UDP - Client: Error handling if no response

Eideen, 2022-07-23:
### Versions

* Python: `3.10.4`
* OS: `Ubuntu 22.04 LTS`
* Pymodbus: `dev`

### Pymodbus Specific
* Server: -
* Client: UDP - sync

### Description

Trying to test/learn Modbus TCP/UDP.

Change `common` example to UDP.

If the UDP client have no, server, or there is lose on the way. 

There is no function that will resend it. 

### Code and Logs

#### Log output

none. 

#### File changes

note: @@ -110,22 +110,15 @@ def run_server(): changes is different bug.

````diff
--- a/examples/common/synchronous_client.py
+++ b/examples/common/synchronous_client.py
@@ -13,7 +13,7 @@ import logging
 # --------------------------------------------------------------------------- #
 # import the various client implementations
 # --------------------------------------------------------------------------- #
-from pymodbus.client.sync import ModbusTcpClient as ModbusClient
+from pymodbus.client.sync import ModbusUdpClient as ModbusClient
 
 # from pymodbus.client.sync import ModbusUdpClient as ModbusClient
 # from pymodbus.client.sync import ModbusSerialClient as ModbusClient

--- a/examples/common/synchronous_server.py
+++ b/examples/common/synchronous_server.py
@@ -19,7 +19,7 @@ from pymodbus.datastore import (
 # from pymodbus.server.sync import StartUdpServer
 # from pymodbus.server.sync import StartSerialServer
 from pymodbus.device import ModbusDeviceIdentification
-from pymodbus.server.sync import StartTcpServer
+from pymodbus.server.sync import StartUdpServer
 

     # ----------------------------------------------------------------------- #
     # run the server you want
     # ----------------------------------------------------------------------- #
     # Tcp:
-    StartTcpServer(context, identity=identity, address=("", 5020))
+    StartUdpServer(context, identity=identity, address=("", 5020))
     #
     # TCP with different framer
     # StartTcpServer(context, identity=identity,

````

janiversen, 2022-07-23:
Well it is an example, so there are no demand to have error detection, or did you mean something different ?

Eideen, 2022-07-23:
True the file is a example. The example uses a key function, `Modbus*Client`, that can call under sub function. First one in the example is `read_coils` (that inheritance ha lot of functions,  if did my `Pdb` correct it stops waits `return self.socket.recvfrom(size)[0]`). Since this functions expects a replay for the server, you can't build logic outside of it. 

Then am think that logic need to happen inside happen in `ModbusUdpClient` function. Or am i misunderstanding it?

janiversen, 2022-07-23:
I really do not follow you. The client waits until receives a response, the connection closes or timeout. The server does the same but without timeout.

Error handling are in general left outside the library which merely report errors (exceptions).

Eideen, 2022-07-23:
Yes and No.
Take TCP, if the lower stack is unable to establish connection, error is raced. If there is packet loss or TCP Fast Retransmission the packet will be resend.

UDP does not have this functionality and except high level (library) to handle unable to establish connection and packet loss. I believe this should be a part of the library, to some limited extend. there is code in `transaction.py`, to handle some errors, with a variable for `Retries` in `constants.py`. But there is no timeout. 
One can think of it as short timeout with `Retries` and long timeout, that library merely report errors (exceptions).

Any thing above this is to be handled outside of the library.

janiversen, 2022-07-23:
patches are welcome. But you are not totally correct, we have 2 tcp clients one is reconnecting.

in TCP resend is handled very limited in the socket library. In UDP there are no send failures (the definition is send without acknowledgment), please remember UDP is typically used to do broadcast while TCP is always a point to point connection.

janiversen, 2022-07-23:
And while I think of it, you cannot just blindly repeat the client call for a couple of reasons:
- it is unsafe to repeat a write (it might have severe consequences)
- The file transfer commands are multiple messages, and you are not allowed to repeat a single message
- UDP do have a connect (but a sendTo) since it is connectionless, so no connect repeat.

Eideen, 2022-07-23:
I wanted to make a contribute, with a simple server where the values are always changing, time and run counter. 

As much as I wanted to contribute on a lower level, it would be a long time before i am able to contribute do to the complexity of the project, and the time available to me. 

I have never used MODbus to do file transfer, when there is so many other better more purpose build protocols. 

In most case I know of the write command is idempotent. If there is scenarios where repeating a write have consequences, it something I would like to know more of.

janiversen, 2022-07-23:
just think of writing to a counter.

Your idea of an example where values are always changing is quite interesting, but making that should really be independent of udp/tcp/serial or tls.

Eideen, 2022-07-23:
My plan is bit of topic.

### For run counter:

Client -> modbus write (udp/tcp/serial or tls) -> Server -> Logic write to `Holding Registers`, copy it to  `Input Registers`
Client -> modbus read (udp/tcp/serial or tls) -> Server  `Input Registers`

For industry it used to make sure the remote PLC/RIO have its logic working. 

### For a timer:
it is similar but the `Input Registers` is write to by the server it self, something like `datetime.now()`

By the looks of it we don't have any example of Modbus server write to it self will running.

janiversen, 2022-07-23:
A lot of devices do not behave as you want....

The same with your earlier comment about file-transfer, if your device is a Modbus device you have to use Modbus file transfer.

Why would a Modbus server write to itself ???? the Modbus server is a slave (controlled from the client) and cannot itself add requests. May I politely suggest you read the Modbus documentation, it seems you missed some very basic concepts.

Eideen, 2022-07-23:
[image]
A very normal MODbus server that write to itself, is a RIO.
It reads inputs analog and digital, a write to the corresponding Registers. 

A other example is PLS, Where SCADA writes a run counter down, the PLS displays that is a live, by write to a `Input Registers`.

We have seen example of logic stopping, and TCP socket stays alive, and because of the logic of MODbus client reading TCP-connect = Remote logic alive, important information about the health condition of the server was hidden.

janiversen, 2022-07-23:
There are no such thing as a Modbus server that writes to itself !!

Please read the Modbus specification https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf and 
https://modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf
The simple definitions are: 
- Modbus server (normally a physical device) receives requests and send responses.
- Modbus client (normally a visualisation platform, e.g. SCADA) sends requests and receives responses.

As a consequence a RIO is a Modbus Server combined with a Modbus client (actually just like your drawing), they typically share the same address space, so there are NO special logic for read/write. Those exist as real devices especially for changing medium (e.g. using tcp communication upwards, and RS485 serial communicating downwards).

So please stop saying that a normal Modbus server can write to itself! or show me where the definitions allow that.

janiversen, 2022-07-23:
I have not heard the term PLS, I suppose you mean PLC, which is the term we use in the industry.

Eideen, 2022-07-24:
I think we are talk passed each other.

I am not talking about the Modbus protocol. A am talk about the system as a whole, where the MODbus server is a part of.

Lets define what a RIO is:

* Physical input and output, analog and digital.
* Network interface
* Management, nowdays a web server. 
* Logic to move information.
* MODbus server.

For a RIO they may share the memory space, or they a have a API.

| Modbus Server               | Shard memory space | Logic           |
|-----------------------------|--------------------|-----------------|
| Write/Read Coils            | ->                 | Write to output |
| Read Discrete Inputs        | <-                 | Read from input |
| Write/ReadHolding Registers | ->                 | Write to output |
| Read Input Registers        | <-                 | Read from input |

### What would this look link in a python program?
````python
#!/usr/bin/env python3
from calendar import month
from pymodbus.datastore import (    ModbusSequentialDataBlock,    ModbusServerContext,    ModbusSlaveContext,)
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.server.sync import StartUdpServer as MODbus_server
from pymodbus.version import version
from datetime import datetime
from time import sleep

UNIT = 0x1

identity = ModbusDeviceIdentification(
    info_name={
        "VendorName": "Pymodbus",
        "ProductCode": "PM",
        "VendorUrl": "https://github.com/riptideio/pymodbus/",
        "ProductName": "Pymodbus Server",
        "ModelName": "Pymodbus Server",
        "MajorMinorRevision": version.short(),
    }
)

### Server starts, and is forket
server = MODbus_server(identity=identity, address=("", 5020))

while True:
    ## Run counter
    
    ## Write content from Holding Register[1] to Input Registers[1]
    server.write_Input_Registers(1,server.Holding_Registers[1], unit=UNIT )
    
    ## write time
    time = datetime.now()
    server.write_Input_Registers(8, [
        time.year, 
        time.month, 
        time.day, 
        time.hour, 
        time.minute, 
        time.second, 
        round(time.microsecond/100), ### Size is to big for a 16 bit int 100 000 
        time.isoweekday() ], unit=UNIT )
    sleep(0.1)
````

### Comment to the program above.

[... 19 lines cut ...]
* `server.read_exception_status`                  Read the contents of eight Exception Status outputs in a local          device.
* `server.read_holding_registers`                 Read `count` number of holding registers starting at `address`.
* `server.read_input_registers`                   Read `count` number of input registers starting at `address`.
* `server.readwrite_registers`                    Read `read_count` number of holding registers starting at `read_address`  and write `write_registers`starting at `write_address`.
* `server.report_slave_id`                        Report information about local server ID.
* `server.return_bus_exception_error_count`       Diagnostic sub command, Return count of Modbus exceptions  returned by local server.
* `server.return_bus_message_count`               Diagnostic sub command, Return count of message detected on bus  by local server.
* `server.return_diagnostic_register`             Diagnostic sub command, Read 16-bit diagnostic register.
* `server.return_iop_overrun_count`               Diagnostic sub command, Return count of iop overrun errors  by local server.
* `server.return_query_data`                      Diagnostic sub command , Loop back data sent in response.
* `server.return_slave_bus_char_overrun_count`    Diagnostic sub command, Return count of messages not handled  by local server due to character overrun condition.
* `server.return_slave_busy_count`                Diagnostic sub command, Return count of server busy exceptions sent  by local server.
* `server.return_slave_message_count`             Diagnostic sub command, Return count of messages addressed to  local server.
* `server.return_slave_no_ack_count`              Diagnostic sub command, Return count of NO ACK exceptions sent by local server.
* `server.return_slave_no_response_count`         Diagnostic sub command, Return count of No responses  by local server.
* `server.silent_interval`                        Read Only!
* `server.state`                                  Read Only!
* `server.write_coil`                             Write `value` to coil at `address`.
* `server.write_coils`                            Write `value` to coil at `address`.
* `server.write_register`                         Write `value` to register at `address`.
* `server.write_registers`                        Write list of `values` to registers starting at `address`.

* `server.write_discrete_inputs`                   write `count` number of discrete inputs starting at offset `address`.
* `server.write_holding_registers`                 write `count` number of holding registers starting at `address`.
* `server.write_input_registers`                   write `count` number of input registers starting at `address`.

janiversen, 2022-07-24:
Feel free to submit a PR with the above, but this is not exactly the mainstream of modbus devices do not have configurable logic, but the sensors etc are hardwired to addresses. The idea that a write to a holding register is written to an input register is something I have not seen in the many years I have worked with SCADA systems. I really cannot see the reason, if you want you can just read the holding register, so in essence you loose a register.

We use the REPL server a lot for functional tests, but it would not hurt to expand that to something easier to use, but our focus is to mimic real life devices.

janiversen, 2022-07-24:
What would be nice to have is a RIO-Repeater, (a modbus server, with multiple clients), since it would allow users to install it on e.g. a raspberry PI and thus convert medium very cheap.

janiversen, 2022-07-24:
Feel free to open a discussion, if you have more questions or just wants comments on your upcoming PR. I am happy to help you on your quest (after all it seems we both are vikings).

As explained earlier due to the nature of the UDP protocol (multicast and no acknowledgement) error handling similar to TCP cannot be done, so I am closing this.

closed 2022-07-24 by janiversen