pymodbus-dev/pymodbus #981: Pymodbus 2.5.3 TypeError: SerialPort.__init__() got an unexpected keyword argument 'timeout'

willydwi77, 2022-07-26:
### Versions

* Python: 3.10.5 64-bit
* OS: Windows 11 Home 21H2 Build 22000.675
* Pymodbus: 2.5.3
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: rtu - async

### Description

When I run program I got this message "TypeError: SerialPort.__init__() got an unexpected keyword argument 'timeout'", yes I've read this issue from [here](https://github.com/riptideio/pymodbus/issues/580) but I've uses the latest version from pymodbus. Should I remove timeout from SerialPort in twisted library? Or maybe there's any solution for this.

### Code and Logs

Here my simple code, there's no different from pymodbus example

```python
#!/usr/bin/env python3

#from twisted.internet.task import LoopingCall
from threading import Thread
from time import sleep
from pymodbus.datastore import (
    ModbusSequentialDataBlock,
    ModbusServerContext,
    ModbusSlaveContext
)
from pymodbus.server.asynchronous import StartSerialServer

import logging

log = logging.getLogger()
log.setLevel(logging.DEBUG)

def updating_writer(extra):
     log.debug('updating the context')
     sleep(2)
     context = extra
     register = 8 # panjang data
     slave_id = 0x01 # slave id 0x01 = 1
     address = 0x0a # address 0x01 = 1
     values = context[slave_id].getValues(register, address, count=5)
     values = [v + 1 for v in values]
     log.debug(f"new values: { str(values) }")
     context[slave_id].setValues(register, address, values)

def run_server():
    store = ModbusSlaveContext(
        di = ModbusSequentialDataBlock(0x05, [0] * 100),
        co = ModbusSequentialDataBlock(0x00, [0] * 100),
        hr = ModbusSequentialDataBlock(0x01, [0] * 100),
        ir = ModbusSequentialDataBlock(0x0a, [0] * 100))
    context = ModbusServerContext(slaves=store, single=True)

    #time = 5 # 5 detik untuk delay
    #loop = LoopingCall(f = updating_writer, a = (context,))
    #loop.start(time, now = False) # inisialisasi delay dengan time

    Thread(target=updating_writer, args=(context,)).start()
    StartSerialServer(context, port='com2', timeout=1, baudrate=9600, autoreconnect=True)

if __name__ == "__main__":
    run_server()
```

Traceback
```s
Traceback (most recent call last):
  File "C:\Users\Lenovo\Desktop\PythonModbus\rasp-slave-gpio.py", line 57, in <module>
    run_server()
  File "C:\Users\Lenovo\Desktop\PythonModbus\rasp-slave-gpio.py", line 53, in run_server    
    StartSerialServer(context, port='com2', baudrate=9600, autoreconnect=True)
  File "C:\Users\Lenovo\Desktop\PythonModbus\lib\site-packages\pymodbus\server\asynchronous.py", line 338, in StartSerialServer
    SerialPort(protocol, port, reactor, baudrate=baudrate, parity=parity,
TypeError: SerialPort.__init__() got an unexpected keyword argument 'timeout'
```

janiversen, 2022-07-26:
For now please remove the timeout. It will be added again in the version 3 line, however the twisted library are gone (we only use asyncio).

willydwi77, 2022-07-26:
Okey sir, thanks. My code has fixed with asyncio module

closed 2022-07-26 by willydwi77