pymodbus-dev/pymodbus #987: AttributeError: 'ReadInputRegistersResponse' object has no attribute 'bits'

armenmolnar, 2022-07-27:
Somebody can help me ? What is the probelem?

from pymodbus.client.sync import ModbusTcpClient
client = ModbusTcpClient('192.168.0.200')
connection = client.connect()
print("Connected: " + str(connection))
while(True):
result=client.read_coils(0, True , unit=1)
if result.bits[0] == True :
client.write_coil(16, True , unit=1)
else:
client.write_coil(16, False , unit=1)

Error message:

AttributeError: 'ReadInputRegistersResponse' object has no attribute 'bits'

janiversen, 2022-07-27:
An input register is not a bit, but 16 bits.
```
    rr = client.read_input_registers(1, 8, unit=UNIT)
    assert not rr.isError()  # nosec test that we are not an error
    txt = f"### address 1 is: {str(rr.registers[0])}"
```
But your code example and the AttributeError do not match. ReadInputRegistersResponse comes from a read_input_registers call.

armenmolnar, 2022-07-27:
Can you write the correct code for me. Im a beginner! THX

janiversen, 2022-07-27:
There are a ton of examples in the repository, use them.

Apart from that, when you report error, please make sure the error match to the code.

closed 2022-07-27 by janiversen