pymodbus-dev/pymodbus #1011: TypeError: BaseModbusClient.__init__() got multiple values for argument 'framer'

gam-abbott, 2022-08-05:
* Python: 3.10.2
* OS: Windows 11
* Pymodbus: 3.0.0dev4
* Modbus Hardware (if used): usb to rs485

### Pymodbus Specific

* Client: rtu - sync

### Description
I am just trying to set the framer and it throws this error. Most of my pymodbus applications are through tcp but I am testing a new device and my modbus poller has no problems reading this device. My python script however is throwing this framer exception and I am not sure exactly why. Very simple piece of code.
-> TypeError: BaseModbusClient.__init__() got multiple values for argument 'framer'

### Code and Logs

from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.transaction import ModbusRtuFramer
client = ModbusClient(port="COM5",framer=ModbusRtuFramer,baudrate=9600)
rr = client.read_holding_registers(0,2,unit=2)
print(rr)
print(rr.registers)

janiversen, 2022-08-05:
I think you stumbled across an error I solved a while ago, problem is that "framer=" is present 2 times in **kwargs.

I recommend you try with current dev, as I am having problems getting the maintainer to make a new release candidate.

Btw. During the next week or so, there will be a lot of changes in the client part of the lib, because I am working on streamlining the interface as well as the code.

gam-abbott, 2022-08-05:
I figured as much. I was doing a primer on async and it forced me to update to 3.0.0dev4 or whatever and that solved that issue and now I am faced with this one. I am aware that there are many changes going on internally. Please let me know if there is anything I can do to help further this project because I am heavily dependent on it for work. Thank you.

janiversen, 2022-08-05:
Patches are welcome. We have open TODO lists, and there are open issues.

An easy start issue is #966

closed 2022-08-06 by janiversen