pymodbus-dev/pymodbus #1022: Does pymodbus support to lanuch multiple tcp client with the same ip and port to connect other modbus tcp server, like modbus poll simulator?

LinusLing0910, 2022-08-09:
Does pymodbus support to lanuch multiple tcp client with the same ip and port to connect other modbus tcp server, like modbus poll simulator?

janiversen, 2022-08-09:
yes, if you talk about port= which is the port the clients connect to.

You cannot have multiple client with the same source_address (host:port) because that is not possible with sockets.

LinusLing0910, 2022-08-09:
it there any other workaround to do like the modbus poll simulator to launch multiple client?
[image]

janiversen, 2022-08-09:
Why do you want to do a "workaround" ? Pymodbus support this out of the box. 

Just go ahead and write your code, with multiple client instances:
```
   client1 = ModbusTCPClient("host",port=502)
   client2 = ModbusTCPClient("host",port=502)
   client3 = ModbusTCPClient("host",port=502)
   client4 = ModbusTCPClient("host",port=502)
```

LinusLing0910, 2022-08-09:
@janiversen so pymodbus support the host with the same ip?

closed 2022-08-09 by LinusLing0910