pymodbus-dev/pymodbus #1048: Automate tests needs a solution for how to mock a tty port on all platforms.

janiversen, 2022-08-18:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

dhoomakethu, 2022-09-14:
There is no common solution for mocking serial ports on all 3 platforms (Linux, Windows and Mac). We can cover linux and mac with `socat` , there are unofficial ports of `socat` for windows. May be we can try that ?

janiversen, 2022-09-14:
I was thinking of mocking py serial but an unofficial so at seems better, where can I find that?

dhoomakethu, 2022-09-14:
https://github.com/StudioEtrange/socat-windows/blob/master/EXAMPLES

closed 2022-09-14 by janiversen