pymodbus-dev/pymodbus #1057: Multiple slave contexts seems broken

smurfix, 2022-08-21:
### Versions

* Python: 3.10
* OS: Debian testing
* Pymodbus: 3.0.0.dev4

### Description

I was adapting the provided forwarder example for serial/RTU to access multiple slaves.

### Code and Logs
```diff

--- examples/modbus_forwarder.py	2022-08-17 09:31:37.863088411 +0200
+++ /tmp/modbus_forwarder.py	2022-08-21 15:30:00.245183579 +0200
@@ -13,19 +13,30 @@
 import argparse
 import logging
 
-from pymodbus.client import ModbusTcpClient
+from pymodbus.client import ModbusSerialClient
 from pymodbus.server.sync import StartTcpServer
 from pymodbus.datastore import ModbusServerContext
 from pymodbus.datastore.remote import RemoteSlaveContext
+from pymodbus.transaction import ModbusRtuFramer
 
 
 def run_forwarder():
     """Run forwarder setup."""
     port_server, port_client, slaves = get_commandline()
 
-    client = ModbusTcpClient(
-        host="localhost",
+    client = ModbusSerialClient(
         port=port_client,
+        framer=ModbusRtuFramer,
+        timeout=2,
+        #    retries=3,
+        #    retry_on_empty=False,
+        #    close_comm_on_error=False,.
+        #    strict=True,
+        # Serial setup parameters
+        baudrate=9600,
+        bytesize=8,
+        parity="E",
+        stopbits=1,
     )
 
     # If required to communicate with a specified client use unit=<unit_id>
@@ -71,7 +82,7 @@
     parser.add_argument(
         "--port_client",
         help="the port to use",
-        type=int,
+        type=str,
     )
     parser.add_argument(
         "--slaves",
@@ -84,9 +95,9 @@
     # set defaults
     _logger.setLevel(args.log.upper() if args.log else logging.INFO)
     if not args.port:
-        args.port = 5020
+        args.port = 502
     if not args.port_client:
-        args.port_client = 5010
+        args.port_client = "/dev/ttyUSB0"
 
     return args.port, args.port_client, args.slaves
 
```
`python3 modbus_forwarder.py --slaves 1 --slaves 2 --log debug`

```python

15:20:08 DEBUG sync:79 Datastore unable to fulfill request: 'dict' object has no attribute 'validate' Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pymodbus/server/sync.py", line 70, in execute
    response = request.execute(context)
  File "/usr/lib/python3/dist-packages/pymodbus/register_read_message.py", line 190, in execute
    if not context.validate(self.function_code, self.address, self.count):
AttributeError: 'dict' object has no attribute 'validate'
```

Setting a breakpoint:
```
> /usr/lib/python3/dist-packages/pymodbus/register_read_message.py(191)execute()
-> if not context.validate(self.function_code, self.address, self.count):
(Pdb) pp context
{b'\x02': <pymodbus.datastore.remote.RemoteSlaveContext object at 0xb603e178>}
(Pdb) u
> /usr/lib/python3/dist-packages/pymodbus/server/sync.py(70)execute()
-> response = request.execute(context)
(Pdb) l
 65  	                # if broadcasting then execute on all slave contexts, note response will be ignored
 66  	                for unit_id in self.server.context.slaves():
 67  	                    response = request.execute(self.server.context[unit_id])
 68  	            else:
 69  	                context = self.server.context[request.unit_id]
 70  ->	                response = request.execute(context)
 71  	        except NoSuchSlaveException:
 72  	            txt = f"requested slave does not exist: {request.unit_id}"
 73  	            _logger.debug(txt)
 74  	            if self.server.ignore_missing_slaves:
 75  	                return  # the client will simply timeout waiting for a response
(Pdb) pp self.server.context
<pymodbus.datastore.context.ModbusServerContext object at 0xb603e0a0>
(Pdb) pp self.server.context[2]
{b'\x02': <pymodbus.datastore.remote.RemoteSlaveContext object at 0xb603e178>}
(Pdb) 
```

smurfix, 2022-08-21:
also happens with 2.5 apparently. If I'm doing something wrong then so does the example code :-P

janiversen, 2022-08-21:
The example was updated some weeks ago, please see current dev.

janiversen, 2022-08-25:
For some reason the example code works when I test it manually (we do not have automated testing of the forwarder), can you please check that you use the newest dev.

Btw. I like the idea of doing serial as well in the example, it would be nice to see a pull request, e.g. with an extra parameter to use client-tcp or client-serial.

janiversen, 2022-08-27:
Just tested your call:
```
python3 modbus_forwarder.py --slaves 1 --slaves 2 --log debug
```
it works on current dev, but you use the parameters in a non-intended way (--slaves takes an arrays of slaves).

```
python3 modbus_forwarder.py --slaves=1,2 --log=debug
```

closed 2022-08-27 by janiversen