pymodbus-dev/pymodbus #1058: assign identity for multiple slaves

aasick, 2022-08-22:
### Description

I am trying modbus server using 5 slaves. I am stuck on identity assigning to multiple slave.

Below code context contain 5 slave details. In that I need to set identity parameter for two power meter at the time of server running time only one slave able to set identity details.

### Code and Logs

```

def start_server(context, address, comport, baud_rate):
    update_time = 65

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Schneider'
    identity.ProductCode = 'A155'
    identity.MajorMinorRevision = '6.2'

    # identity.VendorName = 'Siemens'
    # identity.ProductCode = '67'
    # identity.MajorMinorRevision = '124'
    # identity.VendorUrl = 'www.Siemens.com.tr'
    # identity.ProductName = 'POWER'
    # identity.ModelName = '1298'

    server = ModbusSerialServer(context, framer=ModbusRtuFramer, identity=identity, port=comport, stopbits=1, bytesize=8,
                                parity='E', baudrate=baud_rate, timeout=1)

```
@dhoomakethu you have any suggestions, how to configure identity for multiple slaves

janiversen, 2022-08-22:
With the current dev you cannot have different identities. The identity structure is protected by a singleton, so the last change wins.

It is on my plan to change that, but it will take a while, sorry for that.

dhoomakethu, 2022-08-22:
If you are comfortable with pymodbus codebase and need a quick hack, you can try to move the `identity` to `Slave Context`, it is not straight forward and might break `ReadDeviceIdentification` and `ReportSlaveID` functions, so take it with a pinch of salt.

janiversen, 2022-08-22:
I just reread this issue, are you sure you should use a server and not a client ?

A client connects to slaves (or a server). A server receives requests from 1 client and executes them. A server cannot connect to slaves.

You can make a combination of server/slave, this is called a forwarder:
Server receives a request (from an external client) forward it to an internal client, that sends the request to a slave.
this is typically used to change transport medium (e.g. primary tcp and then change to serial).

my-life-in-codes, 2022-08-23:
@janiversen I am running as server continuously using loopingcall, whenever client request at the time server should serve information. here issue is if I make identity for two slaves only last identity reflect on pymodbus communication
```
server = ModbusSerialServer(context, framer=ModbusRtuFramer, identity=identity, port=comport, stopbits=1, bytesize=8,
                                parity='E', baudrate=baud_rate, timeout=1)

    threading.Thread(target=server.serve_forever, daemon=True).start()
    loop = LoopingCall(f=updating_writer, a=(context, address))
    loop.start(update_time, now=False)
    reactor.run()
```

janiversen, 2022-08-23:
yes an instance of the library can only have one identity, which actually makes sense since you only have one server.

closed 2022-08-27 by janiversen