pymodbus-dev/pymodbus #1060: Forwarder example rewrites calls to "write_register" to "read_holding_register" (func 06 to 03)

TheChatty, 2022-08-22:
### Versions

* Python: 3.10.6
* OS: Win11 x64
* Pymodbus: 3.0.0.dev4
* Modbus Hardware (if used): n/a

### Pymodbus Specific
* Server: definately tcp, maybe rtu/ascii - sync/async
* Client: definately tcp, maybe rtu/ascii - sync/async

### Description

When utilizing the [forwarder example](https://github.com/riptideio/pymodbus/blob/dev/examples/modbus_forwarder.py) with ModbusTcpClient all writes will be rewritten to reads.

### Code and Logs

```python
client = ModbusClient("modbus_forwarder", port=5020, timeout=10)
client.connect()

rr = client.read_holding_registers(0x90, 1, unit=247)
rr = client.write_register(0x90, 0x1234, unit=247)	# will be executed as the read above
```

Please don't remove the "read only" feature - it was helpful in a certain case. But please make it optional and documented.

janiversen, 2022-08-22:
I am not aware to have removed any “read only” parameter….where did you see that earlier?

Writes should not be reads let me test what happens. I am pretty sure this was not changed from the old example.

janiversen, 2022-08-22:
and please do remember that Pull Requests are welcome, that is the best way to get things changed.

TheChatty, 2022-08-22:
As much as I'd like to help... the inner workings of `DataStore` are way over my head.

janiversen, 2022-08-22:
you wrote "Please don't remove the "read only" feature" and I am still not aware of having removed that, I am also not aware of such a feature in the example.

TheChatty, 2022-08-22:
Well... the example is not talking about any rewrite feature. Even I assumed it would just forward the request as is.

Then I found out it does rewrite requests which I (formerly unknowingly) made use of.

So at first someone needs to find out why and where this happens. Then it should be fixed. But with the option to keep this behavior.

janiversen, 2022-08-22:
I think it states it pretty clear:
```
This is a repeater or converter and  an example of just how powerful datastore is.
```
Using a "datastore" means storing the data and thus rewrite the requests.

If is true that a request forwarder would be nice to have as example, and are in my plans. Actually it is quite easy to make using the "handle=" parameter in the server.

TheChatty, 2022-08-22:
It would be pretty "clear" if it would state it forwards in a read only fashion. Simple storing and fetching does in no way imply it will be changed in between. So could you please hint at where I have to look to make it forward **as is**?

janiversen, 2022-08-22:
Pull requests are welcome. And yes storing and fetching implies directly that the original request is not forwarded.

TheChatty, 2022-08-22:
Actually the usage of DataStore is not that obvious in the forwarder example (I know it's in the imports). But from what do you derive it is modifying the requests? And please guide me to make it not modify it. Once solved I'll create a PR for the forwarder example to make this the default behavior.

janiversen, 2022-08-25:
line 40:
```
        store = RemoteSlaveContext(client)
    context = ModbusServerContext(slaves=store, single=True)
```
Interconnect server datastore and client datastore.

janiversen, 2022-11-04:
Confirmed bug, it seems to be a more general problem.

PR #1166 (2022-11-05): Update datastore remote to handle write requests

fixes #1060
closed 2022-11-05 by janiversen