pymodbus-dev/pymodbus #1078: modbus_mapper.py doesn't work

pedrorovi, 2022-09-08:
### Versions

* Python: 3.10
* OS: Ubuntu 22.04.1 LTS
* Pymodbus: 3.0.0.dev4
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

I've been trying the modbus_mapper example (https://github.com/riptideio/pymodbus/blob/dev/examples/contrib/modbus_mapper.py) with a csv, but is no longer compatible with Python 3.
I've made some changes but I'm still having problems.
modbus_context_decoder function returns a dictionary inside the context where I think it should create a data store.
Has anyone experimented with this library?

I leave the link to my github repository, you can see the logs and the code here:
https://github.com/pedrorovi/pymodbus_mapper

### Code and Logs

Server logs:
```python
raw_mapping
defaultdict(<class 'dict'>,
            {'1': {'address': '1',
                   'description': 'Comm. count PLC',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'value': '1'}})
slave_context.store
{'c': <pymodbus.datastore.store.ModbusSequentialDataBlock object at 0x7fde0e856c50>,
 'd': <pymodbus.datastore.store.ModbusSequentialDataBlock object at 0x7fde0e856b00>,
 'h': {'1': '1'},
 'i': <pymodbus.datastore.store.ModbusSequentialDataBlock object at 0x7fde0e856e90>}
2022-09-08 13:35:23,592 DEBUG    sync           :372      Started thread to serve client at ('127.0.0.1', 46073)
2022-09-08 13:35:23,593 DEBUG    sync           :44       Client Connected [('127.0.0.1', 46073)]
2022-09-08 13:35:23,593 DEBUG    sync           :202      Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x1
2022-09-08 13:35:23,593 DEBUG    socket_framer  :165      Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x1
2022-09-08 13:35:23,593 DEBUG    factory        :216      Factory Request[ReadHoldingRegistersRequest': 3]
2022-09-08 13:35:23,593 DEBUG    context        :64       validate: fc-[3] address-1: count-1
2022-09-08 13:35:23,594 DEBUG    sync           :78       Datastore unable to fulfill request: 'dict' object has no attribute 'validate' Traceback (most recent call last):
  File "/home/pedro/.local/lib/python3.10/site-packages/pymodbus/server/sync.py", line 69, in execute
    response = request.execute(context)
  File "/home/pedro/.local/lib/python3.10/site-packages/pymodbus/register_read_message.py", line 129, in execute
    if not context.validate(self.function_code, self.address, self.count):
  File "/home/pedro/.local/lib/python3.10/site-packages/pymodbus/datastore/context.py", line 65, in validate
    return self.store[self.decode(fc_as_hex)].validate(address, count)
AttributeError: 'dict' object has no attribute 'validate'

2022-09-08 13:35:23,594 ERROR    pdu            :108      Exception Response(131, 3, SlaveFailure)
2022-09-08 13:35:23,594 DEBUG    sync           :236      send: [Exception Response(131, 3, SlaveFailure)]- b'000100000003008304'
2022-09-08 13:35:23,649 DEBUG    sync           :202      Handling data: 
2022-09-08 13:35:23,650 DEBUG    socket_framer  :165      Processing: 
2022-09-08 13:35:23,650 DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 46073)]

```

Client logs:
```python

defaultdict(<class 'dict'>,
            {'1': {'address': '1',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'size': '2',
                   'type': 'int16'}})
None
2022-09-08 13:35:23,592 DEBUG    sync           :218      Connection to Modbus server established. Socket ('127.0.0.1', 46073)
2022-09-08 13:35:23,592 DEBUG    transaction    :127      Current transaction state - IDLE
2022-09-08 13:35:23,592 DEBUG    transaction    :131      Running transaction 1
2022-09-08 13:35:23,592 DEBUG    transaction    :283      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x1
2022-09-08 13:35:23,592 DEBUG    sync           :77       New Transaction state "SENDING"
2022-09-08 13:35:23,592 DEBUG    transaction    :304      Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2022-09-08 13:35:23,594 DEBUG    transaction    :408      Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2022-09-08 13:35:23,594 DEBUG    transaction    :319      RECV: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x4
2022-09-08 13:35:23,594 DEBUG    socket_framer  :165      Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x4
2022-09-08 13:35:23,594 DEBUG    factory        :347      Factory Response[131]
2022-09-08 13:35:23,594 DEBUG    transaction    :495      Adding transaction 1
2022-09-08 13:35:23,594 DEBUG    transaction    :507      Getting transaction 1
2022-09-08 13:35:23,594 DEBUG    transaction    :229      Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
<pymodbus.pdu.ExceptionResponse object at 0x7f3ca7cb31f0>
Traceback (most recent call last):
  File "/home/pedro/Pedro/Projects/pymodbus_mapper/src/client.py", line 30, in <module>
    response.registers, byteorder=Endian.Big, wordorder=Endian.Big
AttributeError: 'ExceptionResponse' object has no attribute 'registers

```

janiversen, 2022-09-08:
It is one of the examples that needs an update. 
The 
```
AttributeError: 'ExceptionResponse' object has no attribute 'registers
```
Comes the server sends the exception message and not a real response, this is something the client need to check for.

If you have suggested changed, then please make a pull request so that we can review it, and make it available to everyone.

To be honest modbus_mapper.py is in the contrib section (added by developers who use Pymodbus) and it will take a while before we get around to updating that. That being said if you make a Pull Request with your changes, chances are good that one of us maintainers will review it, and see if together we can update it.

janiversen, 2022-09-08:
I know there have been quite a number of updates on dev, and I remember a couple of them relating to

```
AttributeError: 'dict' object has no attribute 'validate'
```

so part of your problem might be solved using current dev (3.0.0dev5 is expected soon).

pedrorovi, 2022-09-08:
We have made a pull request with our changes.

Logs server:
```
raw_mapping
defaultdict(<class 'dict'>,
            {'1': {'address': '1',
                   'description': 'Comm. count PLC',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'value': '100'},
             '2': {'address': '2',
                   'description': 'Comm. count PLC',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'value': '200'}})
slave_context.store
{'c': <pymodbus.datastore.store.ModbusSequentialDataBlock object at 0x7f245daf8220>,
 'd': <pymodbus.datastore.store.ModbusSequentialDataBlock object at 0x7f245daf8670>,
 'h': <pymodbus.datastore.store.ModbusSparseDataBlock object at 0x7f245daf8340>,
 'i': <pymodbus.datastore.store.ModbusSequentialDataBlock object at 0x7f245d490d90>}
2022-09-08 17:26:24,097 DEBUG    sync           :372      Started thread to serve client at ('127.0.0.1', 37189)
2022-09-08 17:26:24,098 DEBUG    sync           :44       Client Connected [('127.0.0.1', 37189)]
2022-09-08 17:26:24,098 DEBUG    sync           :202      Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x2
2022-09-08 17:26:24,098 DEBUG    socket_framer  :165      Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x2
2022-09-08 17:26:24,098 DEBUG    factory        :216      Factory Request[ReadHoldingRegistersRequest': 3]
2022-09-08 17:26:24,098 DEBUG    context        :64       validate: fc-[3] address-1: count-2
2022-09-08 17:26:24,098 DEBUG    context        :78       getValues: fc-[3] address-1: count-2
2022-09-08 17:26:24,099 DEBUG    sync           :236      send: [ReadHoldingRegistersResponse (2)]- b'000100000007000304006400c8'
2022-09-08 17:26:24,103 DEBUG    sync           :202      Handling data: 
2022-09-08 17:26:24,103 DEBUG    socket_framer  :165      Processing: 
2022-09-08 17:26:24,103 DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 37189)]
2022-09-08 17:27:07,408 DEBUG    sync           :372      Started thread to serve client at ('127.0.0.1', 43727)
2022-09-08 17:27:07,408 DEBUG    sync           :44       Client Connected [('127.0.0.1', 43727)]
2022-09-08 17:27:07,408 DEBUG    sync           :202      Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x2
2022-09-08 17:27:07,408 DEBUG    socket_framer  :165      Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x2
2022-09-08 17:27:07,408 DEBUG    factory        :216      Factory Request[ReadHoldingRegistersRequest': 3]
2022-09-08 17:27:07,408 DEBUG    context        :64       validate: fc-[3] address-1: count-2
2022-09-08 17:27:07,408 DEBUG    context        :78       getValues: fc-[3] address-1: count-2
2022-09-08 17:27:07,409 DEBUG    sync           :236      send: [ReadHoldingRegistersResponse (2)]- b'000100000007000304006400c8'
2022-09-08 17:27:07,413 DEBUG    sync           :202      Handling data: 
2022-09-08 17:27:07,413 DEBUG    socket_framer  :165      Processing: 
2022-09-08 17:27:07,413 DEBUG    sync           :52       Client Disconnected [('127.0.0.1', 43727)]

´´´

Logs client: 
```
defaultdict(<class 'dict'>,
            {'1': {'address': '1',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'size': '2',
                   'type': 'int16'},
             '2': {'address': '2',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'size': '2',
                   'type': 'int16'}})
defaultdict(<class 'dict'>,
            {'1': {'address': '1',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'size': '2',
                   'type': <function ModbusTypeDecoder.parse_16bit_int at 0x7fca6bdb0c10>},
             '2': {'address': '2',
                   'function': 'hr',
                   'name': 'Comm. count PLC',
                   'size': '2',
                   'type': <function ModbusTypeDecoder.parse_16bit_int at 0x7fca6bdb0c10>}})
HR
2022-09-08 17:27:07,408 DEBUG    sync           :218      Connection to Modbus server established. Socket ('127.0.0.1', 43727)
2022-09-08 17:27:07,408 DEBUG    transaction    :127      Current transaction state - IDLE
2022-09-08 17:27:07,408 DEBUG    transaction    :131      Running transaction 1
2022-09-08 17:27:07,408 DEBUG    transaction    :283      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x0 0x0 0x2
2022-09-08 17:27:07,408 DEBUG    sync           :77       New Transaction state "SENDING"
2022-09-08 17:27:07,408 DEBUG    transaction    :304      Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2022-09-08 17:27:07,409 DEBUG    transaction    :408      Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2022-09-08 17:27:07,409 DEBUG    transaction    :319      RECV: 0x0 0x1 0x0 0x0 0x0 0x7 0x0 0x3 0x4 0x0 0x64 0x0 0xc8
2022-09-08 17:27:07,409 DEBUG    socket_framer  :165      Processing: 0x0 0x1 0x0 0x0 0x0 0x7 0x0 0x3 0x4 0x0 0x64 0x0 0xc8
2022-09-08 17:27:07,409 DEBUG    factory        :347      Factory Response[ReadHoldingRegistersResponse': 3]
2022-09-08 17:27:07,409 DEBUG    transaction    :495      Adding transaction 1
2022-09-08 17:27:07,409 DEBUG    transaction    :507      Getting transaction 1
2022-09-08 17:27:07,409 DEBUG    transaction    :229      Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
<pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7fca6b487610>
2022-09-08 17:27:07,409 DEBUG    payload        :299      [100, 200]
[1]	100
[2]	200

´´´

closed 2022-09-14 by dhoomakethu