pymodbus-dev/pymodbus #1087: ModbusRtuFramer object is not callable

thesems, 2022-09-20:
### Versions

* Python: 3.10
* OS: Ubuntu 22.04
* Pymodbus: 3.0.0.dev5 9bc2fb0 (latest dev branch commit)
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: rtu async
* Client: rtu async

### Description

I'm trying to run serial rtu async client with the code below, however I'm running into some issues.
Directly after calling the .connnect() function, I receive that ModbusRtuFramer' is not callable.
The other error "int has not len()" is received after calling read_holding_registers.

What am I missing?

Thank you.

### Code and Logs
```
13:10:52 WARNING serial:111 Failed to connect: 'ModbusRtuFramer' object is not callable
```

```
File "/home/-/projects/-/src/main_meter_3.py", line 17, in <module>
    asyncio.run(run(client), debug=True)
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/-/projects/-/src/main_meter_3.py", line 12, in run
    await run_async_client(client)
  File "/home/-/projects/-/src/main_meter_3.py", line 6, in run_async_client
    rr = await client.read_holding_registers(1, 1, unit=1) # this has been changed
  File "/home/-/.cache/pypoetry/virtualenvs/--XI0Oy1ie-py3.10/lib/python3.10/site-packages/pymodbus/client/mixin.py", line 118, in read_holding_registers
    return self.execute(request)
  File "/home/-/.cache/pypoetry/virtualenvs/--XI0Oy1ie-py3.10/lib/python3.10/site-packages/pymodbus/client/base.py", line 179, in execute
    return self.transaction.execute(request)
  File "/home/-/.cache/pypoetry/virtualenvs/--XI0Oy1ie-py3.10/lib/python3.10/site-packages/pymodbus/transaction.py", line 168, in execute
    response, last_exception = self._transact(
  File "/home/-/.cache/pypoetry/virtualenvs/--XI0Oy1ie-py3.10/lib/python3.10/site-packages/pymodbus/transaction.py", line 323, in _transact
    result = self._recv(response_length, full)
  File "/home/-/.cache/pypoetry/virtualenvs/--XI0Oy1ie-py3.10/lib/python3.10/site-packages/pymodbus/transaction.py", line 363, in _recv
    if len(read_min) != min_size:
TypeError: object of type 'int' has no len()
```

```python
from pymodbus.client import AsyncModbusSerialClient
import asyncio

async def run_async_client(client):
    rr = await client.read_holding_registers(1, 1, unit=1)
    assert rr.registers[0] == 10  # test the expected value

async def run(client):
    await client.connect()
    await run_async_client(client)

if __name__ == "__main__":
    client = AsyncModbusSerialClient(port="/dev/ttyUSB0", baudrate=115200)
    asyncio.run(run(client), debug=True)
```

janiversen, 2022-09-20:
Sounds like 2 real problem, let me have a look.

thesems, 2022-09-20:
First error has been solved, however second seems to be a separate error and remains the same.

Weirdly, recvPacket() calls the ModbusBaseClient recv(), which returns an int, instead of the derived Serial.

janiversen, 2022-09-20:
The second issue is not as easy to solve, I have just compared to older versions, and do not understand with recvPacket is called at all. I will need to make some test cases before I can solve this, so it might take some time.

dhoomakethu, 2022-09-21:
The `recv` method is not overrided for `AsyncModbusSerialClient` , it should be similar to sync client `recv` refer https://github.com/riptideio/pymodbus/blob/dev/pymodbus/client/serial.py#L293

janiversen, 2022-09-21:
@dhoomakethu that does not sound right, when I go back to v2.5.3 AsyncModbusSerialClient does not have a receive nor can I find it in other commits.

In async normally data_received is called and that starts calling the framer and transaction. For some reason here recvPacket() in transaction is called, as if it is the other way round. I am pretty sure the problem is something different, but look forward to hear why you think recv() is missing.

dhoomakethu, 2022-09-21:
@janiversen the older version of async client had its own `execute` method Refer https://github.com/riptideio/pymodbus/blob/2781ca5f3d48dae22be8afe4d12712082f782f93/pymodbus/client/base.py#L281 which would eventually write over the transport https://github.com/riptideio/pymodbus/blob/2781ca5f3d48dae22be8afe4d12712082f782f93/pymodbus/client/base.py#L376

With the new changes, the execute is being called from the Base class which defaults to the execute method from transaction.py which is usually used for serial clients.

I mentioned its missing `recvPacket` implementation on an assumption that you have rewritten the library to be uniform across sync and async clients. Apologies if that caused confusion.

janiversen, 2022-09-21:
That rewrite is still sitting in my local repo (dev_client_sync) and waiting that I solve the outstanding server problems first.

closed 2022-09-21 by janiversen

dhoomakethu, 2022-09-21:
@janiversen @thesems you change   `await run_async_client(client)` to `await run_async_client(client.protocol)` to get this working. Not an issue with the codebase.

thesems, 2022-09-21:
@janiversen @dhoomakethu thank you for your rash assistance. It is working correctly with the latest commits. It's possible poetry has been caching an older version, though I'm not sure.

PR #1090 (2022-09-21): Solve example server startup and fix #1087

When running the serial async example (server/client) everything works as documented:
```
./server_async.py --comm serial --port /dev/ttys004
13:06:09 INFO  server_async:71 ### Create datastore
13:06:09 INFO  server_async:158 ### start ASYNC server on port /dev/ttys004
13:06:54 ERROR async_io:100 Handler for serial port has been cancelled
```
and
```
./client_async_basic_calls.py --comm serial --port /dev/ttys005
13:06:38 INFO  client_async:52 ### Create client object
13:06:38 INFO  client_async:129 ### Client starting
13:06:38 INFO  serial:115 Seria [... cut]