pymodbus-dev/pymodbus #1100: pymodbus 3.0.0dev5 serial_asyncio moduleNotFound

tomdelahaba, 2022-10-01:
* Python: Python 3.10
* OS: MacOS X 12.6 (Monterey)
* Pymodbus: 3.0.0dev5
* Modbus Hardware (if used): none yet

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

Trying to run the example code from:
https://pymodbus.readthedocs.io/en/dev/source/example/client_sync.html

I have a file main.py:
```
import client_sync

print('### STARTING APPLICATION')

client_sync()
```
which is running the client_sync which is copy&paste code from the example above...

I get error:
```
Traceback (most recent call last):
  File "python-project/main.py", line 1, in <module>
    import client_sync
  File "python-project/client_sync.py", line 30, in <module>
    from pymodbus.client import (
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pymodbus/client/__init__.py", line 5, in <module>
    from pymodbus.client.serial import AsyncModbusSerialClient, ModbusSerialClient
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pymodbus/client/serial.py", line 7, in <module>
    from serial_asyncio import create_serial_connection
ModuleNotFoundError: No module named 'serial_asyncio'
```

janiversen, 2022-10-01:
Seems you have not run the “pip install” as documented, that installs pymodbus and it’s dependencies, one of them being serial_asyncio.

janiversen, 2022-10-01:
see requirements.txt.

tomdelahaba, 2022-10-01:
Thanks a lot, I have installed it from the package manager (pypi) so the -r option was not possible. I had to install that 
 pyserial and pyserial-asyncio manually and it seems I moved a little bit with this one. That was exactly what I was looking for (the requirements file, I missed that somehow before).
 
 Thanks a lot

closed 2022-10-01 by tomdelahaba