pymodbus-dev/pymodbus #1104: About serial data sticking package while using modbus rtu

Perry19920630, 2022-10-09:
### Versions

* Python:3.8.10
* OS:ARM64
* Pymodbus:2.5.3
* Modbus Hardware (if used): n/a

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
# ----------------------------------------------------------------------- #
  # Public Member Functions
  # ----------------------------------------------------------------------- #
  def processIncomingPacket(
      self, data, callback, unit, **kwargs
  ):  # pylint: disable=arguments-differ
      """Process new packet pattern.
      This takes in a new request packet, adds it to the current
      packet stream, and performs framing on it. That is, checks
      for complete messages, and once found, will process all that
      exist.  This handles the case when we read N + 1 or 1 // N
      messages at a time instead of 1.
      The processed and decoded messages are pushed to the callback
      function to process and send.
      :param data: The new packet data
      :param callback: The function to send results to
      :param unit: Process if unit id matches, ignore otherwise (could be a
             list of unit ids (server) or single unit id(client/server)
      :param kwargs:
      """
      if not isinstance(unit, (list, tuple)):
          unit = [unit]
      self.addToFrame(data)
      single = kwargs.get("single", False)

      while True: 
          if self.isFrameReady():
              if self.checkFrame():
                  if self._validate_unit_id(unit, single):
                      self._process(callback)
                  else:
                      header_txt = self._header["uid"]
                      txt = f"Not a valid unit id - {header_txt}, ignoring!!"
                      _logger.debug(txt)
                      self.resetFrame()
              else:
                  _logger.debug("Frame check failed, ignoring!!")
                  self.resetFrame()
          else:
              txt = f"Frame - [{data}] not ready"
              _logger.debug(txt)
              break
```

janiversen, 2022-10-09:
I have no idea what your problem is ?

Please try with v3.0.0dev5, this part have had several changes. You can also wait about 1 week and 3.0.0dev5 will be released as 3.0.0.

Perry19920630, 2022-10-09:
> I have no idea what your problem is ?
> 
> Please try with v3.0.0dev5, this part have had several changes. You can also wait about 1 week and 3.0.0dev5 will be released as 3.0.0.

closed 2022-10-09 by Perry19920630

Perry19920630, 2022-10-09:
While received a few frames in one package,only the first frame can be treated by callback at once,and the other frames can not be treated until the next frame is received.This will disorder the communication sequence,so I share my correction above that.

janiversen, 2022-10-09:
It is always nice to see enhancements to the library, but please make a pull request based on the current dev.

Perry19920630, 2022-10-09:
ok,thank you anyway.