pymodbus-dev/pymodbus #1112: pyserial, pyserial-asyncio should be optional in pymodbus 3

chrullrich, 2022-10-13:
### Versions

* Pymodbus: 3.0.0

### Description

pyserial and pyserial-asyncio are listed as optional dependencies, but importing pymodbus.client fails if they are not present.

### Code and Logs

```
>>> from pymodbus.client import ModbusUdpClient
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/pymodbus/pymodbus/client/__init__.py", line 5, in <module>
    from pymodbus.client.serial import AsyncModbusSerialClient, ModbusSerialClient
  File "/pymodbus/pymodbus/client/serial.py", line 7, in <module>
    from serial_asyncio import create_serial_connection
ModuleNotFoundError: No module named 'serial_asyncio'
```

janiversen, 2022-10-13:
Ups, that is a real bug, will fix that

Jonamaita, 2022-10-13:
> Ups, that is a real bug, will fix that

Thank you!!! 

The import of `ModbusSerialClient` module  in `pydmodbus 2.5.3` was done:

`from pymodbus.client.sync import ModbusSerialClient`

To `Pymodbus: 3.0.0`  it was changed too?.

janiversen, 2022-10-14:
Look at the documentation or examples.

PR #1114 (2022-10-14): Make serial import dynamic

fixes #1112
closed 2022-10-14 by janiversen