pymodbus-dev/pymodbus #1118: Repl server startup bug

jugge83, 2022-10-17:
### Versions

* Python: 3.10.8
* OS: win 10
* Pymodbus: 3.0.0
* Typer: 0.6.1

### Pymodbus Specific

* pymodbus. server (repl)

### Description

I was trying to start the repl server with "pymodbus. server run" and got the following error message: 
12:36:59 ERROR main:373 Invalid server ModbusServerTypes.tcp

### Code and Logs

I managed to start the server by adding .value on the enum objects "ModbusServerTypes.tcp" and "ModbusFramerTypes.socket.value" in pymodbus/repl/server/main.py.

The working code snippet looks like this :

```python
# pymodbus/repl/server/main.py
# line 117

@app.command("run", context_settings=CONTEXT_SETTING)
def run(
    ctx: typer.Context,
    modbus_server: str = typer.Option(
        ModbusServerTypes.tcp.value,
        "--modbus-server",
        "-s",
        case_sensitive=False,
        autocompletion=servers,
        help="Modbus Server",
    ),
    modbus_framer: str = typer.Option(
        ModbusFramerTypes.socket.value,
        "--framer",
        "-f",
        case_sensitive=False,
        autocompletion=framers,
        help="Modbus framer to use",
    ),
```

klasrocket, 2022-10-17:
I experienced the same using Python 3.10.6
Got it working with the same workaround, thx!

janiversen, 2022-10-20:
As far as I can see this is not limited to python 3.10 but occurs in earlier versions as well.

serroba, 2022-10-26:
# Versions
* Python 3.9.6
* macOS 12.5.1

# Commits

The latest known tag that the server REPL was working fine is `v3.0.0dev3`

## Proof

```
➜  pymodbus git:(018d618) git checkout v3.0.0dev3
Previous HEAD position was 018d618 Misc updates
HEAD is now at d92516d Bump version to 3.0.0dev3 and update CHANGELOG
➜  pymodbus git:(d92516d) pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///Users/sebastianmachuca/projects/pymodbus
  Preparing metadata (setup.py) ... done
Requirement already satisfied: six>=1.15.0 in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/site-packages (from pymodbus==3.0.0.dev3) (1.15.0)
Requirement already satisfied: pyserial>=3.4 in /Users/sebastianmachuca/Library/Python/3.9/lib/python/site-packages (from pymodbus==3.0.0.dev3) (3.5)
Installing collected packages: pymodbus
  Attempting uninstall: pymodbus
    Found existing installation: pymodbus 3.0.0.dev2
    Uninstalling pymodbus-3.0.0.dev2:
      Successfully uninstalled pymodbus-3.0.0.dev2
  Running setup.py develop for pymodbus
Successfully installed pymodbus
➜  pymodbus git:(d92516d) pymodbus.server run --modbus-server tcp --modbus-framer socket --modbus-unit-id 1 --modbus-unit-id 4 --randomize 1 --modbus-port 5020

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/
                                                                                                1.0.0

SERVER > hekp
Invalid command or invalid usage of command - ['hekp']
SERVER > help
Available commands:
clear                                        Clears screen
manipulator                                  Manipulate response from server.
Usage: manipulator response_type=|normal|error|delayed|empty|stray
	Additional parameters
		error_code=<int>
		delay_by=<in seconds>
		clear_after=<clear after n messages int>
		data_len=<length of stray data (int)>

	Example usage:
	1. Send error response 3 for 4 requests
	   manipulator response_type=error error_code=3 clear_after=4
	2. Delay outgoing response by 5 seconds indefinitely
	   manipulator response_type=delayed delay_by=5
	3. Send empty response
	   manipulator response_type=empty
	4. Send stray response of lenght 12 and revert to normal after 2 responses
	   manipulator response_type=stray data_len=11 clear_after=2
	5. To disable response manipulation
	   manipulator response_type=normal
SERVER >
Bye Bye!!!
^C
Aborted!
```

On the next tag, `v3.0.0dev4` we get this:

```
➜  pymodbus git:(d92516d) git checkout v3.0.0dev4
Previous HEAD position was d92516d Bump version to 3.0.0dev3 and update CHANGELOG
HEAD is now at 95e9581 Merge branch 'setupcfg-updates' into dev
➜  pymodbus git:(95e9581) pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///Users/sebastianmachuca/projects/pymodbus
  Preparing metadata (setup.py) ... done
Installing collected packages: pymodbus
  Attempting uninstall: pymodbus
    Found existing installation: pymodbus 3.0.0.dev3
    Uninstalling pymodbus-3.0.0.dev3:
      Successfully uninstalled pymodbus-3.0.0.dev3
  Running setup.py develop for pymodbus
Successfully installed pymodbus-3.0.0.dev4
➜  pymodbus git:(95e9581) pymodbus.server run --modbus-server tcp --modbus-framer socket --modbus-unit-id 1 --modbus-unit-id 4 --randomize 1 --modbus-port 5020

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \\
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/

SERVER > help
Available commands:
{cmd:45s}{hlp:100s}
{cmd:45s}{hlp:100s}
SERVER >
Bye Bye!!!
^C
Aborted!
```

# GitHub 2 dots comparison

The diff is rather large between the 2 hashes: https://github.com/riptideio/pymodbus/compare/d92516d..95e9581

Hard to point what actually changed

janiversen, 2022-10-26:
Did you try with the release 3.0.0 (above you compare dev3 and dev4) ?

There are some known problems with REPL, which I think @dhoomakethu is working on solving.

serroba, 2022-10-26:
@janiversen we still have errors in 3.0.0

```
➜  pymodbus git:(dev) git checkout v3.0.0
Note: switching to 'v3.0.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at d7834ac Merge branch 'dev' to define v3.0.0
➜  pymodbus git:(d7834ac) pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///Users/sebastianmachuca/projects/pymodbus
  Preparing metadata (setup.py) ... done
Installing collected packages: pymodbus
  Attempting uninstall: pymodbus
    Found existing installation: pymodbus 2.5.3
    Uninstalling pymodbus-2.5.3:
      Successfully uninstalled pymodbus-2.5.3
  Running setup.py develop for pymodbus
Successfully installed pymodbus-3.0.0rc1
➜  pymodbus git:(d7834ac) pymodbus.server --no-repl run
Traceback (most recent call last):
  File "/usr/local/bin/pymodbus.server", line 8, in <module>
    sys.exit(server())
TypeError: server() missing 1 required positional argument: 'ctx'
➜  pymodbus git:(d7834ac) pymodbus.server run
Traceback (most recent call last):
  File "/usr/local/bin/pymodbus.server", line 8, in <module>
    sys.exit(server())
TypeError: server() missing 1 required positional argument: 'ctx'
```

janiversen, 2022-10-26:
ok just wanted to be sure. Let’s wait and see what @dhoomakethu says.

janiversen, 2022-10-26:
Did you check if your local file was updated ?
File "/usr/local/bin/pymodbus.server", line 8, in <module>
    sys.exit(server())

Normally we do not install in /user/local/bin however I am not expert on REPL

dhoomakethu, 2022-10-26:
@serroba  Please give it a try with this PR https://github.com/riptideio/pymodbus/pull/1141 based out of this branch `repl-enhancements-2`

dhoomakethu, 2022-10-26:
```
pymodbus on  repl-enhancements-2 [$?] via 🐍 v3.8.13 (pymodbus) took 2s
❯ pymodbus.server run --modbus-server tcp --framer socket --unit-id 1 --unit-id 4 --random 2 --modbus-port 5020

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \\
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/

SERVER > help
Available commands:
clear                                        Clears screen
manipulator                                  Manipulate response from server.
Usage: manipulator response_type=|normal|error|delayed|empty|stray
	Additional parameters
		error_code=<int>
		delay_by=<in seconds>
		clear_after=<clear after n messages int>
		data_len=<length of stray data (int)>

	Example usage:
	1. Send error response 3 for 4 requests
	   manipulator response_type=error error_code=3 clear_after=4
	2. Delay outgoing response by 5 seconds indefinitely
	   manipulator response_type=delayed delay_by=5
	3. Send empty response
	   manipulator response_type=empty
	4. Send stray response of length 12 and revert to normal after 2 responses
	   manipulator response_type=stray data_len=11 clear_after=2
	5. To disable response manipulation
	   manipulator response_type=normal
SERVER >
```

@serroba  please give it a try now with `repl-enhancements-2`.

serroba, 2022-10-26:
@dhoomakethu I am not sure if I am doing something wrong, but I am still seeing an error. Maybe I am missing an step. This is what I did:

```
➜  pymodbus git:(dev) git pull
remote: Enumerating objects: 64, done.
remote: Counting objects: 100% (64/64), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 64 (delta 41), reused 58 (delta 37), pack-reused 0
Unpacking objects: 100% (64/64), 18.63 KiB | 323.00 KiB/s, done.
From github.com:riptideio/pymodbus
 * [new branch]      dev_datastore_json  -> origin/dev_datastore_json
 * [new branch]      fix-#1118           -> origin/fix-#1118
 * [new branch]      repl-enhancements-2 -> origin/repl-enhancements-2
Already up to date.
➜  pymodbus git:(dev) git checkout repl-enhancements-2
branch 'repl-enhancements-2' set up to track 'origin/repl-enhancements-2'.
Switched to a new branch 'repl-enhancements-2'
➜  pymodbus git:(repl-enhancements-2) git log
➜  pymodbus git:(repl-enhancements-2) pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///Users/sebastianmachuca/projects/pymodbus
  Preparing metadata (setup.py) ... done
Installing collected packages: pymodbus
  Attempting uninstall: pymodbus
    Found existing installation: pymodbus 3.0.0.dev3
    Uninstalling pymodbus-3.0.0.dev3:
      Successfully uninstalled pymodbus-3.0.0.dev3
  Running setup.py develop for pymodbus
Successfully installed pymodbus-3.0.1
➜  pymodbus git:(repl-enhancements-2) pymodbus.server --no-repl run --modbus-server tcp --modbus-framer socket --modbus-unit-id 1 --modbus-unit-id 4 --randomize 1 --modbus-port 5020

Traceback (most recent call last):
  File "/usr/local/bin/pymodbus.server", line 8, in <module>
    sys.exit(server())
TypeError: server() missing 1 required positional argument: 'ctx'
```

I am not very well familiar with python/pip so maybe I am doing something wrong or missing an step

serroba, 2022-10-26:
also as a REPL

```
pymodbus.server run --modbus-server tcp --framer socket --unit-id 1 --unit-id 4 --random 2 --modbus-port 5020
Traceback (most recent call last):
  File "/usr/local/bin/pymodbus.server", line 8, in <module>
    sys.exit(server())
TypeError: server() missing 1 required positional argument: 'ctx'
```

janiversen, 2022-10-26:
I think you need to remove the /usr/local/bin/pymodbus.server or at the very least verify it is renewed.

serroba, 2022-10-26:
Indeed, this seems to fix things:
after removing the file
[image]

serroba, 2022-10-27:
@dhoomakethu although the help menu is back the functionality is not.

When I run the headless server, and try to connect, this is what I get:

[image]

However, when I run the sync server example, things work as expected:

[image]

dhoomakethu, 2022-10-27:
@serroba Can you run with `--verbose` option please?

dhoomakethu, 2022-10-27:
@serroba Can you run with `--verbose` option for the server please?

dhoomakethu, 2022-10-27:
Also can you share the docker command used to run server inside the container? Please try without docker on the same host as well

serroba, 2022-10-27:
@dhoomakethu I used this Dockerfile to build the image: https://github.com/riptideio/pymodbus/pull/1145/files

To build the image, I run this command:

```
docker build --tag serroba/pymodbus:repl-enhancements-2 .
```

I get the very same error with and without docker. This is in my host (OSX):

[image]

When I run in verbose mode I get this:

[image]

```
2022-10-27 13:31:47,054 MainThread      ERROR    tcp            :191      Connection to (localhost, 5020) failed: [Errno 61] Connection refused
```

dhoomakethu, 2022-10-27:
Just to make sure, the issue is only with non repl mode, am I correct with my assumption?

I will take a look in sometime. Thanks for the report

serroba, 2022-10-27:
@dhoomakethu I get the same error also in the REPL. I can, on the other hand, run the `examples/server_sync.py` and things work fine there.

dhoomakethu, 2022-10-27:
I just gave this a try on my local setup (Mac OSX), seems to be working fine with repl mode
[image]
Can confirm, the non repl mode is not working
[image]

serroba, 2022-10-28:
@dhoomakethu I can also confirm. I have pull latest changes, create a new image, test via docker. Things are working as expected :)

commit 30f71669 (2022-11-02): Repl enhancements 2 (#1141)

* Fix #1118 for python 3.9 and above

* Solve isort.

* Support randomizing reads for discrete inputs and input registers

* Fix pylint error for zero comparision

* Fix server help commands

* Fix Reactive server not starting in non-repl mode

* Update REPL server documentation

* Codespell fix
closed 2022-11-02 by janiversen