pymodbus-dev/pymodbus #1126: PyModbus 3.0 module is not found - although the module has been installed

Jmmx1237, 2022-10-21:
### Versions

* Python: 3.9
* OS: Rasbian Os (bullseye)
* Pymodbus: 3.0
* Modbus Hardware (if used): Wago 750

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

On my new hardware with the latest Linux Rasbian OS (Bullseye), my Pymodbus script no longer runs. Somehow, I can not get the PyModbus 3.0 module installed correctly.

On my old system, my script runs with an older version of Pymodbus without any problems:

# My Old system
pip3 show pymodbus
Name: pymodbus
Version: 2.5.3
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/riptideio/pymodbus/
Author: Galen Collins
Author-email: bashwork@gmail.com
License: BSD-3-Clause
Location: /home/philo/.local/lib/python3.8/site-packages
Requires: six, pyserial
Required-by:

How do I get this to work, should I maybe downgrade my python version?

### Error

~/python-skripte $ python3 websocketserver3.py
Traceback (most recent call last):
  File "/home/philo/python-skripte/websocketserver3.py", line 9, in <module>
    from pymodbus.client.sync import ModbusTcpClient as ModbusClient
ModuleNotFoundError: No module named 'pymodbus.client.sync'

~/python-skripte $ sudo python3 websocketserver3.py
Traceback (most recent call last):
  File "/home/philo/python-skripte/websocketserver3.py", line 9, in <module>
    from pymodbus.client.sync import ModbusTcpClient as ModbusClient
  File "/usr/local/lib/python3.9/dist-packages/pymodbus/client/__init__.py", line 5, in <module>
    from pymodbus.client.serial import AsyncModbusSerialClient, ModbusSerialClient
  File "/usr/local/lib/python3.9/dist-packages/pymodbus/client/serial.py", line 7, in <module>
    from serial_asyncio import create_serial_connection
ModuleNotFoundError: No module named 'serial_asyncio'

janiversen, 2022-10-21:
Please read the documentation !

Pymodbus.client.sync no longer exist it is now Pymodbus.client.

You also have not installed serial_asyncio.

There are (as noted) a number of important changes between 2.5.3 and 3.0.0 (as signalled by the change 2 -> 3) and you need you need to modify your script and environment accordingly.

Jmmx1237, 2022-10-21:
Thank you very much, my script was modified and seems now to run perfectly... serial_asyncio I haven't reinstalled it now, I don't use asyncio in my script either, at least it wasn't imported.

janiversen, 2022-10-21:
There is a known bug in 3.0.0 that demands serial_asyncio to be installed, but if your scripts works then its fine.

closed 2022-10-21 by janiversen