pymodbus-dev/pymodbus #1128: Runtime is dependant on pyserial, even when using TCP/IP

alexrudd2, 2022-10-21:
### Versions

* Python: 3.9 or 3.10
* OS: MacOSX
* Pymodbus: 3.0.0
* Modbus Hardware (if used): N/A

### Pymodbus Specific
* Client: tcp async

### Description

pymodbus `ReconnectingAsyncioModbusTcpClient` fails if `pyserial` is not installed.  This is listed as optional in `requirements.txt`
https://github.com/riptideio/pymodbus/blob/b4f5a558254f8242fd65fb49fe10f4f255610b32/requirements.txt#L31-L33

### Code and Logs

```
pip uninstall pyserial
pip install pymodbus==3.0.0

python3
>    from pymodbus.client import AsyncModbusTcpClient
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/client/__init__.py", line 5, in <module>
    from pymodbus.client.serial import AsyncModbusSerialClient, ModbusSerialClient
  File "/opt/homebrew/lib/python3.9/site-packages/pymodbus/client/serial.py", line 7, in <module>
    from serial_asyncio import create_serial_connection
  File "/opt/homebrew/lib/python3.9/site-packages/serial_asyncio/__init__.py", line 21, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
```

janiversen, 2022-10-21:
Correct, this was reported earlier and is solved on dev. We expect to release 3.0.1 next month with this and other bugs solved..

https://github.com/riptideio/pymodbus/commit/7e58d51d1adb431f7f4d110815235af69b8e977d

alexrudd2, 2022-10-21:
Aha, I was in the process of submitting a PR but you are too fast for me :)

This was how I had solved it, but `dev` works fine for me.
```
from pymodbus.client.base import ModbusBaseClient
try:
    from pymodbus.client.serial import AsyncModbusSerialClient, ModbusSerialClient
except ModuleNotFoundError:
    pass
```

closed 2022-10-21 by alexrudd2

janiversen, 2022-10-21:
Heh, I did that PR a couple of days ago...but please keep submitting PR´s, in open source 1+1 = 3 is really true.

alexrudd2, 2022-10-21:
I've spent this week updating a number of drivers we ([numat](https://github.com/numat)) maintain.  So far 3.0.x appears to be working well after the port.

* https://github.com/numat/productivity/pull/42
* [numat/clickplc](https://github.com/numat/clickplc)
* [numat/huber](https://github.com/numat/huber)
* [numat/midas](https://github.com/numat/midas)

Thanks for your work!   We've been anticipating this release for a few years now :)

janiversen, 2022-10-21:
I too !

I am a member of home assistant where Pymodbus is the centre of the modbus integration, so very soon I will upgrade HA.

My plan currently is to provide a monthly maintenance release for a couple of months until it is really stable.