pymodbus-dev/pymodbus #1140: modbus RTU response interpretation

patfrench, 2022-10-24:
### Versions

* Python: 3.7.13
* OS: windows
* Pymodbus: 2.5.3
* Modbus Hardware (if used):

### Pymodbus Specific
* Client: tcp/rtu/ascii - sync/async

### Description

Hello,
I have a particular case: a big memory ! 

### Code and Logs
This is my table modbus (example) 
```
1000 |      | LiveNumeric | 24 byte

1000 | float32_t | Data1 | 4
1002 | float32_t | Data2 | 4
1004 | uint16_t  | Data3 | 2
1005 | uint16_t  | Data4 | 2
1006 | int16_t   | Data5 | 2
1007 | uint16_t  | Data6 | 2
1008 | uint16_t  | Data7 | 2
1009 | uint16_t  | Data8 | 6
```

### Question
Today with "modbus_mapper" I can read without problem but it will be too long because there are too many addresses. I would like to read 24 bytes in one reading. 
Is there a tool to decode a 24 bytes answer with a template (same as modbus_mapper) ?

Thanks in advance

janiversen, 2022-10-24:
If you talk about reading the csv file, then you can simply adapt the modbus_mapper to read 24 bytes and put them in the right place.

patfrench, 2022-10-25:
Hi
yes, that's exactly it: I can't reading (too long)
```
0x1 0x3 0x3 0xe8 0x0 0x2 0x44 0x7b = read 2 word at address 1000 (0x03E8)
0x1 0x3 0x3 0xea 0x0 0x1 0xa5 0xba = read 1 word at address 1002 (0x03EA)
0x1 0x3 0x3 0xeb 0x0 0x1 0xf4 0x7a = read 1 word at address 1003 (0x03EB)
```
with 
```
rr = client.read_holding_registers(1000, 12, unit=1)
0x1 0x3 0x3 0xe8 0x0 0xc 0xc5 0xbf = read 12 word at start address 1000
```
Perfect but now I have to find a way to decode the answers (depending on the type) 
```CSV
address,type,size,name,function
1000,float32,2,Data1,hr
1002,uint16,1,Data2,hr
1003,uint16,1,Data3,hr
1004,uint16,1,Data4,hr
1005,uint16,1,Data5,hr
1006,float32,2,Data6,hr
1008,uint16,4,Data7,hr
```
```Python
rr = client.read_holding_registers(1000, 12, unit=1)
print(rr.registers)
[0, 0, 17408, 0, 0, 0, 0, 0, 0, 0, 0, 0]
``` 
I will try to adapt modbus_mapper

Thank you

janiversen, 2022-10-25:
look at the payload module it has all the functions you want.

patfrench, 2022-10-25:
I'm not a great programmer but 

```Python
response = client.read_holding_registers(address=1000, count=12, unit=1)
decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Little, wordorder=Endian.Little)
print(decoder.decode_32bit_float())
print(decoder.decode_16bit_uint())
print(decoder.decode_16bit_uint())
print(decoder.decode_16bit_uint())
print(decoder.decode_16bit_uint())
print(decoder.decode_16bit_uint())
print(decoder.decode_32bit_float())
print(decoder.decode_16bit_uint())
```
```
0.0
68
0
0
0
0
0.0
0
```
Now I have to find a way to use a CSV for decoding
a CSV to read the registers and a CSV to decode

Thank you

closed 2022-10-28 by patfrench