pymodbus-dev/pymodbus #1146: Custom Response Class registration lost due to ModbusFramer being inialized multiple times

wlcrs, 2022-10-27:
### Versions

* Python: 3.10.2
* OS: Ubuntu 20.04
* Pymodbus: 3.0.0
* Modbus Hardware (if used):

### Pymodbus Specific
* Client: tcp async

### Description

I'm trying to upgrade to 3.0.0, but run into trouble with registering my custom response class.

I do this like this:
```py
client = AsyncModbusTcpClient(host,   port,     framer=ModbusSocketFramer)
client.register(PrivateHuaweiModbusResponse)   
```

This correctly adds the response to the `client.framer.decoder.__lookup` table

However, upon callling `client.connect()`, I see that in `AsyncModbusTcpClient._create_protocol`, it does not pass `client.framer` to the ModbusClientProtocol, but instead the `self.params.framer`:

```py
       protocol = ModbusClientProtocol(framer=self.params.framer, **self.params.kwargs)
 ```

In the protocol-object, the framer is initialized again, without the custom response class this time.

I'm very confused about how both `AsyncModbusTcpClient` and `ModbusClientProtocol` have `ModbusBaseClient` as a parent class. Would it not be more appropriate to remove `ModbusBaseClient` as a parent class from `ModbusClientProtocol` and pass the client-object creating the `ModbusClientProtocol` to it instead?

janiversen, 2022-10-27:
We are moving in the direction of having the code common between sync and async thus the need for a common base class.

Seems however you found a real bug, I will add a test case and solve it.

janiversen, 2022-10-30:
3.0.1 is due beginning November and will include the fix. It would be nice if you could test it before then.

HomeAssistant modbus integration is also due to be bumped to use the new library.

closed 2022-10-30 by janiversen

wlcrs, 2022-10-30:
I was able to successfully complete the migration of my huawei-solar library to pyModbus 3.x. Thank you for the code cleanup. The API looks much nicer now. 👍

I'll release a new version of my HA integration when 3.0.1 is released.

PR #1153 (2022-10-30): Do not initialize framer twice

fixes #1146