pymodbus-dev/pymodbus #1155: tiemout parameter can not be changed. Always takes 3.0 seconds

bsense-rius, 2022-10-31:
### Versions

* Python: 3.8
* OS: ubuntu 20.04
* Pymodbus: 2.5.3
* Modbus Hardware (if used): SHT20 Temperature Sensor

### Pymodbus Specific
* Server: N/A
* Client: serial rtu-async

### Description
No matter the timeout I set, it always takes 3.0 seconds when the modbus peripheral has an issue.

What were you trying, what has happened, what went wrong, and what did you expect?
The sensor fails from time to time. Most of the time is ok, but sometimes is not processing modbus requests. So we are aware of this issue, but we expected the pymodbus library timeout in the given timeout parameter, But it does not. Always takes 3.0 seconds

### Code and Logs

```pythonloop, client =  AsyncModbusSerialClient(schedulers.ASYNC_IO, port=self.commPort, baudrate=self.commSpeed,
                                                            method=self.modbusMode, timeout=self.commTimeout, loop=fakeLoop)
# code and logs here.
loop, client =  AsyncModbusSerialClient(schedulers.ASYNC_IO, port=self.commPort, baudrate=self.commSpeed,
                                                            method=self.modbusMode, timeout=self.commTimeout, loop=fakeLoop)

# please use the following to format logs when posting them here
import pymodbus
pymodbus.pymodbus_apply_logging_config()
```

janiversen, 2022-10-31:
You need to supply a log otherwise we cannot see what happens.

Did you try e.g. with timeout=10 to check if 3 is just your minimum.

Which values did you try less than 3, supply logs please.

janiversen, 2022-10-31:
And I just saw you are using 2.5.3, that version have many known issues with timeout especially in async, please upgrade to 3.0.0

GraciousGpal, 2022-11-02:
This is present in 3.0, I supplied timeout of 30 to  the AsyncModbusTcpClient and it still timeout at 3.0 seconds

janiversen, 2022-11-03:
Thanks for the research, I will take a look soon and expect to solve it before 3.0.1 is released.

PR #1161 (2022-11-03): Transfer parms. to connected client

fixes #1155
closed 2022-11-03 by janiversen