pymodbus-dev/pymodbus #1156: 'ModbusSingleRequestHandler' object is not callable

ElTarget, 2022-10-31:
import os
import logging

from pymodbus.framer.rtu_framer import ModbusRtuFramer
from pymodbus.server.async_io import StartTcpServer, StartSerialServer
from pymodbus.datastore import (
    ModbusSequentialDataBlock,
    ModbusSlaveContext,
    ModbusServerContext,
)

# 创建log对象，log对象为pymodbus
pymodbus_logger = logging.getLogger("pymodbus")
# 设置log记录等级，log等级可以参考
# logging.CRITICAL
# logging.ERROR
# logging.WARNING
# logging.INFO
# logging.DEBUG
pymodbus_logger.setLevel(logging.DEBUG)

if __name__ == "__main__":
    # 设置log记录格式：log时间-log发生对象-log等级-log打印信息
    fmt = logging.Formatter("%(asctime)s - %(name)s - %(levelname)s - %(message)s")
    # 创建log文件对象，文件创建在当前目录，文件名为slave.log
    file_handler = logging.FileHandler(os.getcwd() + "/slave_server.log")
    # 设置文件对象记录log的等级
    file_handler.setLevel(logging.DEBUG)
    # 设置文件对象log的格式
    file_handler.setFormatter(fmt)
    # 添加log文件对象到当前log系统中，这样产生log后，log会被同步写入到文件中
    pymodbus_logger.addHandler(file_handler)

    # 定义co线圈寄存器，存储起始地址为0，长度为20，内容为15个True及5个False
    co_block = ModbusSequentialDataBlock(0, [True] * 15 + [False] * 5)
    # 定义di离散输入寄存器，存储起始地址为0，长度为20，内容为15个True及5个False
    di_block = ModbusSequentialDataBlock(0, [True] * 15 + [False] * 5)
    # 定义ir输入寄存器，存储起始地址为0，长度为10，内容为0~10递增数值列表
    ir_block = ModbusSequentialDataBlock(0, [i for i in range(10)])
    # 定义hr保持寄存器，存储起始地址为0，长度为10，内容为0~10递增数值列表
    hr_block = ModbusSequentialDataBlock(0, [i for i in range(10)])

    # 创建从机，从机的di离散量、co线圈、hr保持寄存器、ir输入寄存器等由上面定义并传入
    slaves = ModbusSlaveContext(di=di_block, co=co_block, hr=hr_block, ir=ir_block)
    # 创建单从机上下文，交由服务器调度
    context = ModbusServerContext(slaves=slaves, single=True)

    # 如果需要创建多个从机，参考如下：从机地址 + 从机配置
    # slaves = {
    # 1:ModbusSlaveContext(di=di_block, co=co_block, hr=hr_block, ir=ir_block),
    # 2:ModbusSlaveContext(di=di_block, co=co_block, hr=hr_block, ir=ir_block),
    # }
    # context = ModbusServerContext(slaves=slaves, single=False)

    # 开启tcp服务器方法
    # kwargs = {'context': context, 'address': ('127.0.0.1', 5020)}
    # StartTcpServer(**kwargs)
    # 开启串行设备服务器方法
    # 参数：从机上下文、从机通行帧格式、监听设备'/dev/ttymxc1'、串口波特率、从机监听超时时间0为不阻塞立刻响应
    StartSerialServer(context=context, framer=ModbusRtuFramer, port="COM9")
Python ：3.9.13
Pymodbus : 3.0
This is the case code obtained from the document. After the library version is upgraded, some codes have been modified. The RTU mode cannot be enabled. Only TCP can be started.

ElTarget, 2022-10-31:
``import os
import logging

from pymodbus.framer.rtu_framer import ModbusRtuFramer
from pymodbus.server.async_io import StartTcpServer, StartSerialServer
from pymodbus.datastore import (
    ModbusSequentialDataBlock,
    ModbusSlaveContext,
    ModbusServerContext,
)

# 创建log对象，log对象为pymodbus
pymodbus_logger = logging.getLogger("pymodbus")
# 设置log记录等级，log等级可以参考
# logging.CRITICAL
# logging.ERROR
# logging.WARNING
# logging.INFO
# logging.DEBUG
pymodbus_logger.setLevel(logging.DEBUG)

if __name__ == "__main__":
    # 设置log记录格式：log时间-log发生对象-log等级-log打印信息
    fmt = logging.Formatter("%(asctime)s - %(name)s - %(levelname)s - %(message)s")
    # 创建log文件对象，文件创建在当前目录，文件名为slave.log
    file_handler = logging.FileHandler(os.getcwd() + "/slave_server.log")
    # 设置文件对象记录log的等级
    file_handler.setLevel(logging.DEBUG)
    # 设置文件对象log的格式
    file_handler.setFormatter(fmt)
    # 添加log文件对象到当前log系统中，这样产生log后，log会被同步写入到文件中
    pymodbus_logger.addHandler(file_handler)

    # 定义co线圈寄存器，存储起始地址为0，长度为20，内容为15个True及5个False
    co_block = ModbusSequentialDataBlock(0, [True] * 15 + [False] * 5)
    # 定义di离散输入寄存器，存储起始地址为0，长度为20，内容为15个True及5个False
    di_block = ModbusSequentialDataBlock(0, [True] * 15 + [False] * 5)
    # 定义ir输入寄存器，存储起始地址为0，长度为10，内容为0~10递增数值列表
    ir_block = ModbusSequentialDataBlock(0, [i for i in range(10)])
    # 定义hr保持寄存器，存储起始地址为0，长度为10，内容为0~10递增数值列表
    hr_block = ModbusSequentialDataBlock(0, [i for i in range(10)])

    # 创建从机，从机的di离散量、co线圈、hr保持寄存器、ir输入寄存器等由上面定义并传入
    slaves = ModbusSlaveContext(di=di_block, co=co_block, hr=hr_block, ir=ir_block)
    # 创建单从机上下文，交由服务器调度
    context = ModbusServerContext(slaves=slaves, single=True)

    # 如果需要创建多个从机，参考如下：从机地址 + 从机配置
    # slaves = {
    # 1:ModbusSlaveContext(di=di_block, co=co_block, hr=hr_block, ir=ir_block),
    # 2:ModbusSlaveContext(di=di_block, co=co_block, hr=hr_block, ir=ir_block),
    # }
    # context = ModbusServerContext(slaves=slaves, single=False)

    # 开启tcp服务器方法
    # kwargs = {'context': context, 'address': ('127.0.0.1', 5020)}
    # StartTcpServer(**kwargs)
    # 开启串行设备服务器方法
    # 参数：从机上下文、从机通行帧格式、监听设备'/dev/ttymxc1'、串口波特率、从机监听超时时间0为不阻塞立刻响应
    StartSerialServer(context=context, framer=ModbusRtuFramer, port="COM9")``

janiversen, 2022-10-31:
Which example are you using ? only the examples directly in examples/*py have been updated, as noted in the readme.

janiversen, 2022-10-31:
If you look in test/test_examples.py you will see that we test all upgraded examples automatically to ensure they work.

Pull requests are welcome, if you would like to update the example.

ElTarget, 2022-10-31:
> If you look in test/test_examples.py you will see that we test all upgraded examples automatically to ensure they work.
> 
> Pull requests are welcome, if you would like to update the example.

From “ https://pymodbus.readthedocs.io/en/latest/source/example/modules.html ” get, but I haven't found any related ones so far. I'm going to try to use your new code as a test service.
And is there a specific example of the file transfer function? Is the file transferred to the corresponding address after reading?

janiversen, 2022-10-31:
but modules.html is a list of all examples, and not a specific example.

I do not recall a specific file transfer example, but have a look...if not pull requests are welcome.

What is the corresponding address ? I do not understand your question.

ElTarget, 2022-10-31:
1.Thank you. I'm going to look at the sample code, but it seems that the data is transferred from the terminal.
2.Read a file in byte mode, slice it, take 2 bytes as a segment of data, and transfer it to the 00001 register, and the second segment to the 00002 register. And so on, until the storage ends.

janiversen, 2022-10-31:
I was not aware that we have any example code that reads from the terminal.

What you describe in 2. is NOT file transfer. File transfer is a specific modbus command, that takes a transfer address and a byte array. Special with this command is that you can transfer in segments (e.g. 1000 bytes in each transfer).

Reading a file and sending it to the modbus device as single register write is not going to work....different registers needs different commands (holding, coils etc.).

if you talk about setting up a datastore, then reading from a file is very easy to do. just load the file as a byte array and define the datastore.

janiversen, 2022-10-31:
I am actually working on a datastore that reads from a json file, see #1157, this will be part of the 3.0.1 release.

ElTarget, 2022-10-31:
OK, I'll check it.

janiversen, 2022-10-31:
Closing as the issue seems solved.

closed 2022-10-31 by janiversen