pymodbus-dev/pymodbus #1169: Undeclared dependency on pygments

chrullrich, 2022-11-05:
### Versions

* Pymodbus: 3.0.1

### Description

`pymodbus/client/main.py` imports `pygments`, but this dependency is not declared.

### Code and Logs

```
$ pymodbus.console
Traceback (most recent call last):
  File "/usr/local/bin/pymodbus.console", line 33, in <module>
    sys.exit(load_entry_point('pymodbus==3.0.1', 'console_scripts', 'pymodbus.console')())
  File "/usr/local/bin/pymodbus.console", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/local/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/local/lib/python3.10/site-packages/pymodbus/repl/client/main.py", line 29, in <module>
    from pygments.lexers.python import PythonLexer
ModuleNotFoundError: No module named 'pygments'
```

janiversen, 2022-11-05:
There are no pymodbus/client/main.py in our repo, so I am quite confused ????

Apart from that the error message is pymodbus/repl/client/main.py which also do not import pygments, see:
https://github.com/riptideio/pymodbus/blob/dev/pymodbus/repl/client/main.py

so I think you have a problem outside pymodbus.

Apart from that please do not use 3.0.1 it is faulty and will be replaced by 3.0.2 in a couple of hours.

chrullrich, 2022-11-05:
Yes, the first file name was a typo. The line in question is https://github.com/riptideio/pymodbus/blob/dev/pymodbus/repl/client/main.py#L29

janiversen, 2022-11-05:
@dhoomakethu: do we have a problem ?

Seems to me like it is actually missing in requirements.txt.

chrullrich, 2022-11-05:
FYI, it's there if pip is involved because it is a transitive dependency:

    Added pygments<3.0.0,>=2.6.0 from https://... (from rich<13.0.0,>=10.11.0->typer[all]>=0.6.1->pymodbus[repl])

janiversen, 2022-11-05:
I am not sure what you mean, it is added to requirements.txt as a fixed requirement for spell, which I am in the process of releasing as 3.0.2

janiversen, 2022-11-05:
fixed in #1170

closed 2022-11-05 by janiversen

chrullrich, 2022-11-05:
My comment was about the requirements.txt as of 3.0.1. Your commit happened while I was digging through the output of `pip install -vv pymodbus[repl]` looking for the dependency chain.

The reason this missing dependency was never noticed before is that when installing pymodbus using pip, pymodbus depends on typer depends on rich depends on pygments, and so it was always present if the [repl] feature was selected (and not needed otherwise).

"spell" above is autocorrectese for repl, I suspect.

dhoomakethu, 2022-11-06:
@chrullrich is this an issue even installing with `pymodbus[repl]` ? Or was this discovered during some other import?

chrullrich, 2022-11-06:
It was the transitive dependency that, until yesterday's commit, prevented the issue when installing pymodbus[repl] via pip or another package manager that implements PEP 508, or 440, or whatever it is this week, so, no, installing pymodbus via pip (except with `--no-deps`) could never have ended up with pygments missing.

I found the problem because I was using a different package manager (creating a FreeBSD port, actually) where Python's dependency declarations do not apply. When I created the package dependencies for that port, I used the requirements.txt (that did not include pygments at that time), and consequently `pymodbus.console` greeted me with the ImportError.

dhoomakethu, 2022-11-07:
Thanks for the confirmation @chrullrich.