pymodbus-dev/pymodbus #1193: Handling synchronous access to hardware - Understanding question -

Jmmx1237, 2022-11-25:
Can accesses be made simultaneously via synchronous pymodbus if one access is read and the other is both read or write? More specifically, I would like to know if this is supported by pymodbus in principle or if it is only limited by the hardware?

For example, I have two threads in which the hardware is accessed. One thread can write and read and the other is only used for reading.

Can it happen that both threads access a register or coil at the same time? For example, while one thread is still writing to an address, the other thread is reading, purely from the code logic. My understanding is that threads are never executed at the same time, but serially, right? So am I on the safe side with my concept or should I consider asyncio?

The only problem is, asyncio is very sophisticated for me as a beginner, which is why I have hopes that my theory will be confirmed here. I have also researched and read that threading for I/O works but is not the perfect method. Even if the issue section is for something other than my comprehension question, I apologise.

janiversen, 2022-11-25:
This is generally not supported. But use async and it works very well.

janiversen, 2022-11-25:
The reason for not supporting this is that the modbus protocol itself does not really support it, nor do the I/O of python.

With async you do not need to make threads with all the sync. problems.

Jmmx1237, 2022-11-25:
Thank you for the explanation, that means threading.Threads are not asynchronous by definition?Thank you for the explanation, that means threading.Threads are not asynchronous by definition?

> With async you do not need to make threads with all the sync. problems.

I have tried to search for it explicitly, but I can't find any examples. Could you perhaps give me a short example where async problems occur when you only thread?

In my first attempts with threading, it seems at least that two functions are executed asynchronously when they are called from two different threads.

Thanks again and sorry for all the questions, I just want to understand in detail. After that I will be quiet^^

janiversen, 2022-11-25:
Threading is sync. Task is async. have a look at our examples. With threading and the modbus protocol you are bound to have problems or a lot of io locks. With async the parallelism is given.

janiversen, 2022-11-25:
Closing this, since it does not seem to be an issue with pymodbus.

closed 2022-11-25 by janiversen