pymodbus-dev/pymodbus #1194: After the client is closed, the function will be executed again.

ElTarget, 2022-11-26:
[image]

janiversen, 2022-11-26:
Please add a proper debug log and tell us which version etc. you are using, as per documentation. I really cannot see what is happening.

You might refer to a wrong socket, it could be client.protocol.socket.

ElTarget, 2022-11-26:
[image]
My Python 3.9 and pymodbus are 3.02.
I use ModbusSerialClient:
self. client = ModbusSerialClient(……)
Close using self.client.close()
But use self.client.close(), I can still use the related function functions.

ElTarget, 2022-11-26:
> 

[image]
The code can be this view.

janiversen, 2022-11-26:
Yes because pymodbus detects you are not connected and reconnect automatically. It seems you did not send the full debug log, because you should have seen both the close and the reconnect.

If you do not want to communicate anymore delete the client object (or let garbage collect do it)....calling close() and then continue making function calls is not something we consider good programming.

janiversen, 2022-11-26:
Actually what you see, is a feature for many, because when a connection breaks it automatically reopens.

ElTarget, 2022-11-26:
> Yes because pymodbus detects you are not connected and reconnect automatically. It seems you did not send the full debug log, because you should have seen both the close and the reconnect.
>
> If you [...]

Because I am cooperating with QT in the test. After clicking Disconnect, the program on another page is still sending the read message. So there will be the current situation.

janiversen, 2022-11-26:
Well then you need to delete the client object or have a semaphore that tells if the connection is open.

We might make client.close() stricter in a future version, but that will only lead to an exception in the following calls.

ElTarget, 2022-11-26:
> Actually what you see, is a feature for many, because when a connection breaks it automatically reopens.

So I may need to add a judgment on whether the modbus is connected during the execution of related threads.

ElTarget, 2022-11-26:
Yes, thank you very much!

janiversen, 2022-11-26:
No, you need to add a judgement if your application have called client.close() or if your application still consider communication valid.

ElTarget, 2022-11-26:
> 不，如果您的应用程序调用了 client.close（） 或者您的应用程序仍然认为通信有效，则需要添加判断。

> No, you need to add a judgement if your application have called client.close() or if your application still consider communication valid.

Yes.

closed 2022-11-26 by ElTarget