pymodbus-dev/pymodbus #1202: Server thread not closing

HamishSwanbarton, 2022-11-28:
### Versions

* Python: 3.10
* OS: Windows
* Pymodbus: Dev branch, Commit "Fix #1188 and some pylint issues (#1189)" 46473ffa
* Modbus Hardware (if used): None

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

When running a Modbus TCP server in a thread, the thread does not consistently exit after `ServerStop()` is called. This is in a unit test where both client and server are run. A workaround appears to be connecting a client and closing it, immediately before the `ServerStop()` call occurs, however if there is any (substantial) code in between `client.close()` and `ServerStop()` the thread remains active.

To reproduce: use the test in test/test_examples.py and run this test: test_exp_sync_server_client

Putting a `print(threading.enumerate())` shows only the main thread active at the end (good), however adding a `time.sleep(1)` just before the `ServerStop()` call causes 3 threads to be printed out

### Code and Logs

```python
# Test looks like this: (I changed the port offset to 0 so that it worked locally)
@pytest.mark.parametrize("test_port_offset", [0])
@pytest.mark.parametrize("test_comm, test_framer, test_port", [TEST_COMMS_FRAMER[0]])
def test_exp_sync_server_client(
    test_comm,
    test_framer,
    test_port_offset,
    test_port,
):
    """Run sync client and server."""
    print("start threads:", threading.enumerate())
    args = Commandline.copy()
    args.comm = test_comm
    args.port = test_port + test_port_offset
    args.framer = test_framer
    run_args = setup_server(args)
    thread = Thread(target=run_sync_server, args=(run_args,))
    thread.daemon = True
    thread.start()
    sleep(1)
    test_client = setup_sync_client(args=args)
    run_sync_client(test_client, modbus_calls=run_sync_calls)
    time.sleep(1)
    ServerStop()
    print("end threads", threading.enumerate())
```
```
# Logs:
============================= test session starts =============================
collecting ... collected 1 item

test_examples.py::test_exp_sync_server_client[tcp-ModbusSocketFramer-5020-0] 

============================== 1 passed in 5.24s ==============================

Process finished with exit code 0
PASSED [100%]start threads: [<_MainThread(MainThread, started 17248)>]
end threads [<_MainThread(MainThread, started 17248)>, <Thread(Thread-1 (run_sync_server), started daemon 16032)>, <Thread(asyncio_0, started daemon 11232)>]

```

janiversen, 2022-11-28:
Well, I am not sure what the problem is, because this one is tested on different platforms and different versions of Python by the CI (with every commit) all without problems.

If it is hanging somewhere, we need to know more about your environment (python platform etc) and please run it with debug enabled as pr documentation. The debug log might give a hint where it is hanging.

HamishSwanbarton, 2022-11-29:
Just to clarify, the base tests do work without problems for me locally. However, when I add the `time.sleep(1)` right before the `ServerStop()` in the test mentioned above, it is only then that the `ServerStop()` doesn't shut the extra threads down. My hope is that this is reproducible.

This commit 3b29057fc1c7aab6bdda7041f30932a2721684e8 shows the changes required to make the test fail.

Environment is pop-os 22.04 LTS (Ubuntu), python 3.10. The output from `tox -e py310` [here](https://github.com/riptideio/pymodbus/files/10112738/log.txt) (while on the above commit).

Thanks for the response, let me know if you'd like anything else.

janiversen, 2022-12-07:
There are clearly a problem, I have found a couple of smaller items, but I am still looking for the main reason. you can follow my progress on branch "test_server_task".

HamishSwanbarton, 2022-12-07:
Thanks for looking into it, best of luck with it!

janiversen, 2023-01-15:
Should be solved with #1256

closed 2023-01-15 by janiversen

HamishSwanbarton, 2023-01-25:
Thank you! This has helped a lot