pymodbus-dev/pymodbus #1203: AsyncModbusTcpClient ignores reconnect_delay == 0

Cavaler, 2022-11-28:
pymodbus 3.0.2

Documentation says:
> reconnect_delay doubles automatically with each unsuccessful connect. Set reconnect_delay=0 to avoid automatic reconnection.

However, creating

Cavaler, 2022-11-28:
--- tcp.old	2022-11-28 12:22:03.442917900 +0200
+++ tcp.py	2022-11-28 18:58:22.041898200 +0200
@@ -69,7 +69,7 @@
         """Stop client."""
 
         # prevent reconnect.
-        self.params.host = None
+        self.params.delay_ms = 0
 
         if self.connected and self.protocol and self.protocol.transport:
             self.protocol.transport.close()
@@ -104,7 +104,8 @@
         except Exception as exc:  # pylint: disable=broad-except
             txt = f"Failed to connect: {exc}"
             _logger.warning(txt)
-            asyncio.ensure_future(self._reconnect())
+            if self.params.delay_ms > 0:
+                asyncio.ensure_future(self._reconnect())
         else:
             txt = f"Connected to {self.params.host}:{self.params.port}."
             _logger.info(txt)
@@ -130,7 +131,7 @@
 
         self.connected = False
         self.protocol = None
-        if self.params.host:
+        if self.params.delay_ms > 0:
             asyncio.ensure_future(self._reconnect())
 
     async def _reconnect(self):

janiversen, 2022-11-28:
Nice enhancement, would you mind submitting a pull request ?

janiversen, 2022-11-28:
Thanks again for the PR, its merged to dev, and will be available in version 3.1

closed 2022-11-28 by janiversen